pavel-demin/red-pitaya-notes #596: RP Network address alpine OS

hb9pae, 2017-10-09:
Ho Pavel
I'm enjoying using the Red Pitaya with your software. I updated now my SD-Card to run the Multiband WSPR TRX using the new OS alpine.
Unfortunately I can't connect to the RP. Wether DHCP nor the static address 192.168.44.2 seems to work.  Even network-scan is not successful. I don't have a WIFI dongle, just connect via networkable.
I believe the SD-Card is correct, the LEDs of Ethernet are flashing.
 
What is the default Network address? How to connect the RP by SSH? 

Thanks in advance.
Peter, hb9pae

pavel-demin, 2017-10-09:
Hello Peter,

All the network configuration is in `/etc/dhcpcd.conf`.

By default, it's configured to request an IP address via DHCP. If no IP address is provided by a DHCP server, then it falls back to a static IP address `192.168.1.100`.

For a fully static address, replace
```
interface eth0
fallback static_eth0
```
with
```
interface eth0
static ip_address=192.168.1.100/24
static routers=192.168.1.1
static domain_name_servers=192.168.1.1
```

Then restart dhcpcd:
```
service dhcpcd restart
```

Don't forget to save the configuration changes to SD card:
```
lbu commit -d
```

Best regards,

Pavel

hb9pae, 2017-10-09:
Hi Pavel
Thank you for support. I got the problem now: my DHCP Server did not respond to RP, using an other Server, all ok.
73 de Peter HB9PAE

closed 2017-10-09 by hb9pae