pavel-demin/red-pitaya-notes #604: XMing and SSH

jkhgroves, 2017-10-12:
Hi Pavel,

Can you share your settings to get SSH (PuTTY) and XMING to work as I'm failing here.

Back in the '90s when I was using Solaris and Exceed it was quite simple but whatever I try to get the X11 windows to work it all just fails.

Thanks in advance,

Regards,

Julian.

pavel-demin, 2017-10-12:
Hi Julian,

There is nothing special in my PuTTY and Xming settings. Enabling X11 forwarding in PuTTY should be enough to get the X11 windows.

Here is a link to one of many how-to guides available on the web:

http://laptops.eng.uci.edu/software-installation/using-linux/how-to-configure-xming-putty

Best regards,

Pavel

jkhgroves, 2017-10-12:
Hi Pavel,

Thanks for that. The link went through all of what I had already done.

As a test I installed x11-apps on my Debain development server and now I can prove X11 windows works.

As a side note I downloaded the Vivado Web Installer package and it's that which is failing.

The Vivado file you link to if over 20GB so I tried the smaller one, looks like I have to bite the bullet and download the huge file.

Kind Regards,

Julian.

closed 2017-10-12 by jkhgroves