pavel-demin/red-pitaya-notes #605: [adc_recorder_trigger] triggering issue

adria-amezaga, 2017-10-12:
Hello,

I'm using the adc_recorder_trigger code to record a repeating signal. An external trigger arrives every few tenths of milliseconds and the goal is to record only a certain portion of the signal after each trigger event and save the data in a file.

Before doing that, I'm trying to record only one segment. It is my understanding that only the immediate next triggering event after the oscilloscope is started should be recorded. However, when I look at the data, it seems that some previous triggering events have remained undetected.

The number of samples before trigger is set to zero.

I can't see why this happens. Any thoughts?

pavel-demin, 2017-10-12:
> It is my understanding that only the immediate next triggering event after the oscilloscope is started should be recorded.

I think that your understanding is correct.

The oscilloscope module does the following steps:
 - waits for `run_flag` pulse
 - starts running and writing the data to a ring buffer
 - writes `pre_data` samples (zero in your case)
 - waits for `trg_flag` pulse
 - writes `tot_data - pre_data`  samples
 - returns to the first step

> However, when I look at the data, it seems that some previous triggering events have remained undetected.

I tested the project by sending a single trigger pulse and it didn't miss that single pulse. I also use the oscilloscope module in the mcpha project and it doesn't miss the trigger pulses.

Could you try to generate a single trigger event and check if it gets detected or not?

adria-amezaga, 2017-10-13:
I have not made the single trigger event check yet but I have performed another test. I modified the Verilog source for the scope (axis_oscilloscope.v) to fit my application. While experimenting I have seen that by removing the s_axis_tvalid condition in the following portion of code where trg_flag is checked, all trigger events are detected correctly. In my case, an external trigger signal is generated every 20ms and the data I expect to see after each trigger is written correctly.

In my modified code, if I remove the comments the external trigger events are detected erratically:

`//      begin
//        if(s_axis_tvalid)
        begin
          if(trg_flag)
          begin
            int_enbl_next = 1'b1;
            int_case_next = 2'd2;
          end
        end
//      end`

pavel-demin, 2017-10-13:
Many thanks for finding the source of the problem!

I think that [this commit](https://github.com/pavel-demin/red-pitaya-notes/commit/b0ebef5866ca4886605a41dbf95d1ca49eb2a394) should fix it.

pavel-demin, 2017-10-13:
BTW. It's also possible to use `run_flag` as a trigger input. I did it in the `radar_test` project:

https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/radar_test/block_design.tcl#L335

closed 2017-10-25 by pavel-demin