pavel-demin/red-pitaya-notes #622: SD card image "Alpine with 8 sdr receivers" does not work

ZL3IO, 2017-11-01:
### Description of the setup:

 * Device: Red Pitaya 14 bit
 * SD card image: https://github.com/pavel-demin/red-pitaya-notes/releases/download/20171025/red-pitaya-alpine-3.6-armhf-20171025.zip
 * Application: SDR Receiver (Skimmer Server)
 * Other relevant information: I heard from others the same trouble and unfortunately could not find anyone who got this working from the image. I found one who did a work around using another Linux version and dealing with the app only.

Cheers from New Zealand
Holger

ZL3IO, 2017-11-01:
Hi Pavel
It was my first post here and parts of my message are not shown in the above text. 

Thanks for the fine work. we really appreciate this. here is the lost message describing what I did. You may have some advise.
I extracted your image directly to an empty sd card and moved start.sh into the root. I also extracted to the pc and copied from the pc to the sd card. In no case the skimmer server software can't find a receiver after powering up the Red Pitaya.
When going back to the six receiver version all works fine.

How would I see from outside if the app program is running?

br Holger

ZL3IO, 2017-11-01:
Hi Pavel
close the issue please. I've tried again with another sd card and it is working just fine now.
cheers Holger

closed 2017-11-01 by ZL3IO

pavel-demin, 2017-11-01:
Hi Holger,

Thank you for confirming that the new SD card image works.

I've just tested it myself. The sdr_receiver_hpsdr app automatically starts without any problem.

I've also updated your first comment and uncommented its first part.

Best regards,

Pavel

ZL3IO, 2017-11-01:
Hi Pavel
Thanks for your reply.

A German HAM (DC2PD) developed a receiver pre-amp with impedance matching. The amplification can be adjusted from the pc software per I2C bus through the red pitaya. You may know this. 
For the CW skimmer application this does not work. Skimmer server software does not offer the feature to adjust the pre-amp. I was wondering if in your SW a block can be implemented that sets the amplification for the pre-amp on the I2C bus. Ideally it would allow this independently per frequency band. On the lower frequencies you will need less pre-amp gain than on higher frequencies. The gain or actually attenuation send to the pre-amp will typically only need adjustment once per band and would not change unless another antenna would be used. It may need some testing /adjustment at the beginning.

I was almost hardware modifying the pre-amp to set to 10 dB but a HW modification is not reversible if I ever want to use the SW control. 

Br Holger

pavel-demin, 2017-11-01:
> I was wondering if in your SW a block can be implemented that sets the amplification for the pre-amp on the I2C bus.

On the Red Pitaya side, it's possible to write a small C program to control the variable gain amplifier.

On the PC side, the most difficult part is as always the user interface. Would you be interested to contribute this part?

> Ideally it would allow this independently per frequency band.

I don't understand how it can be done with the CW skimmer server receiving on 8 bands simultaneously.