pavel-demin/red-pitaya-notes #628: Connection refused

PrimLem, 2017-11-17:
### Description of the setup:

 * Device: Red Pitaya (RP)
 * SD card image: red-pitaya-alpine-3.6-armhf-20171025.zip
 * Application: vna-win32-20171012.zip
 * Other relevant information: Windows 7, local IP 192.168.10.x, IP mask 255:255:255:0, MAC 00:26:32:F0:0F:73, both the program and the image downloaded from the links here: http://pavel-demin.github.io/red-pitaya-notes/vna/

### Description of the problem:
The RP image properly installs and boots. The device gets a DHCP address. The vna.exe application starts but when I press the Connect button I receive "Error: Connection Refused" message after a few seconds.
Is it maybe required strictly that the 192.168.1.x subnet is used?

### Steps to reproduce the problem:

 1. Reboot the RP and wait that it is accessible in the local network
 2. Run the vna.exe program (precompiled)
 3. Insert the RP IP address and press Connect button

pavel-demin, 2017-11-18:
How are you starting the VNA application on Red Pitaya?

PrimLem, 2017-11-21:
So far I haven't started it specially. I was sure it was running by default on the RP as soon as it is installed. I tried both options: your image with VNA pre-installed and latest RP SW + VNA application from the marketplace. The result was equal in both cases.

I somehow overlooked that the "start.sh" needs to be copied to the root folder. So I did it now and rebooted the RP, but the result is still the same. How can I see that the VNA application is running on the RP using the ps -ax command, for instance?

If I start the VNA application manually from the command line using "/opt/redpitaya/www/apps/vna/vna &" command I get the following if I check with the "ps -ax"

"[1]+  Bus error               /opt/redpitaya/www/apps/vna/vna"

Do you think it is possible that there is something wrong with my RP (HW-wise) as I am also having problems with my USB serial port occassionaly.

Thanks and regards,

Primoz

pavel-demin, 2017-11-21:
> I somehow overlooked that the "start.sh" needs to be copied to the root folder. So I did it now and rebooted the RP, but the result is still the same.

With `start.sh` in the root folder, the application starts automatically at boot. It's also possible to start the application from the web interface.

> How can I see that the VNA application is running on the RP using the ps -ax command, for instance?

`ps aux | grep vna` should do the job.

> If I start the VNA application manually from the command line using "/opt/redpitaya/www/apps/vna/vna &" command I get the following if I check with the "ps -ax"
> "[1]+ Bus error /opt/redpitaya/www/apps/vna/vna"

It's better to use `start.sh`. This script configures the FPGA before running the program. The error message is most probably due to the FPGA not being correctly configured.

pavel-demin, 2017-11-21:
> I somehow overlooked that the "start.sh" needs to be copied to the root folder.

I've just realized that from your description it's not clear what is the root folder.

If you copy `start.sh` from the Red Pitaya command line, then the path to the topmost directory on the SD card is `/media/mmcblk0p1`.

PrimLem, 2017-11-24:
This worked, thanks. I was not able to perform file copy to/media/mmcblk0p1 in command line as the file system is read-only but in a card reader instead.

closed 2017-11-24 by PrimLem

pavel-demin, 2017-11-25:
Great that it finally works!

For information. `/media/mmcblk0p1` can be remounted for writing with the `mount -o rw,remount /media/mmcblk0p1` command or with the `rw` alias.