pavel-demin/red-pitaya-notes #630: QAM in GNU radio with Red Pitaya

sabine95, 2017-11-21:
### Description of the setup:

 * Device: Red Pitaya
 * SD card image: Red Pitaya for Gnuradio
 * Application: GnuRadio
 * Other relevant information:

### Description of the problem:
Hey,
I've a problem with my Red Pitaya and Gnuradio. I'm using Ubuntu and I installed the Block for the Red Pitaya but when i'm starting a Programm there is no error message but nothing is shown on my oscilloscope.

Im trying to do a Gnuradio Programm and it should QAM and PSK playing and shown them on the oscilloskope.
Is there any solution for it I dont know?

Please help!

Sabine

(sry for my bad english :) )

pavel-demin, 2017-11-21:
I think that you need to enable PTT in the Red Pitaya sink block

sabine95, 2017-11-21:
Thank you so much!!

closed 2017-11-21 by pavel-demin