pavel-demin/red-pitaya-notes #635: Operating under Windows 10

davidhay45, 2017-11-25:
Very impressive, thanks.

I am running Windows 10 and my radio shack is remote from my router and connects by WiFi.  Therefore it is not convenient to connect my Red Pitaya by cable and to connect by WiFi using the default WiFi hotspot on the pitaya I have to disconnect from the internet.  I need to set up the Red Ptaya as a wifi client, this can't be difficult but what is the best way?  Is there a boot ini file?  Could you possibly add this sort of thing to the very informative notes?

As a Windows person, albeit one who spent many years as a software developer in Visual C++, my knowledge of the Linux environment is sketchy.  I may decide to set up a Linux PC but time will tell.  Are their others who use Windows XP onwards where knowledge could be pooled?

Many thanks

David M0TGC

pavel-demin, 2017-11-25:
> I need to set up the Red Ptaya as a wifi client, this can't be difficult but what is the best way?

If you're using the SD card image based on Alpine Linux (red-pitaya-alpine-3.6-armhf-20171025.zip), then switching to the client Wi-Fi mode can be done in two relatively easy steps:

1. Connect to the Red Pitaya command line interface via SSH as described at [this link](http://redpitaya.readthedocs.io/en/latest/developerGuide/os/ssh/ssh.html). For the SD card image based on Alpine Linux, the login is `root` and the password is `changeme`.

2. Type the first three commands from [this link](http://pavel-demin.github.io/red-pitaya-notes/alpine/#useful-commands). Here is a copy of these commands:
```
wpa_passphrase SSID PASSPHRASE > /etc/wpa_supplicant/wpa_supplicant.conf
./wifi/client.sh
lbu commit -d
```
`SSID` and `PASSPHRASE` should be replaced with the correct parameters of your Wi-Fi network.

davidhay45, 2017-11-26:
Thanks, this worked but the delays over wifi were two great so I have connected direct to the PC using another network card configured with an ip address of 192.168.1.202.

closed 2017-11-26 by pavel-demin