pavel-demin/red-pitaya-notes #636: Different signal levels in RX1 and RX2

davidhay45, 2017-11-26:
### Description of the setup:

 * Device:
 * SD card image: red-pitaya-alpine-3.6-armhf-20171025
 * Application: SDR receiver compatible with HPSDR
 * Other relevant information: Windows 10 wired to RP 

### Description of the problem:
The signal levels of RX1 and RX2 are 3 to 5db different, RX2 is lower.  Why is that?

### Steps to reproduce the problem:

 1. Tune RX1 and RX2 to the same frequency.  Set all other controls the same (ATT, SQL, AGC etc)
 2.
 3.

dl1sdz, 2017-11-26:
Check the jumpers from input 1 and 2

73 de

Hajo dl1sdz

---
Cela est bien dit, mais il faut cultiver notre jardin.

http://hajos-kontrapunkte.blogspot.de/

pavel-demin, 2017-11-26:
> Application: SDR receiver compatible with HPSDR
> The signal levels of RX1 and RX2 are 3 to 5db different, RX2 is lower. Why is that?

Can't reproduce this problem.

Here is a screenshot showing RX1 and RX2:

[image]

SDR receiver compatible with HPSDR is running in the default configuration with both RX1 and RX2 connected to IN1.

davidhay45, 2017-11-26:
I can't reproduce it now.  If I see it again I will take a screen shot.

Thanks for looking.

closed 2017-11-27 by pavel-demin

davidhay45, 2017-11-27:
I think I am confused as to how to map RX1 and RX2 to IN1, how is it done please?

pavel-demin, 2017-11-27:
It can be done by editing the last line in [apps/sdr_receiver_hpsdr/start.sh](https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/sdr_receiver_hpsdr/app/start.sh).

The sdr-receiver-hpsdr server has eight command line arguments. Each argument is the input number (1 or 2) corresponding to each of the eight receivers.