pavel-demin/red-pitaya-notes #637: HPSDR transceiver gives firmware error

davidhay45, 2017-11-27:
### Description of the setup:

 * Device:Red Pitaya
 * SD card image:red-pitaya-alpine-3.6-armhf-20171025
 * Application: HPSDR transceiver
 * Other relevant information:Running under widows 10

### Description of the problem:

HPSDR receiver works well on my setup but when I change app to the HPSDR transceiver it gives an error, see screenshot.

It is the same with hardware options Mecury/Penelope and Mecury/Pennylane

[image]

pavel-demin, 2017-11-27:
> when I change app to the HPSDR transceiver it gives an error, see screenshot

I think that it only happens when the Radio Model is set to HPSDR. Setting it to HERMES should solve the problem.

The board ID incompatible with the HPSDR radio model was requested in [this issue](https://github.com/pavel-demin/red-pitaya-notes/issues/501) for better compatibility with [pihpsdr](https://github.com/g0orx/pihpsdr).

pavel-demin, 2017-11-27:
BTW. The firmware check can be disabled via Setup->General->HPSDR->Bypass Firmware Check.

closed 2017-11-27 by pavel-demin

davidhay45, 2017-11-27:
Ok I understand.  What is the correct config to set in PowerSDR?  Is it HPSDR + Mercury/Pennylane/Alex?

pavel-demin, 2017-11-27:
> What is the correct config to set in PowerSDR?

HERMES + Alex