pavel-demin/red-pitaya-notes #639: Development Machine

davidhay45, 2017-11-29:
Hi

I am trying to set up a development machine on my Windows 10 PC following the instructions given.

I think I am almost there (!), Under the VM I can login via root and I have downloaded the Xilinx_Vivado_SDK_2016.4_1215_1.tar.gz and unpacked it but I can't run the installer because of the lack of X11.  So I have run SSH with X11 enabled but I can't login to the VM via SSH with root/changeme (password wrong) and when I log in via red-pitaya/changeme I don't have permission to create the xsetup via sed.

So, is the root password via SSH something else?  Or what should I do?  I have many years experience of Windows and Visual C++ but I am a beginner at Linux as you will have guessed.

I very much appreciate all the work you have done here it is amazing.

Thanks

David

dl1sdz, 2017-11-29:
Just a guess:

Root login via ssh could be forbidden. So ssh login with RedPitaya and then
change to su or use sudo.

Gruss
Hajo

---
Cela est bien dit, mais il faut cultiver notre jardin.

http://hajos-kontrapunkte.blogspot.de/

pavel-demin, 2017-11-29:
Hajo is right.

The root login via SSH is disabled by default. So, the easiest way to the root login is to login via SSH as red-pitaya and then run `su -`.

davidhay45, 2017-11-29:
Spot on. 

Thanks

davidhay45, 2017-11-29:
[image]

This is like peeling an onion only as I get closer to the core the layers get thinner and there are more of them.

Can anyone give me a clue to the "unsupported auth protocol" please?

pavel-demin, 2017-11-29:
If you login via SSH as red-pitaya and switch to root, then you can't run X11 applications from the root account.

There are several possible solutions:
 - enable root login via SSH,
 - run the installation script from the red-pitaya account.

To run the installation script from the red-pitaya account, you'll need to change the permissions for the /opt/Xilinx directory.
```
su -
mkdir /opt/Xilinx
chown red-pitaya:red-pitaya /opt/Xilinx
exit
```

davidhay45, 2017-11-29:
Next layer reached, thanks.  I now get can't connect to internet to check SDK version and I can't ignore it because I don't have mouse control so more config to do.

Thanks again.

David

davidhay45, 2017-11-29:
Got past the mouse click issue!

closed 2017-11-29 by davidhay45

davidhay45, 2017-11-30:
Well the install completed.  What sort of licence do I need?  For some reason the VM does not have internet connectivity so that is what I will look at next.

pavel-demin, 2017-11-30:
> What sort of licence do I need?

WebPACK. It's enabled by default but it probably requires an Internet connection.

davidhay45, 2017-11-30:
Thanks.  I have fixed the internet connectivity.  How do I restart the licence manager?

Is there debian forum for Linux help?

pavel-demin, 2017-11-30:
> How do I restart the licence manager?

I don't know. I think that it should work without restarting the licence manager. I'd suggest to build the LED blinker project (`make NAME=led_blinker bit`). If it works and outputs led_blinker.bit, then the licence is OK.

> Is there debian forum for Linux help?

https://forums.debian.net

However, I'd suggest to start with some reading. Here are some links that look interesting:

http://www.ee.surrey.ac.uk/Teaching/Unix
https://ryanstutorials.net/linuxtutorial
https://debian-handbook.info
https://www.debian.org/doc

davidhay45, 2017-11-30:
Your support is great, thanks.

Final question for a while (I hope).  What directory structure is expected?  IE where do the app files go and what directory do I run make from?

Thanks

pavel-demin, 2017-11-30:
> What directory structure is expected? IE where do the app files go and what directory do I run make from?

Please have a look at the [LED blinker](http://pavel-demin.github.io/red-pitaya-notes/led-blinker).

I'd say that the 'Source code' and 'Getting started' sections have the answers to your questions.

You can find more details about the directory structure and about my toolchain for ZYNQ in my Club Vivado presentation (slides 90-111):
http://www.origin.xilinx.com/publications/products/vivado/club_vivado/2016/einhoven-2016-presentations.zip

davidhay45, 2017-11-30:
I have done 

make NAME=led_blinker all

and it appeared to ok.  Presumably if it had errors it would say so?

When I try and build a bootable SD card I get:

sudo: /usr/bin/sudo must be owned by uid 0 and have the setuid bit set

Have I gone wrong somewhere?

pavel-demin, 2017-12-01:
> When I try and build a bootable SD card I get:
> sudo: /usr/bin/sudo must be owned by uid 0 and have the setuid bit set
> Have I gone wrong somewhere?

The error message looks pretty scary. Could you check what the following command returns?
```
ls -l /usr/bin/sudo
```

On my system it returns
```
-rwsr-xr-x 1 root root 157760 May 29  2017 /usr/bin/sudo
```

davidhay45, 2017-12-01:
-rwxr-xr-x 1 red-pitaya red-pitaya 157760 May 29  2017 /usr/bin/sudo

I think I must have done it for the /usr/bin/* to try and overcome an older problem.  I suspect I should chown /usr/bin back to root?

I am away until Tuesday but a question, I read your presentation for the Eindhoven conference and would like to comment but not via the forum, is that ok with you?

pavel-demin, 2017-12-01:
>  I suspect I should chown /usr/bin back to root?

Yes. And you should also set the setuid bit:
```
chmod u+s /usr/bin/sudo
```

> I read your presentation for the Eindhoven conference and would like to comment but not via the forum, is that ok with you?

Yes, no problem. You can find my e-mail address in [my profile](https://github.com/pavel-demin).

davidhay45, 2017-12-05:
I am confused by the two types of SD card image, Debian and Alpine.  It would appear that you are only now using the Alpine version, is that correct?

Rolf-L, 2017-12-05:
Both versions are working fine.
Many thanks Pavel for your excellent work.
Rolf -DJ7TH-
www.dj7th.de

davidhay45, 2017-12-05:
Whilst that may be true, are both versions current and why would you use one rather than the other?

pavel-demin, 2017-12-05:
> I am confused by the two types of SD card image, Debian and Alpine. It would appear that you are only now using the Alpine version, is that correct?

Correct. I'm mostly using the SD card image based on Alpine Linux.

> Whilst that may be true, are both versions current and why would you use one rather than the other?

The SD card image based on Debian Linux is usable. However, I can think of the following problems with this image:

* its Linux kernel doesn't have the modules that I recently added to my scripts and patches,
* it can't be rebuilt with the current version of the scripts and patches  (the scripts still have to be adapted to the changes introduced with [this pull request](https://github.com/pavel-demin/red-pitaya-notes/pull/623)).

For me, the SD card image based on Alpine Linux has the following advantages over the SD card image based on Debian Linux:

* fast to build (much faster than the SD card images based on Debian and Ubuntu),
* fast to download (it's much smaller than the SD card images based on Debian and Ubuntu),
* easy to install (just unpack the .zip file to a FAT formatted micro SD card),
* easy to maintain (since it contains all the pre-built applications, I have only one SD card image to maintain).

pavel-demin, 2017-12-05:
I've just updated the pre-built SD card image based on Debian and the debian and ubuntu scripts with [this commit](https://github.com/pavel-demin/red-pitaya-notes/commit/43a9576d0d51f31865300e376ca695e5331002dc). They are now compatible with the current version of the scripts and patches that I use to build Linux.

davidhay45, 2017-12-05:
I have executed the following:

cd /opt/Xi*
cd X*

source /opt/Xilinx/Vivado/2016.4/settings64.sh

git clone https://github.com/pavel-demin/red-pitaya-notes

cd red-pitaya-notes

make NAME=sdr_transceiver_hpsdr bit

It ran for some minutes and then stopped with the message in the screenshot whereit has stayed for the last five hours and is still apparently running.  Is this correct?  The log files do not appear to exist.

[image]

pavel-demin, 2017-12-06:
The messages on the screenshot look OK. The impl_1 step takes long time to complete. However, not more than half an hour for the sdr_transceiver_hpsdr project.

I think that building sdr_transceiver_hpsdr requires more memory than is available on your virtual machine. It starts to swap and everything slows down.

I'd suggest to increase the memory size to 3 or 4 GB.

davidhay45, 2017-12-06:
I increased the memory to 4gb and then re-ran as below.  As expected it skipped the bits it had done but then it gave the error as below.  I had rebooted the PC before starting the job again.  So what is:

ERROR: [Common 17-69] Command failed:  Run'impl_1' is already running and cannot be relaunched.

Any ideas?

<<<
red-pitaya@debian:/opt/Xilinx/Xilinx_Vivado_SDK_2016.4_1215_1/red-pitaya-notes$ make NAME=sdr_transceiver_hpsdr bit
mkdir -p tmp
vivado -nolog -nojournal -mode batch -source scripts/bitstream.tcl -tclargs sdr_transceiver_hpsdr

****** Vivado v2016.4 (64-bit)
  **** SW Build 1733598 on Wed Dec 14 22:35:42 MST 2016
  **** IP Build 1731160 on Wed Dec 14 23:47:21 MST 2016
    ** Copyright 1986-2016 Xilinx, Inc. All Rights Reserved.

source scripts/bitstream.tcl
# set project_name [lindex $argv 0]
# open_project tmp/$project_name.xpr
Scanning sources...
Finished scanning sources
INFO: [IP_Flow 19-234] Refreshing IP repositories
INFO: [IP_Flow 19-1700] Loaded user IP repository '/opt/Xilinx/Xilinx_Vivado_SDK_2016.4_1215_1/red-pitaya-notes/tmp/cores'.
INFO: [IP_Flow 19-2313] Loaded Vivado IP repository '/opt/Xilinx/Vivado/2016.4/data/ip'.
open_project: Time (s): cpu = 00:00:07 ; elapsed = 00:00:07 . Memory (MB): peak = 1084.113 ; gain = 89.605 ; free physical = 3068 ; free virtual = 4591
# if {[get_property PROGRESS [get_runs impl_1]] != "100%"} {
#   launch_runs impl_1 -to_step route_design
#   wait_on_run impl_1
# }
ERROR: [Common 17-69] Command failed:  Run'impl_1' is already running and cannot be relaunched.
INFO: [Common 17-206] Exiting Vivado at Wed Dec  6 12:14:35 2017...
Makefile:152: recipe for target 'tmp/sdr_transceiver_hpsdr.bit' failed
make: *** [tmp/sdr_transceiver_hpsdr.bit] Error 1
red-pitaya@debian:/opt/Xilinx/Xilinx_Vivado_SDK_2016.4_1215_1/red-pitaya-notes$
>>>

pavel-demin, 2017-12-06:
> Command failed:  Run 'impl_1' is already running and cannot be relaunched.

This is a part that isn't handled well by my scripts. I just remove all the project's files before rerunning make:
```
rm -rf tmp/sdr_transceiver_hpsdr*
make NAME=sdr_transceiver_hpsdr bit
```

davidhay45, 2017-12-06:
I found this, it suggests that jobs running concurrently is the issue.

https://forums.xilinx.com/t5/Synthesis/Error-running-Synthesis-via-quot-Generate-Bitstream-quot/td-p/444580

pavel-demin, 2017-12-06:
I'm not sure that the link is about the same problem.

I think that when you stopped the long running build process, Vivado didn't properly remove some files indicating that it was running.

When you restarted the build process, Vivado found the files indicating that it was still running and decided to output the "Run 'impl_1' is already running" error message.

So, I still think that the easiest solution is to remove all the project's files before rerunning make.

davidhay45, 2017-12-06:
However, doing as you suggest appears to have worked, so thank you.  It took about 10 mins to complete the impl_1 step.

During the complete build there were quite a lot of warning messages, eg

WARNING: [BD 5-235] No pins matched 'get_bd_pins fir_1/S_AXIS_DATA'
and
WARNING: [BD 41-1306] The connection to interface pin /tx_0/mult_0/s_axis_b_tdata is being overridden by the user. This pin will not be connected as a part of interface connection S_AXIS_B

plus numerous others.

So my questions now are:
 do the warnings matter?
 how can I know the build worked?

I should explain that I am trying to get to the stage where I am running sdr_transceiver_hpsdr with a system that I built and I can then try various things.

Fascinating stuff, takes me back to my days as a software engineer with Philips Telecom! Thanks.

pavel-demin, 2017-12-06:
> do the warnings matter?

Normally, I only worry about critical warnings. From time to time, I also check the non-critical warnings, but most of the time, I don't know what to do about them.

> how can I know the build worked?

If you have `tmp/sdr_transceiver_hpsdr.bit`, then it's already a good sign.

You can also open the project in Vivado (`vivado tmp/sdr_transceiver_hpsdr.xpr`) and check the 'Project Summary' and 'Design Runs' tabs. If nothing is red, then I consider it OK.

Suggested reading: [UG900](https://www.xilinx.com/support/documentation/sw_manuals/xilinx2016_4/ug900-vivado-logic-simulation.pdf), [UG906](https://www.xilinx.com/support/documentation/sw_manuals/xilinx2016_4/ug906-vivado-design-analysis.pdf).

davidhay45, 2017-12-06:
I do indeed have the .bit file!

My final step is to make a zip of the SD card but your script source (helpers/build-all.sh) references the SD card via mmcblk0p1 which I don't think I have access to from a VM.

What would you advise me to do?  I don't really need any more than sdr_transceiver_hpsdr and so far I have not "made" the other apps.

pavel-demin, 2017-12-06:
> your script source (helpers/build-all.sh) references the SD card via mmcblk0p1 which I don't think I have access to from a VM.

The script works without the SD card. It creates all the required directories including mmcblk0p1.

pavel-demin, 2017-12-06:
alpine.sh requires an additional package:
```
apt-get install squashfs-tools
```

davidhay45, 2017-12-07:
I am trying to get alpine.sh to work in my environment.  I see that the script uses chroot and appears to pipe a set of commands into dash.  I don't understand what this is doing, is there something I can read?

Is it feasible just to recompile sdr_transceiver_hpsdr and replace it on the SD card?

pavel-demin, 2017-12-07:
> I am trying to get alpine.sh to work in my environment.

Normally, it should work out of the box. What error messages do you see?

> Is it feasible just to recompile sdr_transceiver_hpsdr and replace it on the SD card?

Yes. In case of a pre-built SD card based on Alpine Linux, just copy sdr_transceiver_hpsdr.bit to apps/sdr_transceiver_hpsdr and rerun the start.sh script or restart the application via the web interface.

pavel-demin, 2017-12-07:
> I don't understand what this is doing, is there something I can read?

It's a combination of chroot and so called here document:
https://en.wikipedia.org/wiki/Chroot
http://tldp.org/LDP/abs/html/here-docs.html

davidhay45, 2017-12-08:
I don't think the SD card peripheral is available to the VM under Windows 10 so I modified the alpine.sh script to change mmcblk0p1 to sdcard. After the script had run I searched for sdcard and found it at:

/opt/Xilinx/Xilinx_Vivado_SDK_2016.4_1215_1/red-pitaya-notes/alpine-initramfs?/alpine-root?/media/sdcard

where the ? appears to be a \r.  I can't get at the contents of sdcard, perhaps because it gets used under the chroot jail.  Any ideas?

pavel-demin, 2017-12-09:
> I don't think the SD card peripheral is available to the VM under Windows 10 so I modified the alpine.sh script to change mmcblk0p1 to sdcard.

The alpine.sh script doesn't require any SD card. It creates a .zip file that should be unpacked to a FAT formatted micro SD card.

If you rename mmcblk0p1 to sdcard, the resulting .zip file won't work properly.

> Any ideas?

At the moment, I don't understand what problem you're trying to solve. Do you see any error messages when running the unmodified alpine.sh script?

davidhay45, 2017-12-09:
I see errors as it tries to compile transceiver_emb, snb.c and the zip command fails as the zip utility is not installed.  The problem is that I want to create a sd card and I can't.

pavel-demin, 2017-12-09:
> I see errors as it tries to compile transceiver_emb, snb.c

I've just rerun the alpine.sh script and I see many warnings when it builds the wdsp library but no errors.

>  the zip command fails as the zip utility is not installed

Installation of the zip utility was indeed missing from my notes. I've just added it with [this commit](https://github.com/pavel-demin/red-pitaya-notes/commit/0816a08387a3da2d48fe499c636adbaa040eb065).

You can install it with the following command:
```
apt-get install zip
```

davidhay45, 2017-12-09:
That appeared to work but still gave warnings re transceiver_emb, why?

pavel-demin, 2017-12-09:
> That appeared to work but still gave warnings re transceiver_emb, why?

Most of the warnings are from the compilation of the [WDSP library](https://github.com/g0orx/wdsp). If you think that some or all of them should be fixed, then I'd suggest to contact the developers of the [WDSP library](https://github.com/g0orx/wdsp).

davidhay45, 2017-12-10:
I am now running my RP on an SD card that I have generated so this stage is complete. Many thanks.

pavel-demin, 2017-12-10:
Great! Thanks for going through the instructions and for highlighting all the problems!

davidhay45, 2017-12-12:
I thought I would summarize the problems I came across:

Initial Vivado loading has to be with a graphical interface loaded, I had difficulty in getting X11 to work.
Under a SSH client (I used PuTTY) the root login is disabled.
I was confused by the request to obtain a licence for Vivado, the answer was to ignore the request.
There were apps I needed that had to be loaded manually, squashfs-tools and zip.
The make process virtually stopped because I had only allocated the Virtual Machine 2048mb of memory, On this was increased to 4096mb it went OK.
I was expecting the make all job to be a lot quicker but it got there in the end.

Thanks again to Pavel for all his help.

davidhay45, 2017-12-15:
Hi

I decided to look at the actual apline.sh script I had used and tried to open the VM.  It gave the following:

[image]

The only thing I can think of is that this is a problem with Oracle VM Virtualbox and has shown up because I have upgraded my Windows 10 to the 1709 version.  Have you seen this before?

pavel-demin, 2017-12-15:
It works fine with Windows 7. I don't have Windows 10. I see that a similar problem is discussed on the VirtualBox forum:
https://forums.virtualbox.org/viewtopic.php?f=6&t=82249
https://forums.virtualbox.org/viewtopic.php?f=6&t=85395

pavel-demin, 2017-12-15:
One of the posts on the VirtualBox forum suggest reinstalling VirtualBox to solve this problem:
https://forums.virtualbox.org/viewtopic.php?f=6&t=83696#p405149

davidhay45, 2017-12-16:
Reinstalling VirtualBox removed the “invalid data” but now I can’t PuTTY to the VM, it gives “host not found”.  I have posted this to the virtualbox forum.

Rgds

David

From: Pavel Demin 
Sent: Friday, December 15, 2017 19:14
To: pavel-demin/red-pitaya-notes 
Cc: davidhay45 ; State change 
Subject: Re: [pavel-demin/red-pitaya-notes] Development Machine (#639)

One of the posts on the VirtualBox forum suggest reinstalling VirtualBox to solve this problem:
https://forums.virtualbox.org/viewtopic.php?f=6&t=83696#p405149

davidhay45, 2017-12-19:
The VirtualBox forum provided a solution to this.

https://forums.virtualbox.org/viewtopic.php?f=6&t=85910&p=409261#p409261

davidhay45, 2017-12-20:
Is it possible to compile a c file on the RP under Alpine?  Alternatively is it possible to use ftp on the RP into the Debian VM or direct into my Win 10 PC?

Thanks

pavel-demin, 2017-12-20:
> Is it possible to compile a c file on the RP under Alpine?

Yes. The compiler is on the SD card but not installed by default. The compiler and make can be installed with the following command:
```
apk add make gcc
```
If you prefer to keep them installed, then write the changes to the SD card with the following command:
```
lbu commit -d
```

> Alternatively is it possible to use ftp on the RP into the Debian VM or direct into my Win 10 PC?

SCP can be used to transfer files. It works with both a Linux command line and with graphical clients like WinSCP:
https://winscp.net

davidhay45, 2017-12-21:
I think winscp is wonderful and really works well with my Windows 10/Debian/RP.  I am not sure if you mention it elsewhere but it is worth it.