pavel-demin/red-pitaya-notes #648: error while making adc_test project

parveennisha, 2017-12-07:
Hi, Pavel
I am trying to run adc_test project from your red-pitaya-notes. I am following
 https://github.com/pavel-demin/red-pitaya-notes/tree/master/projects/bare_metal_test t 
this after running 
make NAME=adc_test all
i am getting the following error.

make[1]: Leaving directory `/home/optics-19/red-pitaya-notes/tmp/linux-xlnx-xilinx-v2016.4'
make: *** [uImage] Error 2

please help how to run adc_test project.

Thanks for the help.

pavel-demin, 2017-12-07:
Hi,

I'm not sure if the bare_metal_test instructions are usable with the adc_test project.

The FPGA configuration bitstream can be built with
```
make NAME=adc_test bit
```

Then `adc_test.bit` can be used with any pre-built SD card image.

parveennisha, 2017-12-07:
i tried using above instuction i am getting this error

 make NAME=adc_test.bit
make: *** No rule to make target `tmp/adc_test.bit.fsbl/executable.elf', needed by `boot.bin'.  Stop.

pavel-demin, 2017-12-07:
The source of the problem is the dot that you've added between adc_test and bit.

parveennisha, 2017-12-07:
I tried this first and got this

make NAME=adc_test bit
make: Nothing to be done for `bit'

pavel-demin, 2017-12-07:
> make: Nothing to be done for `bit'

It means that everything is OK and the .bit file (`tmp/adc_test.bit`) is already built.

parveennisha, 2017-12-08:
Thank you for the help it is working.

closed 2017-12-08 by pavel-demin