pavel-demin/red-pitaya-notes #649: direct access of RF ports and slow analog pins

parveennisha, 2017-12-08:
Hi Pavel,
    Actually we have tried accessing the slow analog pins using the API which gives upto 1 kHz with noise. The regenerated output from the RF ports using API is discontinuous due to delay in the acquisition and generation.

We have also tried using programming logic without using ZYNQ processor, but the bit file is not been dumped.  

Is there any project related to direct access of  RF ports and slow analog pins at 100 kHz? If not can you suggest what can be done. 

Thanks.

pavel-demin, 2017-12-08:
Hi Parveen,

I'm afraid that I don't clearly understand what problem you're trying to solve and what do you mean by "direct access". It would be helpful if you could provide some more details.

If your project requires continuously reading the ADC samples at 125 million samples per second, then I think that the only solution is to develop a custom FPGA (programmable logic) configuration according to the requirements of your project.

The slow ADC (XADC) can be configured to continuously output data via its AXI4-Stream interface. You can find more details about XADC in [PG091](https://www.xilinx.com/support/documentation/ip_documentation/xadc_wiz/v3_3/pg091-xadc-wiz.pdf).

So, if your project requires reading the data from both fast ADC and slow ADC, then I'd say that your FPGA configuration should at least have two modules:
- [xadc-wiz](https://www.xilinx.com/products/intellectual-property/xadc-wizard.html) for reading the data from the slow ADC,
- [axis_red_pitaya_adc](https://github.com/pavel-demin/red-pitaya-notes/tree/master/cores/axis_red_pitaya_adc_v2_0) or something similar for reading the data from the fast ADC.

Best regards,

Pavel

parveennisha, 2017-12-09:
Hey Pavel,

First of all, thank you for the quick reply!!

Actually we are trying to **regenerate** (generate the same data that are continuously acquired by the ADC) a analog signal using slow analog pins and fast analog ports individually. 

As mentioned above, using slow analog ports with API works fine only till 1 kHz but after that its frequency decreases and the output is distort, 

We have also tried to Regenerate a signal using the fast ADC with API, the output is discontinuous, there is a data loss at every 1 millisecond.

So now we are trying to accesses the slow analog ports and fast analog ports separately using FPGA so that we can get regenerated efficient output of at least 100 kHz. 

Hope you can help us with necessary guidance. Thanks again

With Regards,
Parveen

pavel-demin, 2017-12-09:
Thanks for the details about your project.

I once posted on the Red Pitaya forum a very simple project with the output of the ADC module directly connected to the input of the DAC module. Here is a link to this post:
http://forum.redpitaya.com/viewtopic.php?t=1785&p=8212#p6868

I've just updated the projects to make it compatible with Vivado 2016.4. Here is a link to this project:
https://www.dropbox.com/sh/5fy49wae6xwxa8a/AACdFuF0AOc1hL3ePXU3D44ya/dac_adc.zip?dl=1

I've also added the source code of this project to this repository:
https://github.com/pavel-demin/red-pitaya-notes/tree/master/projects/dac_adc

I don't use the PWM pins of the Red Pitaya board in my projects. So, I can't help you with the PWM pins.

closed 2017-12-11 by pavel-demin