pavel-demin/red-pitaya-notes #651: Control of LPF and BPF boards

davidhay45, 2017-12-10:
Using sdr_transceiver_hpsdr. It looks like band switching controls the pins on the extension socket E1 of the RP pcb via a status register (sts) in the C source file. Is there a spec of the E1 pin usage please.

pavel-demin, 2017-12-10:
Have you seen the information at the following link?
http://pavel-demin.github.io/red-pitaya-notes/sdr-transceiver-hpsdr

davidhay45, 2017-12-10:
Yes but I'd forgotten where it was, quite the obvious place.  Is the communication controlled via the sts register in the C file?

closed 2017-12-10 by davidhay45

pavel-demin, 2017-12-10:
>  Is the communication controlled via the sts register in the C file?

pins DIO0_N - DIO3_N connected to the sts register.
pins DIO0_P - DIO7_P connected to the cfg register.