pavel-demin/red-pitaya-notes #652: Redpitaya with hardware manager connectivity

Abinayasivam, 2017-12-11:
Hi Pavel, 
I am trying to connect redpitaya in vivado hardware manger, it is getting programmed but i am not able to find ILA cores. I am using vivado 2015.4 in windows 7 using putty please refer attached image.

[image]

If i wanted to program the ltx file using putty what is the command?

thank you for the help

pavel-demin, 2017-12-11:
> If i wanted to program the ltx file using putty what is the command?

I don't know what is the ltx file.

Normally, I use .bit files to program the Red Pitaya's FPGA.

It's possible to copy a .bit file from Windows to Red Piataya using [WinSCP](https://winscp.net). Once the .bit file is copied to Red Pitaya, I use the following command to program the FPGA:
```
cat led_blinker.bit > /dev/xdevcfg
```
This command should be run from the Red Pitaya command line when you connect to it using [PuTTY](http://www.putty.org).

closed 2017-12-11 by pavel-demin

Abinayasivam, 2017-12-11:
Hi Pavel,
Normally after dumping the bit file in vivado, we can dump the  ltx file(debug_nets.ltx) for vivado analyzer
ltx file is the probe files for Integrated logical Analyzer(ILA). I wanted to see my output in ILA. So I just connected the output of each blocks in ILA(Debugging the nets) while generating the bit file.

I have dumped the bit file and ltx file in redpitaya using hardware manager. but I am not able to see any ILA cores at hardware manager.

pavel-demin, 2017-12-11:
Thanks for the explanation.

Abinayasivam, 2017-12-11:
So, how can  you  verify the output at each IP blocks whether it is working correctly or not?

Thanks,
Abinaya

pavel-demin, 2017-12-11:
> So, how can you verify the output at each IP blocks whether it is working correctly or not?

No idea :smile:

Since I never used the FPGA debugging tools like ChipScope, SignalTap, ect, I don't think that I'm the right person to answer this question.

Maybe it could be done via the JTAG interface of the Red Pitaya board or maybe there are other solutions.