pavel-demin/red-pitaya-notes #653: How to see output of adc_test

parveennisha, 2017-12-11:
Hi Pavel,

I wanted to access DDR RAM regarding that i was referring your adc_test project. I am not able to understand what is happening in the project and how to see the output. I was using monitor statement with address 0x00000008 in redpitaya window and it was giving some hex value. 

Thanks for the help.

pavel-demin, 2017-12-11:
The main purpose of the adc_test project was to test how fast the ADC samples could be transferred to a remote PC. Here is a link to a post on the Red Pitaya forum with the results of those tests:
http://forum.redpitaya.com/viewtopic.php?t=317&p=2118#p2118

You can find more details about the adc_test project in [this issue](https://github.com/pavel-demin/red-pitaya-notes/issues/528).

At the moment, I'm not planning to make a usable application out of this project nor to document it.

Some ideas of the adc_test project are used in [RedPitayaDAQServer](https://github.com/tknopp/RedPitayaDAQServer).

closed 2017-12-11 by pavel-demin

parveennisha, 2017-12-12:
Hi Pavel, 
Thanks for the quick reply. After going through the link i understood that by using sever.c and client.c i can see the output.
"try to use the adc_test project with one of my Debian based images (LED blinker or Debian with Red Pitaya ecosystem), "
 
I am using ubuntu 16.04 and i am new to using ubuntu can you please tell me how to run in ubuntu and i want to know what Red Pitaya ecosystem do?

   " build boot files (uImage, boot.bin and devicetree.dtb) using my scripts (make NAME=adc_test all) and try to use these boot files with your preferred SD card image.

BTW. The ideas from [2] are implemented in the following patches:

    u-boot-xlnx-xilinx-v2016.4.patch,
    devicetree.patch."
In [2] what the .patch do?

pavel-demin, 2017-12-12:
> can you please tell me how to run in ubuntu

```
cat adc_test.bit > /dev/xdevcfg
gcc -O3 adc-test-server.c -o adc-test-server
./adc-test-server
```
> i want to know what Red Pitaya ecosystem do?

It's all the applications and the web interface provided by Red Pitaya.

parveennisha, 2017-12-12:
[image]
After running that code i am getting this messages. What is the use of client.c

pavel-demin, 2017-12-12:
The screenshot looks OK.

adc-test-server.c runs on Red Pitaya and sends the data to a remote PC.
adc-test-client.c runs on a remote PC and receives the data.

I'd much appreciate if you'd try to figure out this kind of trivialities by yourself.

parveennisha, 2017-12-12:
[image]

Sorry for asking silly questions, i am new to this board and i don't know C programming. After running the code it is not showing anything. 

I was running the client code in my PC i am getting the following error, i think all the packages are interrelated how can i use all the packages?
[image]

pavel-demin, 2017-12-12:
> After running the code it is not showing anything.

It's OK. If you want it to show something, you'll need to add some print statements to the code.

> I was running the client code in my PC i am getting the following error

I run adc-test-client.c on a PC with Windows 7 and I use [win-builds](http://win-builds.org) to compile the code. To compile and run it under Ubuntu Linux, the code should be slightly modified.