pavel-demin/red-pitaya-notes #654: Redpitaya- DAC

Abinayasivam, 2017-12-11:
Hi Pavel,
I am trying to use your ADC and DAC program for my appliction, and I am going to add my counter logic for counting the input digital pulse.

Here why we need ODDR part? what it is doing?
Can I create the design without zynq processor?

Thanks,
Abinaya

pavel-demin, 2017-12-11:
Hi Abinaya,

ODDR is a hardware primitive provided by Xilinx. More information can be found on pages 126-133 in [UG471](https://www.xilinx.com/support/documentation/user_guides/ug471_7Series_SelectIO.pdf).

I use ODDR in [axis_red_pitaya_dac](https://github.com/pavel-demin/red-pitaya-notes/blob/master/cores/axis_red_pitaya_dac_v1_0) to communicate with the [DAC1401D125](http://www.idt.com/document/dst/dac1401d125-datasheet) DAC when it operates in the interleaved mode.

Best regards,

Pavel

closed 2017-12-11 by pavel-demin

Abinayasivam, 2017-12-12:
Hi pavel,

I have used your ADC and DAC logic with my counter logic. I have given the digital pulse as an input from E1 connector(DIO5_P) and one analog input from SMA. I have taken the output from DAC through DSO. but I am not getting anything from DAC. First I  need to ensure that whether the digital input pin is taking my input or not. please refer the bd and help me to figure out this.
[image]

pavel-demin, 2017-12-12:
If I share the source code of my applications for Red Pitaya and answer some questions about them, it doesn't mean that I'm interested in providing a free FPGA consulting service.