pavel-demin/red-pitaya-notes #655: sdr_transceiver_hpsdr question

davidhay45, 2017-12-13:
Can anyone tell me what transmit power is produced by the RedPitaya running sdr_tranceiver_hpsdr and to what extent this is controlled by PowerSDR mRx PS running under Win10.  Someone told me this is 10mW, is this correct?

Thanks

pavel-demin, 2017-12-14:
Depending on settings, the output voltage can be as high as 2.0-2.2 Vpp over a 50 Ohm load. It corresponds to 10-12 mW. The output voltage is controlled by Drive and by Settings->PA Settings->PA Gain.

closed 2017-12-14 by davidhay45

davidhay45, 2017-12-14:
I am considering a RF amp that has an input range of up to 2.5mw.  Would it be sensible to use the settings to limit the power so much or should I use an attenuator?

Thanks

pavel-demin, 2017-12-14:
I heard that the output opamps on the Red Pitaya board performed better when the output voltage was around 1 Vpp. So, I think that it should be OK to limit the output power using the settings in PowerSDR mRX PS.

ka6s, 2017-12-14:
David,

I had luck with the class B amplifier design used in the HPSDR Penelope
transmitter.  It is simple to understand - just Resistor Ratios setting the
gain.  You can find the schematics through the openHPSDR.org website.

Steve