pavel-demin/red-pitaya-notes #657: Alpine SD system - question

davidhay45, 2017-12-15:
Is there a way to use SSH to log on to the RP running the Alpine SD system, ie not using a browser.  The default passwords for root and red-pitaya do not work.

pavel-demin, 2017-12-15:
Last time I tested it was with the latest release of the pre-built SD card image. SSH worked with the root account without any problem.

There is no `red-pitaya` account on the SD card image.

The `root` account is enabled with [this patch](https://github.com/pavel-demin/red-pitaya-notes/blob/master/scripts/alpine.sh#L135) in [alpine.sh](https://github.com/pavel-demin/red-pitaya-notes/blob/master/scripts/alpine.sh) and the password is set to `changeme` on the [next line](https://github.com/pavel-demin/red-pitaya-notes/blob/master/scripts/alpine.sh#L137).

closed 2017-12-15 by davidhay45

davidhay45, 2017-12-16:
Is there a way to make the SD card not read-only?  I want to log on the the RP, modify scripts and run them to check.
Thanks

pavel-demin, 2017-12-16:
> Is there a way to make the SD card not read-only?

Just type `rw` on the command line.

`rw` is an alias for `mount -o rw,remount /media/mmcblk0p1`.

There is also `ro`, an alias for `mount -o ro,remount /media/mmcblk0p1`.

The `rw` and `ro` commands are shown in the last example from the [Useful commands](http://pavel-demin.github.io/red-pitaya-notes/alpine/#useful-commands) section.

davidhay45, 2017-12-16:
I am wondering if the wspr app works in Alpine mode?  I logged onto the RP and ran the wspr transmit-wspr.sh and got the following output, the script is below.

[image]
The sh was unmodified:
<<<
#! /bin/sh

DIR=`readlink -f $0`
DIR=`dirname $DIR`

TRANSMITTER=$DIR/transmit-wspr-message
CONFIG=transmit-wspr-message.cfg

GPIO=$DIR/gpio-output

SLEEP=$DIR/sleep-to-59

date

test $DIR/$CONFIG -ot $CONFIG || cp $DIR/$CONFIG $CONFIG

echo "Sleeping ..."

$SLEEP

sleep 1
$GPIO 1
sleep 1

date
TIMESTAMP=`date --utc +'%y%m%d_%H%M'`

echo "Transmitting ..."

killall -q $TRANSMITTER
$TRANSMITTER $CONFIG
>>>

I think the -a comes from $0 giving -ash on my system.

pavel-demin, 2017-12-16:
> I am wondering if the wspr app works in Alpine mode?

It certainly works with the latest release of the pre-built SD card image.

For more information, please read my notes about the multiband WSPR transceiver at
http://pavel-demin.github.io/red-pitaya-notes/sdr-transceiver-wspr

In the [Software section](http://pavel-demin.github.io/red-pitaya-notes/sdr-transceiver-wspr/#software), you'll find an explanation of how the scripts are running.

The transmit-wspr.sh script is disabled by default and should be enabled by uncommenting the corresponding line in wspr.cron.

The correct command to run the transmit-wspr.sh script manually is
```
./transmit-wspr.sh
```

davidhay45, 2017-12-16:
Sorry, I was using the wrong shell!

pavel-demin, 2017-12-16:
I think that the problem isn't with the shell but with the way you run the script. It should be executed but not sourced.