pavel-demin/red-pitaya-notes #658: WSPR app

davidhay45, 2017-12-15:
I want to modify the wspr app to modify the power used and I am trying to understand the transmit-wspr-message.c code.  It looks like the transmit trigger is done by accessing memory to get the config and then using *fifo to initiate the transmission.  Is there further documentation to help me work out what is happening?

Is opening an "issue" the best way to ask this sort of question?

Thanks very much.

pavel-demin, 2017-12-15:
> I want to modify the wspr app to modify the power used

The TX power is currently hardcoded in the [C code](https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/sdr_transceiver_wspr/app/transmit-wspr-message.c#L106).

> Is there further documentation to help me work out what is happening?

No, just the code. The TX part is very easy. It's mostly just FIFO connected to the frequency configuration port of DDS. The C code calculates frequencies and writes them to FIFO. 

> Is opening an "issue" the best way to ask this sort of question?

Yes.

closed 2017-12-15 by davidhay45

davidhay45, 2017-12-17:
OK, I see it uses memory mapped I/O and the power is set at cfg+36 and 38.  I assume the power level set at 32766 is the maximum at 0x7FFE of a 16 bit value, is that correct?

I plan to add the power control to the transmit-wspr-message.cfg file and make it possible to override it by a command line parameter.  Unless that is planned already.

pavel-demin, 2017-12-17:
> I assume the power level set at 32766 is the maximum at 0x7FFE of a 16 bit value, is that correct?

Yes, it's correct.

> I plan to add the power control to the transmit-wspr-message.cfg file and make it possible to override it by a command line parameter. Unless that is planned already.

The reason why it isn't done yet is because I don't know what should be the units (dBFS, dBm, mW, none) and the range (-40-0, -30-10, 0.001-10, 0.0-1.0) of the parameter's value in the configuration file.

davidhay45, 2017-12-17:
For the moment I will stick to 0-32766 to get it going.

davidhay45, 2017-12-21:
A couple of questions if I may please:

What controls how long wspr transmits for each time?

I want to make the frequency (and power) command line parameters for transmit-wspr-message and have more than one instance scheduled in the wspr.cron, maybe two sets each running every alternate 10 minutes so as to explore two (or more) bands.  Any issues there?

Thanks

David

PS Have a good festive season

pavel-demin, 2018-01-08:
> What controls how long wspr transmits for each time?

Technically, it's the number of symbols/values in the FIFO buffer that controls the length of the transmission.

The number of  symbols per transmission and the duration of the transmission is fixed by the WSPR protocol.

> I want to make the frequency (and power) command line parameters for transmit-wspr-message and have more than one instance scheduled in the wspr.cron, maybe two sets each running every alternate 10 minutes so as to explore two (or more) bands. Any issues there?

I think that the following should work:
```
1-51/10 * * * * ...
3-53/10 * * * * ...
5-55/10 * * * * ...
7-57/10 * * * * ...
9-59/10 * * * * ...
```

DH1KLM, 2018-09-04:
Interesting topic, any progress so far?

I have just dealt with WSPR,
I noticed that the transmission power can not be adjusted.
Similarly, LPF / BPF can not be controlled from WSPR like it is done in HPSDR.

That may not be a problem for someone who only does WSPR,
but if a BPF / LPF and also a PA is integrated in a multimode system, problems arise. 
For example, if WSPR goes in TX, the PA will be switched to TX but the BPF / LPF will remain uncontrolled. In my case, all LPF relays are open and the power is not routed to the responsible LPF. Likewise, the PA is driven at full power from Red Pitaya, which would result in a 100W WSPR signal being sent if LPF were switched.

Would it be possible to expand WSPR with these settings?
For example, in the file "transmit-wspr-message.cfg" the power out from RP and LPF / BPF setting.

// frequency
freq = 10.1402; 
lpf = 1,0,0,1 (30M in my case); 
pout-level = 2dBm (or any other unit)