pavel-demin/red-pitaya-notes #659: VNA application: I don't know how to find the Redpitaya address

MxQL, 2017-12-16:
### Description of the setup:

 * Device: RedPitaya
 * SD card image: red-piraya-alpine-3.6-armhf-20171130
 * Application: VNA
 * Other relevant information:

### Description of the problem: I don't know how to find the IP address of the Red Pitaya board. Before, it was 192.168.0.24, but with the new SD card image, it has changed. I also tried the default address given by your VNA.py software, but it doesn't work.

Could you help? Thanks a lot.

Maxim

### Steps to reproduce the problem:

 1.
 2.
 3.

pavel-demin, 2017-12-16:
I can think of several possible solutions:
 - run `apk -a` from the command line of your computer and look for a line containing the MAC address written on the Ethernet connector of your Red Pitaya board,
 - if your computer has DNS Service Discovery (DNS-SD, Bonjour, etc) installed, then run `ping rp-f0xxxx.local`, where `f0xxxx` are the last 6 characters from the MAC address,
 - connect to your network router and check the list of all the connected devices,
 - connect to the Red Pitaya command line interface via the USB/console port and run `ifconfig`.

You can find more hints in the troubleshooting section of the Red Pitaya documentation:
http://redpitaya.readthedocs.io/en/latest/quickStart/troubleshooting/troubleshooting.html

MxQL, 2017-12-17:
Dear Pavel,
Thank you very much for your advice.
I am using a RedPitaya board purchased in 2015. I have bought 2 boards but only opened one. With the 1st board, I used to type 192.168.0.24 in my Web browser and get the Redpitaya's Web server. This works until April 2017.
I am very happy to find your VNA project. For a long time I have been looking for a real FPGA based VNA project. I often need to measure both amplitude and phase of a signal issued from my sensors.
Yesterday I tried to download your VNA project and install it on a new uSD card. I put the address 192.168.0.24 as before but this doesn't work. I removed the uSD card and replaced it by the old one (which still worked in April), but I couldn't find the Redpitaya's Webserver either. I tried this old uSD card with my 2nd Redpitaya board but this new board couldn't connect.

I also tried the command rp-f0xxxx.local/ unsuccessfully.

This is quite amazing. I thought that the Redpitaya hardware should remain unchanged, but all happen as if something has changed with a new uSD card installed.

The problem is I don't have much knowledge about Ethernet. I will have to learn more about all this, and get back to you later.

Anyway, thank you very much for your great work.

Regards

dl1sdz, 2017-12-17:
Hi WithoutaName,

install nmap and use something like this: nmap -sP 192.168.1.*
You will get a list of devices in your lan. The Red Pitaya with the correct
IP number should be visible, if he is switched on and connected.

73 de Hajo dl1sdz

Gruss
Hajo

---
Cela est bien dit, mais il faut cultiver notre jardin.

http://hajos-kontrapunkte.blogspot.de/

MxQL, 2017-12-17:
Dear Hajo and Pavel,

Thank you so much for the good advice: I tried with nmap and found 192.168.0.20 where the Redpitaya's server is. It works as before.

I am still not be able to use the command of rp-F010FE.local/ as indicated on the stemlab documentation (http://redpitaya.readthedocs.io/en/latest/quickStart/first.html), the computer doesn't find the Redpitaya. (The MAC address of my board is 00:26:32:F0:10:FE). I will keep trying.

I will try the VNA application of Pavel with much more confidence, now.

Maxim.

MxQL, 2017-12-18:
The VNA application works!
Thank you very much, Pavel.

Maxim.

pavel-demin, 2017-12-18:
It's great that VNA finally works you.

For information. If you want to communicate with the VNA server running on Red Pitaya from some other programs, then I'd suggest you to have a look at the [MATLAB example](https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/vna/matlab/vna.m).

closed 2017-12-18 by pavel-demin