pavel-demin/red-pitaya-notes #660: The board works perfectly, what does it measure?

MxQL, 2017-12-23:
Dear Pavel,

I saw the Matlab example. It is very clear, a very good educational tool. However, I would like to know more about the hardware configuration.

1. I can understand that the Redpitaya sends out 4 single precision numbers. I have looked around but not found a detailed hardware description. I suppose the 4 received numbers are the real and imaginary parts of the 2 input channels, but I am not sure. Could you confirm?

I would also like to make some suggestions:

2.  Could you give a more detailed command list of the Redpitaya and the data transfer protocole. 

3. Since there is one output left, could you create the quadrature signal of the sinusoidal exciting output, which has same amplitude but 90°-out of phase? This can help to make an analog modulator, or a rotating magnetic field, etc.

4. Are there any possibilities to have some GPIO or auxiliary DACs?

Thanks a lot

Maxim

pavel-demin, 2018-01-08:
Dear Maxim,

> I suppose the 4 received numbers are the real and imaginary parts of the 2 input channels, but I am not sure. Could you confirm?

Yes, it's correct.

> Could you give a more detailed command list of the Redpitaya and the data transfer protocol.

Normally, all the commands are in the Matlab example.

> Since there is one output left, could you create the quadrature signal of the sinusoidal exciting output, which has same amplitude but 90°-out of phase? This can help to make an analog modulator, or a rotating magnetic field, etc.

In the current configuration the second output (OUT2) is already used. It outputs the same signal as the first output (OUT1).

It's possible to add an AXI4-Stream switch between the DDS and the DAC and to output various combination of the sine and cosine signals generated by the DDS.

> Are there any possibilities to have some GPIO or auxiliary DACs?

The pins DIO0_P - DIO7_P are already configured as output in the VNA configuration.

For the auxiliary DACs, the easiest would be to use a quad I2C DAC (MCP4728 or similar).

Best regards,

Pavel

MxQL, 2018-01-08:
Dear Pavel,

Thank you very much for your explanations. I really appreciate your great work.

Could you explain how to use the DIO0_P - DIO7_P digital outputs? Maybe you have an Matlab example?

Best regards

pavel-demin, 2018-01-09:
> Could you explain how to use the DIO0_P - DIO7_P digital outputs? Maybe you have an Matlab example?

The command that controls the digital outputs is in on the lines 30-32 of the Matlab script:
https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/vna/matlab/vna.m#L30-L32

pavel-demin, 2018-03-05:
> Since there is one output left, could you create the quadrature signal of the sinusoidal exciting output, which has same amplitude but 90°-out of phase? This can help to make an analog modulator, or a rotating magnetic field, etc.

The code that controls the phase offsets of both outputs is now in the [latest release](https://github.com/pavel-demin/red-pitaya-notes/releases/tag/20180305).

closed 2018-03-06 by pavel-demin