pavel-demin/red-pitaya-notes #662: Questions about SDR transceiver compatible with HPSDR

davidhay45, 2018-01-09:
Using SD card Alpine 3.6 and PowerSDR mRX v3.4.7.

1) The gpio pins D102_P and D103_P (attenuator 20/10db) do not appear to be used.  The preamp control pin D101_P is set only when the PowerSDR ATT is set to 0 db and not set when ATT is 10/20/30 db.  Is this correct and does it mean that D102_P and D103_P are available for other uses?

2) The tx level has a range of 0-32000 but the max value when Drive=100 is 25688 and appears to be controlled by PowerSDR.  Is this correct?

3) When VAC1 or VAC2 is enabled or disabled the Hermes control pins momentarily select another band and then select the correct band again.  For example, if the band is set to 80m and the Hermes ctrl pins are 0010 and VAC1 is enabled or disabled then the Hermes cntrl pins change to 0100 and then back to 0010. Similarly when the band selected is 20m the Hermes cntrl pins are 0101 but change momentarily to 1010 and then back to 0101.  This looks like a bug or perhaps a benign feature, if not what is is happening?

Many thanks.

David

pavel-demin, 2018-01-09:
> The gpio pins D102_P and D103_P (attenuator 20/10db) do not appear to be used.

As I recall, these two pins work when PowerSDR mRX PS is in the HPSDR mode.

> The tx level has a range of 0-32000 but the max value when Drive=100 is 25688 and appears to be controlled by PowerSDR. Is this correct?

The tx level depends on Drive and PA Gain. The max value is 32110.

> When VAC1 or VAC2 is enabled or disabled the Hermes control pins momentarily select another band and then select the correct band again.

Looks like a bug in PowerSDR mRX PS.

Rolf-L, 2018-01-09:
I'm using Alpine3.6 + PowerSDR mRX V3.4.7 in Hermes-Mode + Alex on.
All Pins + Drive + VAC1+2 are working fine. 
You have to set Receive Pins and Transmit Pins in Hermes Crl to correct values depending on your ext. filter or antenna controller.

davidhay45, 2018-01-09:
That is not the point, mine is also working correctly, I know about the Hermes setup.

closed 2018-01-09 by davidhay45