pavel-demin/red-pitaya-notes #663: SDR Receiver - Red Pitaya unable to talk with HDSDR / SDR#

samlo20, 2018-01-11:
Hello Pavel,

I am currently setting up an SDR Receiver with a Windows image file copied on the SD Card, as followed by this link: 
http://pavel-demin.github.io/red-pitaya-notes/sdr-receiver/

I have also installed the required software including the DLL extension for the Red Pitaya interface, however, the SDR# and HDSDR both cannot receive signals from the Red Pitaya, is there any method to resolve this? 

Regards,
Sam

pavel-demin, 2018-01-11:
Hello Sam,

I'd suggest you to try the  SDR transceiver:

http://pavel-demin.github.io/red-pitaya-notes/sdr-transceiver/

It's a more up-to-date application with the same functionality as the SDR receiver and more.

Anyway, to debug this kind of problems, I'd suggest the following steps:
- connect to the Red Pitaya USB/serial console as described at [this link](http://redpitaya.readthedocs.io/en/latest/developerGuide/os/console/console.html),
- check if Red Pitaya boots,
- check the IP address (for example, using `ifconfig` command),
- check the network connection between Red Pitaya and PC.

If it doesn't help, then please provide more information about your setup.

Best regards,

Pavel

samlo20, 2018-01-12:
Hello Pavel,

I have connected the Red Pitaya into the PC using the USB console, and it
automatically boots.
However, I have tried to use the automatic IP address, as well as changing
to other IP addresses, but it doesn't work.
Is there any method to resolve this?
Regards,
Sam

pavel-demin, 2018-01-12:
> However, I have tried to use the automatic IP address, as well as changing to other IP addresses, but it doesn't work.

Could you please type the following command on the Red Pitaya command line and post its output?
```
ifconfig
```

samlo20, 2018-01-12:
where can I find the command line utilities from the my red pitaya home
page (the page after I typed the red pitaya address, like rp-f05cb4.local /
)?

regards
Sam

pavel-demin, 2018-01-12:
> I have connected the Red Pitaya into the PC using the USB console, and it automatically boots.

> where can I find the command line utilities from the my red pitaya home page (the page after I typed the red pitaya address, like rp-f05cb4.local / )?

I'm afraid I don't have enough information to understand what is your problem. Please provide more details.

From what you write, I understand that you have access to the Red Pitaya USB/console command line.

The command line utilities should be used from the Red Pitaya USB/console command line.

samlo20, 2018-01-12:
Hi Pavel,

This is the output status after I connect the Red Pitaya with using the arp
-a command:

rp-f05cb4.local (169.254.115.191) at 0:26:32:f0:5c:b4 on en0 [ethernet]
? (169.254.255.255) at (incomplete) on en1 [ethernet]
? (224.0.0.251) at 1:0:5e:0:0:fb on en1 ifscope permanent [ethernet]
? (239.255.255.250) at 1:0:5e:7f:ff:fa on en1 ifscope permanent [ethernet]

This is the relevant section after using the ifconfig command:
en0: flags=8863<UP,BROADCAST,SMART,RUNNING,SIMPLEX,MULTICAST> mtu 1500
    options=10b<RXCSUM,TXCSUM,VLAN_HWTAGGING,AV>
    ether 3c:07:54:1a:ef:e3
    inet6 fe80::1c12:1100:4c99:95be%en0 prefixlen 64 secured scopeid 0x4
    inet 169.254.202.28 netmask 0xffff0000 broadcast 169.254.255.255
    nd6 options=201<PERFORMNUD,DAD>
    media: autoselect (1000baseT <full-duplex,flow-control>)
    status: active
en1: flags=8863<UP,BROADCAST,SMART,RUNNING,SIMPLEX,MULTICAST> mtu 1500
    ether 28:cf:da:d7:52:18
    inet6 fe80::107e:a1c3:b6cd:edd5%en1 prefixlen 64 secured scopeid 0x5
    inet 138.38.174.3 netmask 0xffffe000 broadcast 138.38.191.255
    nd6 options=201<PERFORMNUD,DAD>
    media: autoselect
    status: active

Please check if there are any issues with the output

Regards,
Sam

pavel-demin, 2018-01-12:
I'd say that the IP address of your Red Pitaya board is 169.254.115.191. And it's the IP address that should be used in the HDSDR/SDR#.

It's strange that ifconfig shows two interfaces. Could you please confirm that you run ifconfig on Red Pitaya?

samlo20, 2018-01-12:
I ran it on the PC terminal, and the en0 link is connected to the red
pitaya with the mac address ending with f05cb4

pavel-demin, 2018-01-12:
> I ran it on the PC terminal, and the en0 link is connected to the red pitaya with the mac address ending with f05cb4

Thanks for checking.

Have you tried to run HDSDR or SDR# setting the IP address to 169.254.115.191 as shown by the arp -a command?

samlo20, 2018-01-12:
I did, but no signal has appeared on the software

pavel-demin, 2018-01-12:
> I did, but no signal has appeared on the software

Do you have any error messages? If not please post a screenshot of the software.

samlo20, 2018-01-12:
Here is the screenshot, as there is no signal coming in, although it has
the correct IP address​
 20180112_213111.jpg
<https://drive.google.com/file/d/0B40Gytb_QnJVZ2poZFpINU1pTUFCOU82STYybkZqMmsyYXdZ/view?usp=drive_web>
​

samlo20, 2018-01-12:
Hello Pavel,

Here is the screenshot I took after connecting to the Red Pitaya with the
correct IP address

Regards,

Sam

pavel-demin, 2018-01-13:
Hello Sam,

Thanks for the screenshot.

It looks like the signal level shown by the level meter is lower than -130 dBm. 

Maybe the signal is just too low to be seen on the spectrum display.

Could you try to decrease the lower limit of the y scale on the spectrum display?

Best regards,

Pavel

samlo20, 2018-01-16:
Hello I have lowered down to -160 dBm in HDSDR but there is no results seen
on the display

pavel-demin, 2018-01-23:
I'm closing this issue because apart from the steps listed in my first comment I can't think of any other solution.

closed 2018-01-23 by pavel-demin

samlo20, 2018-01-23:
Don't worry I managed to connect it now