pavel-demin/red-pitaya-notes #664: hpf problem ?

F6ITU, 2018-01-15:
### Description of the setup:
bad hpf switching behaviour
  * Device : spi interface (https://github.com/F6ITU/Alexandrie)
 * SD card image: don't remember (but not the last one)
 * Application: PowerSDR mRX ver 3.4.2
 * Other relevant information: none

### Description of the problem:
It seems that the lpf spi commands are incorrectly handled for the 9.5, 13 and 20 MHz filter commands.
All other switching are perfectly decoded by the TPIC6B595 and working (Att 1 & 2, by-pass, 6m_LNA/50 MHz, 1.5 MHz and 6.5 MHz)

Once the 6.5 MHz limit is passed, the 6.5 hpf filter stays on and is not released when passing the next crossover point (theorically 9.5 MHz) 
... and thus till the frequency reachs 50 MHz. 

### Steps to reproduce the problem:

 1. plug an spi interface into E1 of the Red Pitaya
 2.Power-up, launch PowerSDR with "Alex" filter set declared in the general configuration screen, Alex filter configured in "firmware" mode
 3. sweep frequencies from 0 to 60 MHz

I built 2 different interfaces just in case the problem could be caused by a faulty TPIC6B595. Same mis-functionning 

... it could also be a PEBCAK question, but I couldn't find where I made a mistake.

Marc

pavel-demin, 2018-01-19:
Are you using  the firmware or manual Alex configuration (Setup -> General -> Alex -> HPF/LPF)?

pavel-demin, 2018-01-19:
Could you please check if it works better in the manual mode?

I've checked the code but can't see any problem.

Could it be due to the following line in  (sdr-transceiver-hpsdr.c)[https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/sdr_transceiver_hpsdr/server/sdr-transceiver-hpsdr.c]?
```
freq = freq_data[1] < freq_data[2] ? freq_data[1] : freq_data[2];
```
The minimal frequency of the the two receiver frequencies is used for the high-pass filter. This is how the Hermes firmware behaves (look for C122_freq_min in [Hermes.v](http://svn.tapr.org/repos_sdr_hpsdr/trunk/Hermes/Source/Hermes_v3.1/Hermes.v)).

F6ITU, 2018-01-19:
Thks Pavel
I'm QRM for the next 7 days.. I'll check this (hard and soft ) asap and will keep you informed.

As far as I can remembrer, the "manual mode" was not responsive, but I didn't pay attention to the general configuration of PowerSDR (considering that the default config was "firmware")

Sorry for the delay
73'
Marc

closed 2018-02-13 by pavel-demin