pavel-demin/red-pitaya-notes #665: SDR UDPdata timeout issue

bradman77, 2018-01-16:
### Description of the setup:

 * Device:  125-14 Board
 * SD card image:  Ver 098 build 617
 * Application:SDR Receiver compatible with HPSDR  (1.0-20171130)
 * Other relevant information:
Repeated skimmer server error: SDR Failure: Timed out Waiting for UDP data.
Hermes log shows:
"2018-01-15 21:23:20.765: StartRx
2018-01-15 21:23:20.765: Worker thread running
2018-01-15 21:23:20.776: SetRxFrequency Rx# 0 Frequency: 1845500
2018-01-15 21:23:20.786: SetRxFrequency Rx# 1 Frequency: 3545500
2018-01-15 21:23:20.796: SetRxFrequency Rx# 2 Frequency: 7045500
2018-01-15 21:23:20.807: SetRxFrequency Rx# 3 Frequency: 10145500
2018-01-15 21:23:20.817: SetRxFrequency Rx# 4 Frequency: 14045500
2018-01-15 21:23:20.827: SetRxFrequency Rx# 5 Frequency: 18113500
2018-01-15 21:23:27.111: Timed out waiting for UDP data"

### Description of the problem:
Repeated error, does not resolve.  Please help.

### Steps to reproduce the problem:

 1. reboot, same issue
 2. uninstall SDR Receiver compatible with HPSDR  (1.0-20171130), reinstall, same issue
 3.  reinstalled hermesintf.dll, same issue.

pavel-demin, 2018-01-16:
Could you please check/confirm the following?
- SDR Receiver compatible with HPSDR is started via the Red Pitaya web interafce.
- The browser window showing SDR Receiver compatible with HPSDR remains open.

bradman77, 2018-01-16:
Yes to both questions.

pavel-demin, 2018-01-16:
Thanks for checking.

Two more questions.

Do you have the latest hermesintf.dll from the following link?

https://github.com/k3it/HermesIntf/releases

In the Skimmer Server window, what is the receiver name (Unknown or Red Pitaya) shown in the Skimmer tab?

bradman77, 2018-01-16:
hermesintf.dll is current
yes, Skimmer server shows RP- MAC address

I will also note that this problem started to occur when putting a second
RP online on the same network.  I have since coupled the hermesintf.dll
specifically to the IP address of the RP, however, the UDP error is
persistent.  The second RP was removed from the network until this issue
can get resolved so as to not confuse network packets.

One other observation I see frequently with the RP is that it gets confused
about the status of the RP being online.  For example, if trying to check
for a OS update the error persists that I need to put the RP online, not
sure if this could also be related.

Thanks for the help...

pavel-demin, 2018-01-17:
> I will also note that this problem started to occur when putting a second RP online on the same network. I have since coupled the hermesintf.dll specifically to the IP address of the RP, however, the UDP error is persistent.

I think that adding the last two bytes of the MAC address to HermesIntf.dll (as explained in [README](https://github.com/k3it/HermesIntf/blob/master/README.md)) is a better solution.

closed 2018-01-23 by pavel-demin

bradman77, 2018-01-23:
For any benefit to others the only fix that worked was a complete
reimagining of the SD card.