pavel-demin/red-pitaya-notes #666: HPSDR Transceiver 2nd RX question

davidhay45, 2018-01-18:
### Description of the setup:

 * Device:
 * SD card image:Alpine 3.6
 * Application:sdr_transceiver_hpsdr
 * Other relevant information: OpenHPSDR mRX PS v3.4.7 on Widows 10

### Description of the problem:
Gain of the second receiver lower by 15 dbm.

Have I got something wrongly configured?

[image]

pavel-demin, 2018-01-18:
Do you have the same signal connected to both inputs (IN1 and IN2) of the Red Pitaya board?

It would be helpful to see a screenshot showing the spectra from both receivers.

davidhay45, 2018-01-18:
I removed and replaced the jumpers.  The signal levels are shown as the same (-89dbm) but the noise levels are about 10dBm different.  Is this "just" a hardware problem with my IN2?

[image]

pavel-demin, 2018-01-19:
I don't see the 10dBm difference on you screenshot.

davidhay45, 2018-01-22:
I am away but will post clearer screenshots on my return.

davidhay45, 2018-01-23:
What appears to be happening is that the noise floor of the second receiver is 5db lower than the first receiver.  In this case I am running sdr_receiver_hpsdr with RX1 and RX2 using IN1.
 The cursor was on the red line in each case.

[image]
[image]

pavel-demin, 2018-01-23:
Thanks for the new screenshots. Now, I see the difference.

However, both RX meters show exactly the same values and the RX1 spectrum is smoother than the RX2 spectrum. So,  I'd say that the display settings are different for RX1 and RX2 and that's why the spectrum levels are shown differently.

Please check the display settings (Setup->Display->RX1 and Setup->Display->RX2).

davidhay45, 2018-01-23:
There were indeed different and when made the same the 5db level difference disappears.  Why were they different I do not know, I certainly have not modified them.

Anyway, problem solved, thanks.

closed 2018-01-23 by davidhay45