pavel-demin/red-pitaya-notes #667: Wifi client mode problem

tok2010, 2018-01-22:
Hello Pavel,

I encountered the following problem;

 * Device: STEMlab Red Pitaya 125-14 starter kit, wifi USB dongle Realtek RTL8188CUS
 * SD card image: red-pitaya-alpine-3.6-armhf-20171130
 * Application: Wifi Client mode
 * Other relevant information:
With redpitaya_ubuntu_17-11-35_01-sep-2017, wifi client mode is working, but not Access Point mode because of USB dongle chipset according Red Pitaya documentation

### Description of the problem:
I would like to run wifi client mode with redpitaya alpine, since usb console is not convenient and that eth0 is used for powersdr.

### Steps to reproduce the problem:

 1. set a wifi usb dongle, China made Realtek RTL8188CUS I bought on ebay
 2. run redpitaya alpine
 3. according the procedure described in Useful commands of Alpine with pre-built applications of Red Pitaya Notes
4. no ip address assigned for Red Pitaya USB dongle

lsusb shows:
Bus 001 Device 002: ID 0bda:8176
Bus 001 Device 001: ID 1d6b:0002

dmesg | grep wlan0
IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready

ifconfig shows:
ifconfig
eth0      Link encap:Ethernet  HWaddr 00:26:32:F0:5A:66
          inet addr:192.168.1.100  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: fe80::226:32ff:fef0:5a66/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1261891 errors:0 dropped:0 overruns:0 frame:0
          TX packets:8831615 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:1337293197 (1.2 GiB)  TX bytes:894826066 (853.3 MiB)
          Interrupt:148 Base address:0xb000

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:76 errors:0 dropped:0 overruns:0 frame:0
          TX packets:76 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:6312 (6.1 KiB)  TX bytes:6312 (6.1 KiB)

wlan0     Link encap:Ethernet  HWaddr 00:13:EF:80:5B:90
          inet6 addr: fe80::213:efff:fe80:5b90/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:837 errors:0 dropped:3796 overruns:0 frame:0
          TX packets:201 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:213331 (208.3 KiB)  TX bytes:53797 (52.5 KiB)

Powersdr works fine with eth0.
is My USB wifi dongle not compatible with Redpitaya Alpine? However this dongle works fine with Redpitaya ubuntu.

Regards,
Tokio

pavel-demin, 2018-01-23:
Hello Tokio,

Since ifconfig shows wlan0, the  USB WiFi dongle is compatible with the SD card image based on Alpine Linux.

I'd suggest to check that wpa_supplicant was configured with the correct SSID and passphrase parameters corresponding to your WiFi network.

Best regards,

Pavel

tok2010, 2018-01-23:
Hello Pavel,

Thank you very much for your prompt reply.
Your suggestion has enlightened me on wifi connection. My wpa_supplicant is quite correct, however the access point itself has not been working properly. I never imagined that.

I have two wifi routers of the manufacturer NEC which are very commonly used in Japan. One is attached to the broadband modem, and functions as router too. This wifi router is connected to the other that functions in bridge mode. Red Pitaya was using the latter wifi "bridge" access point. I couldn't explain why this access point has not been working, but Red Pitaya has made this access point of wifi "bridge" idle. After the shutdown of Red Pitaya, I re-start the wifi "bridge", then the access point is brought active again until Red Pitaya makes an access to this. Therefore, the solution for me is to change the access point described in wpa_supplicant of Red Pitaya to another access point of the wifi router.

Now all of them work fine. I use SSH to connect Red Pitaya via Wifi, and PowerSDR is working very well.

Red Pitaya wifi may be not compatible with my wifi "bridge".

Best regards,
Tokio

pavel-demin, 2018-01-23:
I'm glad that you have found a solution that works for you. I'm closing this issue.

closed 2018-01-23 by pavel-demin

pavel-demin, 2018-05-01:
Hello Tokio,

I'm afraid that I don't understand your question.  What do you mean by "subject input on E1"?

Best regards,

Pavel

tok2010, 2018-05-01:
Hello Pavel,

I'm now implementing an extension board for Red Pitaya for the purpose of
HPSDR.
According to the document, "GPS PPS, In" is assigned on the pin E1 DIO3_N.
I would lik to know this "GPS PPS, In" functionality.

Best regards,

Tokio

pavel-demin, 2018-05-01:
The GPS PPS input is used in the multiband WSPR transceiver for the frequency and time calibration. In the SDR transceiver compatible with HPSDR, the GPS PPS input isn't used.

tok2010, 2018-05-01:
Hello Pavel,

Thank you very much for your kind explanation.

73,

Tokio