pavel-demin/red-pitaya-notes #670: Creating a Linux bootable image for Red Pitaya

carlos-e-teixeira, 2018-02-09:
Hi Mr. Pavel!

I'm trying to create my own Linux boot image for my Red Pitaya.

I've already created my hardware platform on Vivado. The corresponding bitstream was generated successfully and I've already exported the hardware to SDK. On SDK, I also created the FSBL ('fsbl.elf').

From now on, I don't know exactly what I have to do. I'm still learning about Linux environment.

If I'm correct, I still need to create the BSP and configure it in order to have the 'U-boot.elf'. In addition, I also need the device tree ('.dtb' file) corresponding to the Red Pitaya. Finally, I need to have 'uImage.bin' and 'uRamDisk.image.gz' files. After having all these files, I think I can create my boot image (.bin) using Bootgen and put it on my SD card.
Is all this correct?

If so, where can I find all these remaining files? Or how can I create them?

I will appreciate any help and/or comments!
Thank you in advance!
Best regards!

pavel-demin, 2018-02-09:
My scripts that build SD card images are in the [scripts](https://github.com/pavel-demin/red-pitaya-notes/tree/master/scripts) directory.

Some more details can be found in the [LED blinker](http://pavel-demin.github.io/red-pitaya-notes/led-blinker) notes.

closed 2018-02-09 by pavel-demin

carlos-e-teixeira, 2018-02-09:
Thank you for the answer, Mr. Pavel.
I found that so complex, but I will try to understand it.
At least, is it correct, the process I described?

pavel-demin, 2018-02-09:
> At least, is it correct, the process I described?

I'm not sure if I understand your description. You can find my description of the process on the page 11 of [this PDF file](https://www.dropbox.com/sh/5fy49wae6xwxa8a/AACl--BhQvcNgjeQLRaiX9dha/ClubVivado2016_Pavel_Demin.pdf?dl=1) and compare it with yours.

carlos-e-teixeira, 2018-02-20:
No problem!
This document helps me, thank you!