pavel-demin/red-pitaya-notes #673: Is the QUISK 4.1.12 "Bscope" function working with the red-pitaya HPSDR compatible TXV  ?

satfan52, 2018-02-11:
Hello,

I am running the latest version of the HPSDR compatible SDR transceiver as well as QUISK 4.1.12 under ubuntu 16.04. I have defined both "red pitaya" and "hermes" radios in QUISK. The issue I am reporting here applies to two types of radios.

Everything works fine with QUISK except the "Bscope" screen/function. When selecting the "Bscope" screen/function, I am able to tune anywhere in the  baseband of the RP (so between 0 and 60 Mhz)  by clicking on the frequency scale. That works ok, thanks!   However, there is no waterfall and no signal displayed in reception mode to show the entire HF spectrum between 0 and 60 Mhz (roughly)!!! 

Not sure it is a QUISK software issue or an issue with your HPSDR compatoble sofwtare, or even with my own environment. 

However since this Bscope feature works ok with Hermes Light II, and since everything else works ok in my environment, my logical deduction is that it there is  maybe  (?) a compatibility issue regarding the emulation of the Hermes Light 2 protocol for this specific Bscope function.

It would be very useful to have this feature available, but again, not sure where the fix must be done, in QUISK, the HPSDR compliant software, or my own environment

Thanks
Regards
73
Peter/ON7YI

pavel-demin, 2018-02-11:
Hello Peter,

>  However, there is no waterfall and no signal displayed in reception mode to show the entire HF spectrum between 0 and 60 Mhz (roughly)!!!

I think that the Bscope function in Quisk depends on the raw ADC data transmitted via EP4 (page 4 of the HPSDR protocol description). This part of the protocol isn't implemented in my software. Implementing it would require the sample rate conversion from 125 to 122.88 MSPS (down 3125 and up 3072).

Best regards,

Pavel

closed 2018-02-13 by pavel-demin

satfan52, 2019-11-03:
1) Quisk 4.1.45 introduces a new way of dealing with the so called "bandscope functionality" 
_

> Quisk Version 4.1.45  September 2019
> ====================================
> I added a config screen hardware option "Hermes known IP". If you know the IP address of the Hermes
> hardware, enter it here. Otherwise Quisk will search for the hardware using the usual broadcast method.
> 
> **The Python Rx samples interface can now accept bandscope data, that is, raw samples from the ADC. See
> the file quisk_hardware_model.py for documentation. I fixed a bug in the bandscope that added noise**

_

Does this means that the bandscope functionality could now possibly be implemented in QUISK with out any changes to the software of the redpitaya or are some changes still required ?  Thanks for clarifying.

2) See also SparkSDR - known issues with the redpitaya

http://www.ihopper.org/radio/sparksdr2.htm

> Known Issues

FT8/JT9 conversation view disabled.

Slow initial display switching receivers.

The 32 bit version has had less testing and generally has lower performance and slower startup.

The red pitya running Parvel's 8 channel hpsdr receiver code does not provide the bandscope data so the rf agc and adc level meter don't work.

Filter switching is untested on anything other than Hermes 

> Lite2

satfan52, 2019-11-03:
1) Quisk 4.1.45 introduces a new way of dealing with the so called "bandscope functionality" 
_

> Quisk Version 4.1.45  September 2019
> ====================================
> I added a config screen hardware option "Hermes known IP". If you know the IP address of the Hermes
> hardware, enter it here. [...]

_

2) See also SparkSDR - known issues with the redpitaya

http://www.ihopper.org/radio/sparksdr2.htm

_```
> Known Issues

FT8/JT9 conversation view disabled.

Slow initial display switching receivers.

The 32 bit version has had less testing and generally has lower performance and slower startup.

**The red pitya running Parvel's 8 channel hpsdr receiver code does not provide the bandscope data so the rf agc and adc level meter don't work.**

Filter switching is untested on anything other than Hermes 

> Lite2
```_