pavel-demin/red-pitaya-notes #675: XADC Wizard - Reading analog inputs

carlos-e-teixeira, 2018-02-20:
Hi Mr. Pavel,
I'm trying now to read some analog values using the IP XADC Wizard.
For hardware configuration, I used the TCL code you implemented.

```
cell xilinx.com:ip:xadc_wiz:3.3 xadc_0 {
  DCLK_FREQUENCY 143
  ADC_CONVERSION_RATE 100
  XADC_STARUP_SELECTION independent_adc
  CHANNEL_ENABLE_VAUXP0_VAUXN0 true
  CHANNEL_ENABLE_VAUXP1_VAUXN1 true
  CHANNEL_ENABLE_VAUXP8_VAUXN8 true
  CHANNEL_ENABLE_VAUXP9_VAUXN9 true
  CHANNEL_ENABLE_VP_VN true
} {
  Vp_Vn Vp_Vn
  Vaux0 Vaux0
  Vaux1 Vaux1
  Vaux8 Vaux8
  Vaux9 Vaux9
}

apply_bd_automation -rule xilinx.com:bd_rule:axi4 -config {
  Master /ps_0/M_AXI_GP0
  Clk Auto
} [get_bd_intf_pins xadc_0/s_axi_lite]

set_property RANGE 64K [get_bd_addr_segs ps_0/Data/SEG_xadc_0_Reg]
set_property OFFSET 0x40020000 [get_bd_addr_segs ps_0/Data/SEG_xadc_0_Reg]
```

For my C application, I declared volatile a pointer and the mapping is also according to your code ('sdr-transceiver-hpsdr.c').

```
volatile int32_t *xadc;
xadc = mmap(NULL, 16*sysconf(_SC_PAGESIZE), PROT_READ|PROT_WRITE, MAP_SHARED, fd, 0x40020000);
```

From now on, I'm not sure about the next steps (pieces of code) to be implemented in order to get 4 analog values corresponding to the 4 slow analog inputs available on the connector E2 of the Red Pitaya.

I did not understand what you've implemented. I know the XADC block only converts analog values to digital vectors of 12 bits, doesn't it? 

Also, I know the XADC block accepts bipolar analog voltages (-0.5V to +0.5V) and unipolar analog voltages (0V to +1V). Since I'm using the temperature sensor LM35, I think I will not have any problem to read it.

I appreciate again in advance your help!
Thank you so much!
Best regards!

pavel-demin, 2018-02-20:
Please, don't call me Mr Pavel.

The only program where I'm using the XADC block is [sdr-transceiver-hpsdr.c](https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/sdr_transceiver_hpsdr/server/sdr-transceiver-hpsdr.c#L1249-L1264). I'm reading the data with the following commands:

```
int32_t value;
value = xadc[144] >> 3;
value = xadc[145] >> 3;
value = xadc[152] >> 3;
value = xadc[153] >> 3;
```

The addresses are from the [XADC Wizard documentation](https://www.xilinx.com/products/intellectual-property/xadc-wizard.html#documentation) ([PG091](https://www.xilinx.com/support/documentation/ip_documentation/xadc_wiz/v3_3/pg091-xadc-wiz.pdf), page 18).

closed 2018-02-20 by pavel-demin

carlos-e-teixeira, 2018-02-21:
Sorry Pavel, I was trying to be courteous.

So... I totally understand now, but just one more question.
Why do you use a right-shift operation of 3 positions? Would not it be a right-shift operation of 4 positions, since we have 16-bits registers that keep the 12-bit MSB justified result of A/D conversion on the corresponding auxiliary analog input.

Thank you again, Pavel! You help me a lot!
Best regards!

pavel-demin, 2018-02-21:
> Why do you use a right-shift operation of 3 positions? Would not it be a right-shift operation of 4 positions, since we have 16-bits registers that keep the 12-bit MSB justified result of A/D conversion on the corresponding auxiliary analog input.

You're right. The right-shift by 4 is the most correct way to process the XADC data. 

However, in my program I was looking for a way to obtain 12-bit values for the unipolar analog voltages and the right-shift by 3 worked well enough.