pavel-demin/red-pitaya-notes #677: DAC problem when restart Red Pitaya in dac_test project

jonathanepn, 2018-02-22:
Dear Pavel,

Thanks for taking the time to open this issue! 

I'm testing dac_test project on Debian 8.6 when a restart Red Pitaya:

1) I can see a voltage level on the DAC output (using an oscilloscope). This voltage level was setting with the last test before turn off.
2) This voltage level return to cero when DAC was reseting with a new value.
3) If Red Pitaya is reprogramming with another .bit which doesn't use DAC, this voltage level persisting. For example blinker_led.
4) Also I modificated rc.local file for start up with dac_test project but the default program (blinker_led) is runing before dac_test and the voltage level is setting for a few seconds again.

Could you give some suggestion for resolved this problem.

Specifically, I want start up Red Pitaya with dac_test project where DAC output will be reseting (voltage level with 0V).

I apologize for my bad english.

pavel-demin, 2018-02-22:
I think that the easiest solution to set the DAC outputs to zero would be to load a .bit file of a project containing the zeroer module connected to the DAC interface module. For example, vna.bit.

pavel-demin, 2018-02-22:
> I want running the dac_test project instead blinker_led 

Just build boot.bin for dac_test and copy it to the SD card. Here is the command to build boot.bin:
```
make NAME=dac_test boot.bin
```

closed 2018-02-27 by pavel-demin

jonathanepn, 2018-02-27:
Dear Pavel.
I was testing boot.bin with vna and dac_test projects. I have the same problem when I reset the red pitaya I see a voltage level on the DAC output (using an oscilloscope) . After few seconds the voltage returns to zero.
I don't know why that value is keeping after reset. Can you help me please.

pavel-demin, 2018-02-27:
I'm afraid that the information that you provide isn't enough for me to understand what exactly problem you're trying to solve and what kind of help you expect from me.

So far, you asked me two questions. Here is how I understood and answered them:

1. Your question in my understanding: How to set the DAC outputs to zero?
My answer:  Load a .bit file of a project containing the axis_zeroer module connected to the DAC interface module. For example, vna.bit.

2. Your question in my understanding: How to change the Debian SD card image to start dac_test instead of led_blinker?
My answer: Just build boot.bin for dac_test and copy it to the SD card.

Did I correctly understand your questions? Did my answers help you?

jonathanepn, 2018-02-27:
I am grateful for the interest and for wanting to solve my question.
The information that you have given me is correct and very helpful. Yes you have solved the questions but it seems to me that I have not been able to explain the main problem.
I will try to explain it again.
1.I am using a modification of the dac_test project, in which I enter a desired value from the terminal. For example, in the terminal I enter a value of 0.5 V and obtain a voltage level of 0.5 V measured with an oscilloscope at the dac output. The program is working correctly.
2. I turn off Red pitaya
3. I turn on Red Pitaya again, and in the oscilloscope the same voltage level configured previously is observed without executing the dac_test.bit
4. I run dac_test.bit and in this moment the value of the output on the dac returns to zero.

To solve the problem I have done the following, without results.

1. Modify the rc.local file for starting the dac_test.bit in the Red Pitaya boot, the result was the value of 0.5 V in the output of the dac was maintained for a few seconds and then returned to zero.
2. Generate the boot.bin with the vna project and I repeat the procedure with the dac_test.bit, in this case when restarting the Red Pitaya, the problem remains but the output of the dac returns to zero in a shorter time.

I think that the dac has a memory where the configured value is stored, the memory is not erased when restarting Red Pitaya.

Perform the same test using the Signal Generator application of the Red Pitaya ecosystem. In this case, this value is not maintained and works perfectly when I restart Red Pitaya. I would like to obtain the same result with my modificated program.

pavel-demin, 2018-02-27:
> 2. I turn off Red pitaya
> 3. I turn on Red Pitaya again

When you turn off Red Pitaya, do you disconnect it from the power supply? For how long?

jonathanepn, 2018-02-27:
Hi Pavel.
Yes I disconnect it from the power supply.
I did some test.
1. I connect and disconnect immediately, with the same result.
2. Disconnected and wait 10 seconds then connect, with the same result
3. I did the same procedure above with 30s, 60s and got the same result.
4. Perform a test with 20 minutes and this time the voltage level was negative -1V and after a few seconds return to zero.

Do you have any idea why this happens?

Thanks again

pavel-demin, 2018-02-28:
> I did some test.

Thanks for the tests.

> Do you have any idea why this happens?

I'd say that adding a resistor between the output and GND would discharge the output much faster. As far as I know the Red Pitaya output requires a 50 Ohm load. So, I'd suggest to try adding a 50 Ohm terminating resistor to you setup.

jonathanepn, 2018-02-28:
I have been using a 50 Ohms terminal.
[image]

pavel-demin, 2018-03-05:
I've just checked how the Red Pitaya outputs behave when when the board is switched off and on.

1. When the board is switched off, the outputs return to 0V within a few milliseconds:
[image]

2. When the board is switched on with the led_blinker boot.bin, the outputs have some negative voltages:
[image]

3. When the board is switched on with the vna boot.bin, the outputs return to 0V after about 370-380 milliseconds:
[image]

jonathanepn, 2018-03-09:
Hi Pavel,
Sorry for not answering before. I did not see your message.

It's exactly the problem I was telling you, I do not understand why there is such a voltage level. Also if before turning off Red Pitaya there is a voltage value set at the DAC output, this level reappears when it is turned on again.

I am worried because if we worked with the output of the DAC connected to a device very sensitive to the voltage change, this sudden change in the output of the DAC could cause a problem in the device.

1. Do you have any idea why this happens?

2. If the Red Pitaya ecosystem is tested, this problem does not exist. What would be the difference between the two configurations?

pavel-demin, 2018-03-09:
If the 370-380 milliseconds of unpredictable voltage is a problem for your application, then I'd suggest to add a couple of relays between the Red Pitaya outputs and your sensitive device.

In my understanding, these 370-380 milliseconds is the time required for the ZYNQ chip to start the CPU, to read the FPGA configuration from the micro SD card and to configure the FPGA. Once the FPGA is configured, it can set the DAC outputs to 0V.

This behavior should be the same for the SD card images provided by the Red Pitaya developers.

The picture at [this link](https://i.imgur.com/Dn4jaSa.jpg) clearly shows all the pins of the DAC chip on the Red Pitaya board. The PWD pin is connected to AGND. So, the FPGA doesn't control when the DAC chip is in the active mode and the DAC chip enters the active mode right after you connect the board to a power supply.

> Also if before turning off Red Pitaya there is a voltage value set at the DAC output, this level reappears when it is turned on again.

I can't reproduce this behavior with my board. When I do my tests, I disconnect the Red Pitaya board from the power supply.

As far as I know, the DAC chip doesn't have any non-volatile memory. Once all the capacitors are discharged it looses all the information about its previous state. Is it possible that your setup has a large capacitor that remains charged for a long time?

jonathanepn, 2018-03-09:
In fact my tests were done with an ATX power supply. I think that is not the cause, the problem is the initialization of the DAC as you mention it.

In view of the fact that the problem is in the Hardware I have already tried an activation circuit with a pair of BJT and a Mosfet with which I control an amplification stage between the DAC output and my device. The answer has been satisfactory and it is the option I will take for my project.

I appreciate your help and time to resolve my concerns. Thank you for help me with your knowledge and your contribution.