pavel-demin/red-pitaya-notes #678: Do not succeed to use the VNA app in transmission mode to measure filters

satfan52, 2018-02-27:
Hello Pavel,

I am trying to use the VNA app to measure filters but I have not much success, at least compared to the standard "frequency response analyzer" app that gives much better results. The set up that I use for the measurement is identical in the two experiments (frequency response analyser & VNA) . I use a 10db attenuator on the output side of the RP and 50 ohm load n the input side of the RP to present an impedance as close as possible to 50 ohms. The attenuator and the 50 ohm load are attached to the DUT and connected with good quality and short SMA cables to the RP. I checked the ôpen, short, load and DUT data and I see nothing unusual. It is the GAIN OPEN and GAIN SHORT tabs that  provide strange results. I would like to use the VNA for all my RF measurements but because of this problem, I just can't and must fallback to the standard frequency response analyzer app.  See pictures below

Regards
Peter

pavel-demin, 2018-02-27:
Hello Peter,

Thank you very much for these tests.

If I understand correctly your setup, then I'd say that only the 'Gain short' tab is usable with this setup and the short calibration should be done with the attenuator directly connected to the 50 Ohm load. 

Could you please explain me how you calibrate both applications?

I didn't look at the 'Frequency Response Analyzer' application before. It's good to know that it works and that its source code is available.  So, I can try to implement a similar algorithm in my VNA application.

For my information, here are some links:

 - Frequency Response Analyzer source code
  https://github.com/RedPitaya/RedPitaya/tree/master/apps-free/freqanalyzer

 - Frequency Response Analyzer documnetation
  http://redpitaya.readthedocs.io/en/latest/appsFeatures/marketplace/marketplace.html#frequency-response-analyzer

Best regards,

Pavel

pavel-demin, 2018-02-27:
Looks like the gain calculation algorithms used in both applications are very similar:

https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/vna/client/vna.py#L662
https://github.com/RedPitaya/RedPitaya/blob/master/apps-free/freqanalyzer/src/dsp.c#L199

In both applications, the frequency response is calculated as DUT signal amplitude over short signal amplitude.

satfan52, 2018-02-27:
Dear Pavel,

 I have tried two different set-up, tried both of them in the two experiments (VNA of the frequency response analyse) and I have obtained the same resulsr

set-up#1 using a SMA-T2

a) for calibration
                                                              OPEN standard
                                                                 or
                                                              SHORT standard
                                                                 !
RPinput1 --(sma cable)--SMAT-1 ---- SMA-T2 -----attenuatpr -10 db ---(sma cable)---RPoutput1
                                            !
                                          50  ohms dummy load

I hit the "open button" of the GUI when the OPEN standard is connected to SMA-T2
I hit the "short button" of the GUI when the SHORT standard is connected to SMA-T2
I hit the "load button" of the GUI when the OPEN standard is connected to SMA-T2

b) the actual measurement after following the calibration above

                                                OPEN standard
                                                     !
RPin1 --(cable)--SMAT-1 ---- SMA-T2 ------- Filter (DUT)-attenuatpr -10 db ---(cable)---RPout1
                                  !
                         50  ohms dummy load

I hit the "single button" and see the results indicated  on the pictures

set-up#2 using a  3 bead reflexion bridge instead of SMA-T2

the calibration procedure and the measurement procedure are the same except that I use the 3 beads reflection bridge instead of SMA-T2. Of course the ref port of the bridge is connected to a 50 ohms dummy load

                                                     OPEN standard
                                                     connected to bridge DUT port
                                                             !
RPin1 --(cable)--SMAT-1 ---- (out) bridge (in) ------- Filter (DUT)-attenuatpr -10 db ---(cable)---RPout1
                                  !                         !
      50  ohms dummy load                50 ohms
                                                         connected to ref port

Regards
Peter

pavel-demin, 2018-02-27:
Dear Peter,

Thank you for the detailed description of the calibration procedure.

I'd say that in the set-up 1 the 'Gain open' tab should work.

However, I'd try an even simpler set-up:

 - hit the 'short' button with the following connection:
```
  RPinput1--(sma cable)--SMAT1--attenuator--(sma cable)--RPoutput1
                           !
                       50 Ohm load
```
 - hit the 'single' button with the following connection:
```
  RPinput1--(sma cable)--SMAT1--Filter--attenuator--(sma cable)--RPoutput1
                           !
                       50 Ohm load
```
 - switch to the 'Gain short' tab

If it doesn't work, could you please send me the .cfg files for some of these set-ups? The .cfg files contain the calibration and the measurement data. With these data, I'll try to check the gain calculations.

Best regards,

Pavel

satfan52, 2018-02-27:
Hello Pavel,

I have done the test that you suggested and unfortunately the result obtained is the very same than with the procedure that I followed, it is actually much faster to do (thanks for pointing out the shortcut). I attach the picture and the cfg files of my test.

Regards
Peter  

[image]
[image]
[image]
[vna.ini.zip](https://github.com/pavel-demin/red-pitaya-notes/files/1764351/vna.ini.zip)
[vna.csv.zip](https://github.com/pavel-demin/red-pitaya-notes/files/1764355/vna.csv.zip)

pavel-demin, 2018-02-28:
Hello Peter,

Thank you for the ini file.

The VNA client for MS Windows ([vna-win32-20171130.zip](https://github.com/pavel-demin/red-pitaya-notes/releases/download/20171130/vna-win32-20171130.zip)) works fine:
[image]

When I run the VNA client under Ubuntu 16.04, I obtain exactly the same results as you did.

The numpy versions are 1.13 in vna-win32-20171130 and 1.11 under Ubuntu 16.04. [numpy.interp](https://docs.scipy.org/doc/numpy/reference/generated/numpy.interp.html) in 1.11 doesn't support complex values. I've just added a workaround with [this commit](https://github.com/pavel-demin/red-pitaya-notes/commit/99a2a75c44792895621f980bb7e06b2dbaf8dc79).

You can apply this fix by downloading [vna.py](https://raw.githubusercontent.com/pavel-demin/red-pitaya-notes/master/projects/vna/client/vna.py) to the VNA client directory.

Best regards,

Pavel

satfan52, 2018-02-28:
Hi Pavel,

Yes!! this solves the problem in my linux Ubuntu 16.04 environment.  I tested successfully your fix.
By the way, I am impressed with the dynamic range of my measurements with your VNA software, it is quite close of the theoretical limit of 92 dB for a 14bit DAC.
Thanks Again
Regards
Peter

closed 2018-03-02 by satfan52

pavel-demin, 2018-03-05:
The code that fixes compatibility with older NumPy versions is now in the [latest release](https://github.com/pavel-demin/red-pitaya-notes/releases/tag/20180305).