pavel-demin/red-pitaya-notes #680: build packetizer_test error

mhanuel26, 2018-03-01:
Hello Pavel, 

I am trying to build the packetizer_test project using vivado 2016.4 by doing

make NAME=packetizer_test tmp/packetizer_test.xpr

but it fails with following error.

```
WARNING: [BD 5-235] No pins matched 'get_bd_pins pktzr_0/S_AXIS'
WARNING: [BD 5-235] No pins matched 'get_bd_pins adc_0/M_AXIS'
WARNING: [BD 5-232] No interface pins matched 'get_bd_intf_pins adc_0/M_AXIS'
** ERROR: can't connect pktzr_0/S_AXIS and adc_0/M_AXIS
    while executing
"error "** ERROR: can't connect $cell_name/$local_name and $remote_name""
    ("foreach" body line 14)
    invoked from within
"foreach {local_name remote_name} [uplevel 1 [list subst $cell_ports]] {
    set local_port [get_bd_pins $cell_name/$local_name]
    set remote_port [g..."
    (procedure "cell" line 10)
    invoked from within
"cell pavel-demin:user:axis_packetizer:1.0 pktzr_0 {
  AXIS_TDATA_WIDTH 32
  CNTR_WIDTH 32
  CONTINUOUS FALSE
} {
  S_AXIS adc_0/M_AXIS
  cfg_data slic..."
    (file "projects/packetizer_test/block_design.tcl" line 32)

    while executing
"source projects/$project_name/block_design.tcl"
    (file "scripts/project.tcl" line 66)
INFO: [Common 17-206] Exiting Vivado at Thu Mar  1 16:36:55 2018...
Makefile:144: recipe for target 'tmp/packetizer_test.xpr' failed
make: *** [tmp/packetizer_test.xpr] Error 1
```

The led_blinker and adc_test build correctly. 

I am just noticing this all over

`awk: symbol lookup error: awk: undefined symbol: mpfr_z_sub`

Could you please let em know what could be wrong? 

Thank you in advance,

pavel-demin, 2018-03-01:
Thank you for testing the packetizer_test project. Looks like I removed adc_0 from cfg_test with commit https://github.com/pavel-demin/red-pitaya-notes/commit/ce084086a3c677fe09470b62727becf122b708c9 and forgot to add adc_0 directly to packetizer_test.

I've just fixed it with commit https://github.com/pavel-demin/red-pitaya-notes/commit/a32bfd143bebb0f8d7f17fa5963850a8287b6df2.

mhanuel26, 2018-03-01:
Thanks, it works now. 

Would you allow me a question here. I understand you use the axi_ram_writer for DMA as in adc_test and packetizer_test, in the case of a DAC, what would be the DMA option or how do you achieve high throughput?

Best,

pavel-demin, 2018-03-05:
> in the case of a DAC, what would be the DMA option or how do you achieve high throughput?

If one of my projects would require the DMA functionality for a DAC, then I think that I'd write a new IP core (axi_ram_reader).

There is also [AXI DMA Controller](https://www.xilinx.com/products/intellectual-property/axi_dma.html) provided by Xilinx.

closed 2018-03-05 by pavel-demin