pavel-demin/red-pitaya-notes #685: GPIO problems

shinyRM, 2018-03-08:
hello,

I am trying to work with both the sdr_transceiver and the sdr_transceiver_emb with the red pitaya while still having access to the red pitaya's GPIO.  I have successfully gotten the sdr_transceiver to work with GNU Radio using the latest image with alpine, but cannot seem to access the GPIO.  

I read through issue #476 and have tried using gpio-output.c, but following the instructions I can only enable one pin when the red pitaya is set to be a sdr_transceiver. With the sdr_transceiver_emb project I cannot set any of the pins.  

Are these projects not compatible with sdr_transceiver_wspr/gpio-output.c anymore or am I missing something else? 

Thank you

pavel-demin, 2018-03-08:
Looking at the code, I'd say that sdr_transceiver and sdr_transceiver_wspr are still using the same addresses for the registers controlling the pins: 
https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/sdr_transceiver/server/sdr-transceiver.c#L84
https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/sdr_transceiver_wspr/app/gpio-output.c#L34

To be usable with sdr_transceiver_emb, [gpio-output.c](https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/sdr_transceiver_wspr/app/gpio-output.c#L34) should be slightly modified. `cfg + 2` should be replaced with `cfg + 3`.

shinyRM, 2018-03-09:
Thank you very much for your quick reply.  That change to the gpio-output.c allows me to toggle the GPIO with the sdr_transceiver_emb.

closed 2018-03-09 by shinyRM