pavel-demin/red-pitaya-notes #687: ADC, DAC and filter

One19wil, 2018-03-10:
Hello Pavel,
I start in FPGA Programming with Vivado on the red-pitaya and I would like to realize a simple digital low-pass filter with the FIR Compiler IP.
To test, i connect an adjustable sine generator to the input IN1 of the red. OUT1 is connected to an oscilloscope to see the result. Here is my configuration:
[image]
I used a subset converter to keep that input IN1:

[image]

I add a FIR Compiler configure like this:

[image]
[image]
[image]
[image]

I chose a sampling frequency at 20khz and I created my FIR on TFilter to generate its form:

[image]

finally, I connect the set to another subset to send it to the DAC:

[image]

I compiled the project and then executed it on the red-pitaya. The problem is that the filter does not work, I vary the input frequency from ~100Hz to more than 1MHz, the output is always as 0v.
 
My questions are:
- Is it achievable as I did?
- What does not work?

Thank you for your attention and your work for all fpga projects on Red-pitaya ! 👍
Best Regards,
Willy

pavel-demin, 2018-03-10:
Hello Willy,

The block design configuration looks OK.

However, the input sampling frequency of the FIR Compiler module should be set to 125 MHz. Since the FIR Compiler module is directly connected to the ADC module, the input sampling frequency of the FIR Compiler module should be the same as the output sampling frequency of the ADC module.

Best regards,

Pavel

closed 2018-03-12 by pavel-demin

One19wil, 2018-03-12:
Hello,

I tested what you said and it's work for a sample frequency of 125MHz.
 
With this #352 issue, i figured out how to change my sample rate to create my low pass filter with a cutoff frequency at ~ 4200Hz. ( same as TFilter design )
I have added 2 CIC Compiler working as DDC and DUC. 

Here is my new design:
[image]

CIC1: 
Type: Decimation
Sample Rate fixed at: 6250
Input Sample & Clock Frequency: 125 MHz
Input Data Width: 14
Output Data Width: 24

FIR1:
Same filter design
Input Sample Frequency: 0.02 MHz
Clock Frequency: 125 MHz
Input & Output Data Width: 24

CIC2:
Type: Interpolation
Sample Rate fixed at: 6250
Input Sample Frequency: 0.02 MHz
Clock Frequency: 125 MHz
Input & Output Data Width: 24

Subset Converter:

[image]

As you specified in this post:

[image]

And it's work ! my signal is decreased with a frequency higher than the Fc.
Thank you for your help :+1: and i will continue to test different type of filter.

Best Regards,
Willy

pavel-demin, 2018-03-12:
Thanks for the update. I'm glad that you've found a configuration that works for you.

BTW. For a flat frequency response, the calculations of the FIR filter should take into account the frequency response of the CIC filters as, for example, described in AN 455 by Altera:
https://www.altera.com/en_US/pdfs/literature/an/an455.pdf

AndiCure, 2022-03-01:
Hello,
I see this issue is already closed, but I may try to ask a question.
I'm working with the RedPitaya 125-14 and what I am currently trying do to is simply bypassing incoming signals from Fast ADC to DAC.
It already works, but I have a signal in frequency spectrum that I cannot identify.
I use two RedPitaya's, one for bypassing and one as analyzer.
The incoming signal comes from a Pulse/Function Generator from hp.
I am pretty sure that the strange signal does not come from the generator.

Could it be the difference in input impendancy to 50 Ω?
What i am thinking of is my clocking management. Can you explain to me how the ddr_clk of the DAC has to be clocked. And maybe how you magage the timing in general. PLL or MMCM?
That would be very helpful to me. Thanks

[image]

[image]

[image]
.

pavel-demin, 2022-03-01:
My version of a similar DAC/ADC configuration can be found in the dac_adc project: 
https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/dac_adc/block_design.tcl

In this project there are two 250 MHz PLL outputs for ddr_clk and wrt_clk with different phase offsets (157.5 and 202.5).

I think that this way the PLL outputs are configured to meet the timing requirements shown on page 25 of the AD9767 DAC datasheet:
https://www.analog.com/media/en/technical-documentation/data-sheets/AD9763_9765_9767.pdf

AndiCure, 2022-03-04:
Thanks for the fast response, really appreciate it.
No I can be sure that these two wide signals I was talking about does not come frome the clocking of the DAC.
At least it seems so.
I connected the DAC of RP-Board 1 to ADC of Board 2 without programming the device with the own generated bitstream nor started the Signalgenerator-App on Board 1. Even then I see these "broadband" signals.
I can definitely exclude 
-> the clock of the DAC (because, why should it generate wide signals in general)
-> the used signal generater (build in or the one using in bypass-mode)

I think that it's some onboard communication that interfers the DAC. I think it is intersting but shouldn't bother me in what I am trying to do.

My future plans with the Red Pitaya is long time data Aquicition.
I'm trying to use it as a measurement device for Adaptivity measurements, where I only record if there is any disturbance on specific channels and when it occurs.
An other idea was demodulation of a bursted FSK signal, where I try to detect a start bit in the preamble of a frame. This is for jitter measurements. I would like to only write the timestamp in memory and then evaluate over several packets.

If someone of you tried  something similar, I would be happy for every little hint, documentation, websites or other sources.

Thank you again pavel-denim that you took the time to answer.

kevinamonteverde, 2024-08-25:
> My version of a similar DAC/ADC configuration can be found in the dac_adc project: https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/dac_adc/block_design.tcl
>
> In this project [...]

Hi, I know it's more than two years after the fact, but I'm trying to pass input from the ADC straight to the DAC on red pitaya and I'm having some trouble with how my signal looks. It'd be really useful if your dac_adc project was still up, but it seems like it isn't. Do you have any idea where I could find something to help me do this? 

I'm getting output from both DACs, even when I'm only inputting to one ADC. My output looks like a wave of the correct frequency and phase but with a much lower magnitude (.2V, when my input signal is .8V), with what appears to be noise as high as .8V. 

Because of this, I think I have a problem with timing, particularly with the DAC selector pin.

pavel-demin, 2024-08-25:
There have indeed been many changes over the last few years.

I renamed the `dac_adc` project to `template` and added a short description to the notes. It can be found at [this link](http://pavel-demin.github.io/red-pitaya-notes/axi-hub/#usage-examples).

A pre-built Vivado project can be found in the `template` directory in the [release zip file](https://github.com/pavel-demin/red-pitaya-notes/releases/download/20240204/red-pitaya-notes-20240204.zip).

kevinamonteverde, 2024-08-26:
> There have indeed been many changes over the last few years.
>
> I renamed the `dac_adc` project to `template` and added a short description to the notes. It can be found at [this [...]

Thanks so much. I'm trying to open the block diagram in vivado now but i'm getting some errors such as the following:

 [BD 41-50] Could not find an IP with the given vlnv: pavel-demin:user:axis_red_pitaya_adc:1.0 

When I try to add the IP file I get another error. Please help! I'm new to this.

pavel-demin, 2024-08-26:
If opening the Vivado project does not work for you, then I can only recommend building the `template` project from source using the following commands:
```
source /opt/Xilinx/Vitis/2023.1/settings64.sh

git clone https://github.com/pavel-demin/red-pitaya-notes
cd red-pitaya-notes

make NAME=template bit
```

kevinamonteverde, 2024-09-04:
> If opening the Vivado project does not work for you, then I can only recommend building the `template` project from source using the following commands:
>
> ```
> source [...]

Hi, thanks so much. I built the project and tried programming the red pitaya by sending template.bit to the red pitaya then using the command 

`cat template.bit > /dev/xdevcfg`

And nothing happens. I assume this project passes a wave through, so waves input through ADC 1 should be output through DAC1? Am I missing anything? Please let me know and thanks again.

pavel-demin, 2024-09-04:
> And nothing happens

Please open a new issue and provide all the requested information.

TanviWani, 2025-03-17:
Hello [pavel-demin]. I'm just starting to use RED PITAYA and I'm facing some issues. I absolutely love the way you solve the issues here. I will take time to understand the convo here but I will be grateful to seek your help. Kindly let me know if and how can i seek your help.