pavel-demin/red-pitaya-notes #688: Food for thoughts  - long term evolution of the VNA app

satfan52, 2018-03-22:
Hello Pavel,

I was wondering the VNA client/server apps (?) could be enhanced to be more easily used  as a sweep based Spectrum analyzer  for checking the wide-band spectra of a devices or of antennas for instance... I understand that the VNA app is much more powerful than a simple spectrum analyzer or signal generator but using the VNA app for spectrum analysis or at two tone signal generation for instance is not really convenient (if feasible)  because the App was never designed for that purpose. 

I was hoping to use the Bscope functionality of QUISK for the the purpose of spectrum analysis but since it is not supported  I  had no other choice than using the standard spectrum analyzer apps of the RP environment instead.  I am of course very happy that  standards open-source apps like the frequency analyser one exist and I am VERY thankful to their developers, it  is just that I have to switch environment (I use Alpine) back and forth to go back to the Standard RP environment when I need to do spectrum analysis or simply generate a two tone signal of precise amplitude and precise frequency. So I was wondering if it could be possible to add a sweep based "spectrum analyzer" and maybe also a precise and ham-rf-engineer oriented  "signal generator" functionality (two tone or not)  to the VNA app, so one single app could be used for all measurements purpose in the frequency domain; VNA, Sweep based Spectrum Analysis and Signal generation (two tone included).

Another thing that stroke me recently is very the low cost of chips like the ADF4351 which are able to generate relatively clean signals between roughly 40Mhz and 4.4 Ghz at up to 5dBm. I am wondering to which extend it would be possible or not for the VNA app to pilot these very cheap chips with the objective to extend the frequency range of the VNA app well beyond the current limit of 60 Mhz. In other words use these cheap device (and a mixer) as a sort of "front-end" in order to extend the frequency range of the VNA app in the GHz range at a marginal costs. I know that commercial VNA such as the miniVNA exist on the market for 300-400 euros, but they are not 14 bits and usually do not offer a better dynamic range than a mere RTL-SDR chip. Even top-nodge devices like the Lime-SDR are only 12 bits (and there is not equivalent of the VNA app anyway for the LimeSDR). 

Regards
Peter 

Regards
Peetr

pavel-demin, 2018-03-22:
Hello Peter,

Thank you for all the suggestions.

There already several spectrum analysis tools that can be used with Red Pitaya:
 - [QSpectrumAnalyzer](https://github.com/xmikos/qspectrumanalyzer) can be used with the SDR transceiver application for Red Pitaya. The parameters are `driver=redpitaya, addr=192.168.1.100, port=1001`
 - [soapy_power](https://github.com/xmikos/soapy_power) `soapy_power -d "driver=redpitaya, addr=192.168.1.100, port=1001" -r 500000 -k 20 -g 0.0 -f 1000000:60000000 -B 1000`

The ADF4351 chip can be controlled via a SPI interface. So, it can be controlled from Red Pitaya. A more difficult part would be synchronizing it with the Red Pitaya ADC.

There is an [article about VNA and LimeSDR](https://myriadrf.org/blog/limesdr-made-simple-part-9-vna) with [some python code](https://github.com/myriadrf/pyLMS7002M/tree/master/examples/VNA).

Best regards,

Pavel

satfan52, 2018-03-23:
Thank you for your answer Pavel,

There is also osmocom_fft which I have used successfully with the SDR transceiver. 

Just a clarification regarding the SDR transceiver: From what I understand from my tests with it,  the bandwidth of the spectrum analysis is limited to max 2.5 Mhz (corresponding sampling rate of  the wideband SDR transceiver).  Unless I am mistaken, the only 2 apps that allow to get a wideband DC to 60 Mhz spectrum analysis with the RP are the two ones available on the bazaar market (and again a big thanks to their developers!). 

Of course buying another device with a higher frequency of operation is always an option but the sense of my question was precisely to know if it is possible to avoid that to stay as much as possible in the RP environment (hardware and software wise) ? by building a cheap front-end and making some limited software changes to existing applications ? Think of the role of the R820T2 tuner that is used in the RTL-SDR device for instance (or what other relevant device that might fit the purpose) ?   

Regards

pavel-demin, 2018-03-23:
Both [QSpectrumAnalyzer](https://github.com/xmikos/qspectrumanalyzer) and [soapy_power](https://github.com/xmikos/soapy_power) are sweep based as you described in the first post. They can stitch multiple 2.5 MHz pieces of spectrum to produce a wideband DC to 60 MHz spectrum.

If the idea is to use ADF4351 connected to a mixer for both the TX and RX signals, then I think that the VNA application can be easily modified to control ADF4351.

closed 2018-04-13 by satfan52