pavel-demin/red-pitaya-notes #689: Conversion of WSPR decoder to FT8

azwirko, 2018-03-23:
Hi again Pavel, the following is a continuation of a post I made awhile back

https://github.com/pavel-demin/red-pitaya-notes/issues/629 

Thanks for your advice, which I followed and I was able to successfully compile the latest WSJT-X 1.8 branch - https://sourceforge.net/projects/wsjt/files/wsjtx-1.8.0/ on the Redpitaya.  I was then able to run the "jt9" command line tool with the "-8" option to decode a pre-recorded FT8 transmission (.wav file) - https://sourceforge.net/projects/wsjt/files/samples/FT8/ . The decode took ~1 second on Redpitaya.

My long term goal is to learn Vivado and FPGA, as well as use your WSPR code as a basis to create a multi-band FT8 decoder only, no transmit.  Since FT8 seems more popular and there exists a propagation, mapping server called PSKreporter - https://pskreporter.info/pskmap.html , similar to WSPRnet.org ,  I am hoping to create a receiver to feed into this propagation reporting system.

So, if there is any chance you could answer some questions or point me in the right direction, or provide ideas, I'd much appreciate it, thanks!  If this is too much, I completely understand as I'm just a beginner when it comes to DSP on FPGA.

Questions

1. Is using your WSPR multi-band decoder as a basis for my idea a good starting point, or do you recommend one of your others? As a starting point I would reset the multiple HF frequencies from WSPR to FT8 frequencies - http://www.qsl.net/sv1grb/psk31.htm 

2. Would it be feasible and what would I need to change such that I could capture ~2500-3000 hz of bandwidth above a given FT8 base frequency, instead of the narrow band 375 hz that you are currently capturing for WSPR? 

3. The above stated question implies a ~8x rate increase over your current operation.  Do you think buffer sizes and write speeds will be exceeded with this requirement?  See next question for buffer changes.

4. Can the buffer capture length / time be changed easily? The needed capture time at the above rate is only 13 seconds of data, instead of ~110 seconds for WSPR.  I see you use system clock and cron to kick off process every 2 minutes.  I would need to start capture at 0, 15, 30 and 45 seconds after the minute.  Or to state differently, write a file at 13, 28, 43 and 58 seconds after the minute.  So cron would not be usable, unless I start every minute and run 4 sequential decodes (15 seconds each).  See next question for another issue regarding the file data format.

5. The wsprd decoder expects a complex .c2 data file. Do you think it would be possible in the FPGA to create a WAV file of the USB pass band?  "jt9 -8" expects an FT8 audio file not a c2 complex file. It looks like you have an embedded transceiver, so does any of that project produce straight audio?   I see there is a CSDR DSP library - https://github.com/simonyiszk/csdr that converts IQ to SSB (RTL example) so I guess I could do that conversion on the CPU, but was wondering what could be done on the FPGA. Are there any existing blocks/code?

Even if the Redpitaya could only decode 4 or 5 bands, I think that would still be  useful tool for the community.

As far as the call sign, spot upload feature, WSJT-X already has the routines to send to PSKreporter, so that code I could reuse as is, once I extracted the data from the jt9 decoder.

I've enjoyed looking over your various project notes online and hope I can create something as useful for the ham radio community.

73

andyz - K1RA

pavel-demin, 2018-03-23:
Hi Andy,

It's great that you've managed to build and to run the FT8 decoder. With the 1 second decoding time, it could be still possible to simultaneously decode several bands.

If I'm not mistaken, the FT8 decoder requires 12000 samples per second. I've just checked the code in [ft8_decode.f90](https://sourceforge.net/p/wsjt/wsjt/HEAD/tree/branches/wsjtx/lib/ft8_decode.f90) and it looks like `sync8` reads the samples at 12000 samples per second.

Since 12 kSPS can be obtained from 125 MSPS as `125e6/25/125/(5/3)/2`, the parameters of the CIC and FIR filters in the multiband WSPR transceiver should be modified accordingly.

I've put together a make file that builds `ft8d` without building the entire wsjtx package but when I run `ft8d` it crashes because of a segmentation fault. It's a known problem:
https://sourceforge.net/p/wsjt/mailman/message/35940168/

So, looks like the FPGA configuration can be built based on the multiband WSPR transceiver project. It would be nice to find a working FT8 decoder that doesn't require building the entire wsjtx package.

Best regards,

Pavel

pavel-demin, 2018-03-23:
I've just added a FPGA configuration with all the required modifications:
https://github.com/pavel-demin/red-pitaya-notes/tree/develop/projects/sdr_transceiver_ft8

azwirko, 2018-03-24:
Pavel,

  Thanks for the quick response and all the information and modifications.  I believe the "jt9" binary is the executable we need to run for decoding FT8, I don't think the "ft8d" is up to date or supported.  Here is a copy of the help output for "jt9" as run on my Redpitaya command line

https://docs.google.com/document/d/1FVfpCKlMhSnXaePuC9wQM8OY__xB7V4y-MRuvWZ9om4/edit?usp=sharing 

As you can see the "-8" or "--ft8" options force decoding of FT8 mode WAV files using this 422K executable (see below).  There are several other modes this can do as well "-6" JT65, "-4" JT4, "-9" JT9, etc.

Since I already built WSJT-X on my Redpitaya and installed in /usr/local, today I tar/gzip it up, which only includes just the WSJT-X files from the bin, include, lib and shared folders.  I've made available for download if you wish to try running "jt9" on your system.

https://drive.google.com/open?id=1kKnrWl7XFr2fJ_uC3HLeuyrdAsUJn4zK 

Then just do:

cp wsjt-x-usr-local.tgz /usr/local
cd /usr/local
tar zxvf wsjt-x-usr-local.tgz

And hopefully you will have all the needed binary files to try for yourself.

Thanks again!

andyz - K1RA

pavel-demin, 2018-03-24:
Thank you for sharing the pre-built WSJT-X binaries. However, my point is to have a FT8 decoder that is easy to build and to modify. The examples of such decoders are [wsprd](https://sourceforge.net/p/wsjt/wsjt/HEAD/tree/branches/wsjtx/lib/wsprd/) by Steven Franke, K9AN and [weakmon](https://github.com/rtmrtmrtmrtm/weakmon) by Robert Morris, AB1HL.

BTW, the FT8 decoder by Robert Morris, AB1HL supports two sample rates 12 kSPS and 6 kSPS. It would be interesting to do some tests and to compare the performance of the decoder with different sample rates.

pavel-demin, 2018-03-24:
> I don't think the "ft8d" is up to date or supported.

You're right `ft8d` crashes because it's not up to date. I've just updated it based on the code from [ft8_decode.f90](https://sourceforge.net/p/wsjt/wsjt/HEAD/tree/branches/wsjtx/lib/ft8_decode.f90) that is the code used by [jt9.f90](https://sourceforge.net/p/wsjt/wsjt/HEAD/tree/branches/wsjtx/lib/jt9.f90). 

Now, I can build and run `ft8d`. Here is the repository with a minimal set of files required to build the FT8 decoder:
https://github.com/pavel-demin/ft8d

pavel-demin, 2018-03-24:
I've tested the FT8 decoder with the .wav file from [this links](https://www.dropbox.com/sh/busdna76pjavht7/AACliXhi76ofE9p4YM0_xHdRa/FT8%20-%209%20and%2010%20decodes.zip?dl=1). There are 7 .wav files and 71 decodes. The decoding time is 32-33 seconds when the decoder runs on one CPU.

So, if the receiver works with 8 bands and the FT8 decoded runs on two CPUs, then the decoding time is around 20 seconds. Looks like the CPU isn't fast enough to decode 8 bands every 15 seconds. A solution could be to run it every 30 seconds.

pavel-demin, 2018-03-25:
> Do you think it would be possible in the FPGA to create a WAV file of the USB pass band? "jt9 -8" expects an FT8 audio file not a c2 complex file. It looks like you have an embedded transceiver, so does any of that project produce straight audio?

I think that this problem could be solved by slightly modifying the `four2a` calls in [sync8.f90](https://github.com/pavel-demin/ft8d/blob/master/sync8.f90#L33) and in [ft8_downsample.f90](https://github.com/pavel-demin/ft8d/blob/master/ft8_downsample.f90#L19). It should be enough to set the last argument to 1. The lines [27](https://github.com/pavel-demin/ft8d/blob/master/subtractft8.f90#L27) and [56](https://github.com/pavel-demin/ft8d/blob/master/subtractft8.f90#L56) in [subtractft8.f90](https://github.com/pavel-demin/ft8d/blob/master/subtractft8.f90) should be also modified to work with the complex input samples. With these modifications, the FT8 decoder should work with the complex samples at 6 kSPS.

azwirko, 2018-03-26:
I haven't compiled your ft8d decoder yet, but did download the zip file of WAVs you used and copied to the RP.  I re-ran my earlier timing test, this round using the earlier "jt9" binary I sent and the command "time jt9 -8 170711*.wav" which represents all seven, 15 second decode sequences and got 65 decodes with the following time:

real    0m9.010s
user    0m8.870s
sys     0m0.090s

Full output here

https://docs.google.com/document/d/1VUiG6j6-yWv6NqbwLnl8rx0VrmNUnKbfOogdAR50fdU/edit?usp=sharing 

I also ran just a single 15 second decode sequence WAV file and got

root@rp-f01e82:~# time jt9 -8 170711_215815.wav
215815 -12  0.6  630 ~  KC3DGM ON3MK -16
215815  -1  0.9  775 ~  MJ0LEL IK3FUS R+00
215815 -17  1.2  950 ~  CX1RK NS9I EN64
215815  -5  1.1 1011 ~  EA1AHY K1HTV -15
215815  -8  1.0 1062 ~  VK4WTN DF9OX R-11
215815 -20  1.1 1155 ~  EK6RSC G1GIL R-04
215815  -6  0.6 1371 ~  YO2NAA K8YAH EN80
215815  -9  0.6 1421 ~  CQ IV3DXW JN65
215815  -7  1.2 1785 ~  CQ YI3WHR LM22
215815   1  0.2 2018 ~  CQ ON3BZ JO20
215815 -12  1.1 2094 ~  KB6IBB W4AS 73
<DecodeFinished>   0  11

real    0m1.353s
user    0m1.300s
sys     0m0.050s

I'm not sure why yours is taking so much longer, but will compile and re-run tests to validate here.

Thanks for all your help and work to get this system closer to operational.

pavel-demin, 2018-03-26:
Thanks for checking the timings. I've just found that I can reproduce your result if I set the `ndepth` parameter to 1. If I set it to 2, then the decoding takes much longer:

| ndepth | decodes | time |
| ------- | ------- | ---- |
| 1 | 65 | 8 |
| 2 | 71 | 38 |

pavel-demin, 2018-03-26:
Since we now have a standalone FT8 decoder and the timings are OK, I think that the next steps are
 - modify the FPGA configuration to output the complex samples at 3000 kSPS,
 - modify the FT8 decoder to work with the complex samples at 3000 kSPS.

azwirko, 2018-03-29:
Pavel,
  Thanks for the next steps.  I will look at the FT8 decoder first, and the WSPR decoder which used complex samples.  Hopefully I can use that as a basis to figure out how to make the FT8 work the same way.

andyz

azwirko, 2018-03-30:
Pavel,

OK, understand and appreciate your efforts on attempting to convert the FT8 decoder.  Would the FPGA code still be able to generate 6000 ksps and write to the memory buffer space similar to WSPR? If so, I would like to at least be able to read that into Linux space and then play with CSDR library to convert to USB and then WAV to attempt using with the existing jt9 executable.  

If not, then I guess it'll have to wait until I become more fluent in Vivado, FPGA and able to take it on myself.

Take care

andyz

pavel-demin, 2018-03-30:
Just noticed that I wrote too many zeros in some of my previous comments. It's fixed now.

> Would the FPGA code still be able to generate 6000 sps and write to the memory buffer space similar to WSPR?

I don't think that the existing jt9 executable requires the 6000 SPS sample rate. The current version of the FPGA configuration outputs the complex samples at 12 kSPS. The unmodified FT8 decoder works with the real samples at 12 kSPS. So, the sample rates are OK.

> If so, I would like to at least be able to read that into Linux space and then play with CSDR library to convert to USB and then WAV to attempt using with the existing jt9 executable.

I'd say that using CSDR for converting complex samples to real samples is an overkill. Just take every second sample and that's it. It can be done either in the program that reads the samples from the FPGA or in the FT8 decoder.

> If not, then I guess it'll have to wait until I become more fluent in Vivado, FPGA and able to take it on myself.

Maybe it wasn't clear from [one of my previous comments](https://github.com/pavel-demin/red-pitaya-notes/issues/689#issuecomment-375825207) but the FPGA configuration is ready:
https://github.com/pavel-demin/red-pitaya-notes/tree/develop/projects/sdr_transceiver_ft8

What is needed is some C or Fortran code to efficiently decode the FT8 signals. Would you be interested to do some coding for this project or do you expect me to write all the code?

pavel-demin, 2018-03-31:
I've just tested the FPGA configuration with the FT8 decoder and they seem to work well together. I can see many decodes. Here is a link to the SD card image that I'm using:
https://www.dropbox.com/sh/5fy49wae6xwxa8a/AACLW3qmAFX7mIUsNZxiiVJOa/sdr/red-pitaya-alpine-3.6-armhf-20180331-ft8.zip?dl=1

The application starts automatically at boot. The decodes can be seen in `/dev/shm/decode-ft8.log`. Since it's just the very first working prototype, some functionality is still missing. Pull requests with improvements are welcome.

pavel-demin, 2018-03-31:
Here is a new version:
https://www.dropbox.com/sh/5fy49wae6xwxa8a/AABWCGvJtMydxGMYqpII7VcHa/sdr/red-pitaya-alpine-3.6-armhf-20180401-ft8.zip?dl=1

In this version, the FPGA configuration and the FT8 decoder both work with complex samples at 12 kSPS. The number of decodes doubled.

pavel-demin, 2018-04-01:
Finally, I have a version that works with the complex samples at 4 kSPS. With this sample rate, the FPGA configuration contains only one FIR filter:
[image]

I've started to put together some notes about this project:
http://pavel-demin.github.io/red-pitaya-notes/sdr-transceiver-ft8

azwirko, 2018-04-01:
Pavel, 
  First, happy Easter. I'm just returning from a long weekend with the family and visiting in-laws.  Let me say - Wow!  And thanks again for the work and documents.  I did not expect you to do any of this.  I do expect to work on some level of coding, but as I mentioned its a long term project for me to better understand FPGA, Vivado, etc.  I grew up learning FORTRAN in high school and college and also have worked with C earlier in my professional care, but I'm an systems engineer at heart and mostly enjoy the hardware aspect of systems.  When I usually write code its for my eyes only,, but I hope to be able to write at a level to share with the wider community.  That means I need to learn to document better :)  Again this is all a goal of mine.  Family, work then radio...

Best regards,

andyz - K1RA

pavel-demin, 2018-04-03:
I've modified [write-c2-files.c](https://github.com/pavel-demin/red-pitaya-notes/tree/master/projects/sdr_transceiver_ft8/app/write-c2-files.c) to record 236000 complex samples at 4000 samples per second. This way, it records the four 15s transmission periods every minute and can be started by cron once a minute. With this new version, I see more than 200 decodes per minute and more than 10k decodes per hour.

If you want to give this new version a try, here is a link to the new version of the SD card image:
https://www.dropbox.com/sh/5fy49wae6xwxa8a/AABWCGvJtMydxGMYqpII7VcHa/sdr/red-pitaya-alpine-3.6-armhf-20180401-ft8.zip?dl=1

Looks like the only missing part is an uploader to a remote database.

ZL3IO, 2018-04-04:
Great job again, Pavel! Thanks.

This is from the skimmertalk reflector:
The Reverse Beacon Network (RBN) is starting to incorporate FT8 spots. According to a post by Pete, N4ZR, "We have three Skimmers running an experimental Aggregator that receives UDP messages from WSJT-X and forwards FT8 spots to the RBN...many rare DX stations never explicitly repeat (their) CQs. To get around this, Dick, W3OA, has programmed the Aggregator to respond to a unique standard FT8 message format that is always associated with stations being called, rather than those calling them. We label these as CQ spots, though we're aware it's a misnomer." 

May be that does already what you need to upload spots. 
Cheers Holger, ZL3IO

pavel-demin, 2018-04-05:
Hello Holger,

I've looked at the WSJT-X's UDP protocol description at

https://sourceforge.net/p/wsjt/wsjt/HEAD/tree/branches/wsjtx/NetworkMessage.hpp

and at the Qt serialization algorithm at

https://github.com/qt/qtbase/blob/5.11/src/corelib/serialization/qdatastream.cpp

Looks like this protocol is quite easy to implement. I'll try to add it to the FT8 receiver application for Red Pitaya. 

Best regards,

Pavel

azwirko, 2018-04-05:
For comparison, or as an alternative, PSKreporter at https://pskreporter.info/pskmap.html uses a format that can be found as implemented in

https://sourceforge.net/p/wsjt/wsjt/HEAD/tree/branches/wsjtx/psk_reporter.cpp  

and

https://sourceforge.net/p/wsjt/wsjt/HEAD/tree/branches/wsjtx/psk_reporter.h 

andyz - K1RA

azwirko, 2018-04-06:
Pavel,
  Downloaded and running your image and I'm also seeing ~200 decodes per minute - WOW! I've been running since 0000 GMT and decoding signals on all 8 bands from 160m through 15m - NICE!

andyz

azwirko, 2018-04-06:
For those who want a single, local network, CW Skimmer-like, DXcluster accessible stream for the FT8 decoder, for use with programs like RXClus by HB9BZA http://www.hb9bza.net/rxclus-overview - I've got a simple awk script that can be used with the Redpitaya's onboard 'tail' and 'nc' commands to provide a TCP listener which you can Telnet too, or point your favorite DXcluster client at.  It can be downloaded here

https://drive.google.com/open?id=15wYQwUah1JFLNotUkglsY_2mUNiuFbcR 

Once you SCP to the RP set execute permissions

`rp-f01e82:~# chmod 755 ./ft8-skimmer.awk`

Then start by using following command line

`rp-f01e82:~# tail -f /dev/shm/decode-ft8.log | ./ft8-skimmer.awk | nc -lk -p 7373`

Then from your favorite PC client either get the redpitaya IP address and

`telnet redpitaya 7373`

or put redpitaya:7373 in your DXcluster client app

And you should start seeing CQ only spots like this

```
DX de K1RA-#:    18101.4 N7MDW      FT8  -9 dB CN94 CQ    1805z
DX de K1RA-#:    18101.9 WJ6T       FT8  -7 dB DM05 CQ    1805z
DX de K1RA-#:     7075.2 N9YBK      FT8  -4 dB EN61 CQ    1806z
DX de K1RA-#:     7075.2 N9YBK      FT8  -3 dB EN61 CQ    1806z
DX de K1RA-#:    10137.0 KF6NFW     FT8 -15 dB EN12 CQ    1806z
DX de K1RA-#:    10136.8 K1LOL      FT8  -1 dB FN34 CQ    1806z
```
This method currently only supports a single TCP inbound connection.

73

andyz - K1RA

F1EYG, 2018-04-11:
Very nice job
I testing the last version xxxxx403 
verticale antenna directly connected to IN1 without any adaptation
120 spots decoded for 1mn interval recording :-))
73
Sylvain F1EYG

azwirko, 2018-04-11:
I'm no programmer, but I've recently written a few other scripts (in PERL) that parse the /dev/shm/decode-ft8.log file.  One script creates a WSJT-X UDP compatible stream that will feed GridTracker - https://tagloomis.com/ and the other will post to PSKReporter - you can see what my Redpitaya is copying over the last 15 minutes here:

https://pskreporter.info/pskmap.html?preset&callsign=k1ra&txrx=rx&timerange=900 

You can change the time frame to see more.  

If anyone is interested in playing with these scripts let me know.  I hope to convert over to C at some point, but if you can get PERL loaded on your RP, I'd be happy to share and let you give it a try.  I've been running the decoder and scripts on my RP running Ubuntu 16.04.3 LTS Red Pitaya GNU/Linux Ecosystem V0.98 Build 617

andyz - K1RA

kng, 2018-04-11:
@azwirko 
I would love to try the perl script for pskreporter, I was thinking of cobbling together something of my own (.pl/.py) but doing it directly in C would take quite some time...
Support for pskreporter directly in this project would be absolutely fantastic.

pavel-demin, 2018-04-11:
Hi Andy,

Have you managed to implement the following requirements listed at https://pskreporter.info/pskdev.html?

 - Ideally, a callsign should be reported only once per hour if it has not 'changed'. 
 - The datagrams should be sent at a rate of no more than one every five minutes (unless the packet becomes full).
 - Timed sends of packets must not be synchronized to the system clock.

Best regards,

Pavel

F1EYG, 2018-04-12:
> Pavel,
> I've managed to get call sign deduplication implemented within a single
> minute and was otherwise just happy to get the datagram in the right format
> and implemented such that PSKReporter was accepting my packets. There
> sometimes is a need for more than one datagram with the raw number of
> unique call signs. I still have some work to do as this is surely just a
> prototype. I will hold off publicly posting a link to what I'm currently
> using so the PSK server doesn't become overburdened.
>
> I can easily buffer 5 minutes worth of data before I send it and will work
> on that next. Randomizing the post timing with some sort of delay shouldn't
> be too hard to add as well. More to follow.
>
> Regards
>
> andyz

Tks again Pavel and Andy for your investment in these project :-))
Andy i am not a specialist in programmation, but i am ready to testing the
pskreporter upload script, i running the standalone Alpine version of Pavel
software.

73
Sylvain
F1EYG

azwirko, 2018-04-13:
I believe I've got several stable and usable PERL scripts for the Redpitaya FT8 skimmer decoder that I've been refining and using over the past week.  I still consider these beta, test versions, but believe they are good enough to hand out for others to use.  For anyone interested in trying these, please email me at k1ra@k1ra.us and I will send you a ZIP file with instructions on how to install, start and use.  

I have the following 3 scripts written in PERL, which therefore requires you to first install PERL on your RP. The scripts include:

PSKReporter uploader using IPFIX UDP protocol
WSJT-X UDP broadcaster - for use with apps like GridTracker
DXCluster Skimmer server - for use with RXClus, SpotCollector or Telnet client

All these scripts perform some level of caching and buffering and only report new calls that have not already been seen within the last 5 minutes.

73

andyz - K1RA

F1EYG, 2018-04-17:
Hello 
Result after 12 hours : Redpitaya vertical+antenna +PavelSoftware+FT8Skimmer
1187 spots and in top 10 all bands  on PSK reporter
Very very ...good  job Pavel, & Andy
Tks i continu test for i hope one week
[image]
Tks again
73 Sylvain

kng, 2018-04-17:
I watched myself SA2KNG and K1RA getting in to the top-20 reporters when running the sdr_transceiver_ft8 + ft8-skimmer this weekend. K1RA seems to have gone offline now but I'll try to have mine online as much as possible. I'm now one step below F1EYG but I guess he'll get a bit further towards the top. Very efficient solution, impresed by the results. Thanks for the effort on getting this to work.

azwirko, 2018-04-17:
SA2KNG & F1EYG - Very good on spotting FT8 stations! I was offline because of bad thunderstorms the other day, but back online again and hope to stay online 24x7. Thanks for the feedback on the scripts.  I have noted one issue with the dxc.pl which occasionally spots a 0.0 frequency which I've only seen happen on 160m, but could potentially happen on other bands..  I will post an update soon.  19K+ spot and 119 countries from me in 24 hours.  I'm using 400' horizontal loop. Here's my map

[image]

73

andyz - K1RA

F1EYG, 2018-04-18:
Hello all
I Try to stay on ligne minimum one week  😜for testing stability .i adding  a small fan to pitaya and think add input impédance transformer thé week 😀
73 f1eyg

pavel-demin, 2018-04-18:
> There's also the problem with a growing decode-ft8.log in /dev/shm/, around 10MB per day.

I agree that the outputting of the spots should be done differently. Writing the spots to decode-ft8.log was a quick and dirty solution for the prototype. Now, that the prototype has proved its usefulness, I'll try to improve the code that outputs the spots to a file.

kng, 2018-04-18:
I tried to make a named pipe instead, but you can't have all the ft8d decode to it. So I tried to dump the output of the ft8d's to a temporary file (ft8_$TIMESTAMP.log) and `cat ft8_$TIMESTAMP > ft8-pipe` after the `wait` in the end. Something probably needs to be changed in the pskr.pl to accomodate this however, didn't figure it out on the lunch break.

pavel-demin, 2018-04-19:
> I tried to make a named pipe instead, but you can't have all the ft8d decode to it.

I think that the following code should work:

```
for file in ft8_*_$TIMESTAMP.c2
do
  while [ `pgrep $DECODER | wc -l` -ge $JOBS ]
  do
    sleep 1
  done
  nice -n $NICE $DECODER $file &
done >> ft8_fifo
```
where `ft8_fifo` is a named pipe.

pavel-demin, 2018-04-19:
Would it be OK if I modify the FT8 decoder to output only the CQ message in the following format?
```
time    sync snr  dt    freq    call         grid
202900  33.8  -7  0.33  7076291 3Z0HDE       KN09
202915   4.1 -18  1.76  7074761 EA4GYR       IN80
```

kng, 2018-04-19:
That does work, at least when you have some other program reading from the pipe. But if the reader is not running then it will block the script and cause it to wait, accumulating all the work. Seems like direct pipe writing in the shell is a bad idea for just that reason. Also pskr.pl stops working when the log is truncated. I'm working on some kind of daily logpurge in decode-ft8.sh

kng, 2018-04-19:
> Would it be OK if I modify the FT8 decoder to output only the CQ messages in the following format?

AFAIK, all calls with locator, even the ones not containing CQ can be sent to pskreporter. In the wsjt-x code it looks like it needs to be of standard type (not free text I guess) and contain a valid grid.

pavel-demin, 2018-04-19:
I've updated the FT8 decoder. Now, it outputs all the spots with a grid locator. Here is how the format looks like:
```
time    sync snr  dt    freq    grid call
202915  35.8  -4  2.09  7074969 JO92 SQ5JBL
202915  22.3  -7 -0.01  7075161 IM67 CT4RC
```

azwirko, 2018-04-19:
There is no requirement to have to send a spot with a locator per the PSKReporter specification.  For those messages that have a grid, I send it, but if no grid is detected, I still upload the information from the FT8 message as the call sign, which in turn maps to a country, seems to be the most important piece of information.

From the developers info https://pskreporter.info/pskdev.html , I create a datagram with senderCallsign, frequency, sNR (1 byte), mode (1 byte), informationSource, senderLocator, flowStartSeconds.  If there is no grid in the message I set sendLocator to "" (null).

It seems PSKReporter caches or determines grid squares for stations on its own, as some stations I've uploaded without grid have either a 4, 6, or even 8 digit grid already stored or cached in the PSKReporter database.

andyz - K1RA

pavel-demin, 2018-04-19:
Thanks for the clarification. I can output an empty string for the spots without a grid locator. Is it OK?
```
time    sync snr  dt    freq    call         grid
202900  18.7  -6 -0.76  7074529 CT4VB
202900  24.8 -11  0.22  7074601 EA7FOF       IM77
202900  19.7   0  0.12  7074921 SP5UFE
202900  24.3 -12  0.22  7075265 EA1UY        IN52
```

azwirko, 2018-04-19:
Yes - that is great!  That will be much easier to parse and utilize grid if it is present.

pavel-demin, 2018-04-19:
I've updated the FT8 decoder. It can me reinstalled and rebuilt with the following commands:
```
make clean
make
```

pavel-demin, 2018-04-20:
In the yesterday's version, I forgot to filter out grid locators starting with KA or LA. It's fixed now.

Dick-W3OA, 2018-04-21:
Hi Guys - 
I'm Dick, W3OA, from the Reverse Beacon Network team and the author of Aggregator.  I've just found this conversation via Google and joined GitHub today.  My interest is finding out if your development is heading toward something that RBN node operators can use to send FT8 spots to the RBN.
Pavel, your email to me a few weeks back said you had downloaded the Aggregator beta and I wonder if you had made any progress getting it working with FT8?
The beta filters all the decoded FT8 messages from WSJT-X via UDP looking for both CQ messages and those in a format of call, call, signal report.  Our thinking is the second type will only come from stations that are responding to callers but may not have called CQ.
Additionally, Aggregator will not forward a spot for the same station on the same band until 10 minutes have passed.
73 - Dick, W3OA

pavel-demin, 2018-04-21:
Hi Dick,

Unfortunately, I haven't made much progress with the WSJT-X UDP protocol. However, I think that the WSJT-X UDP broadcaster developed by Andy, K1RA, could be used with the Aggregator beta.

Best regards,

Pavel

azwirko, 2018-04-25:
FYI I've got the beta Aggregator working with jtudp.pl - I've forwarded the latest script with the appropriate changes to Dick so he can try it too.

andyz

Dick-W3OA, 2018-04-25:
Thanks to andyz for making a few changes to his jtudp.pl file so Aggregator can decode its UDP messages.  My RBN node is now sending FT8 spots from Pavel's 8 band decoder to the RBN.

73 - Dick, W3OA

F1EYG, 2018-04-30:
Hello
I stop test these week ;-), I need time for hardware integration and for uptdate the software
i still using the first version of Pavel software with utility from Andy :-))
I have two questions 
- I follow the last progression about the way to upload spots to pskreporter but I do not understand how make the update ( i'm not confortable with linux just enough to survive).
- is that andy's perl script (dxc &  jtudp) still working ?
Tks
73 F1EYG

azwirko, 2018-05-01:
I have not yet updated my PERL scripts to handle Pavel's new log format.  It should be a relatively easy change, but I've been side tracked here.  I will attempt to compile the latest Pavel code and adjust my scripts accordingly and post any updates as I complete them.

73

andyz - K1RA

F1EYG, 2018-05-02:
Ok tks Andy , i take this time for make a special multibande antenna for this job 😓 
Best 73 
F1eyg

dj1an, 2018-10-14:
The FT8 decoder is working very well,
is there any development ongoing regarding pskreporter spotting?

@azwirko can you provide me your scripts mentioned in https://github.com/pavel-demin/red-pitaya-notes/issues/689#issuecomment-381208997?
tried to mail you, but no response.
Even if they are not working right now, i could update the Scripts to handle the new Log format.

73 de DJ1AN, Jan

pavel-demin, 2018-10-14:
Yesterday, I've finally found how to improve the few lines of code that I added to the [ft8_downsample](https://github.com/pavel-demin/ft8d/blob/master/ft8_downsample.f90) subroutine. The previous version was causing loss of some decodes that were very close to the center frequency. Now, it's fixed with [this commit](https://github.com/pavel-demin/ft8d/commit/f9e209f399f7bdc9080534c6b62f0ccf252ea338).

kng, 2018-10-15:
I'm running this on a ubuntu 0.98-617 installation and cannot get the ft8d to run on the latest commit. At ab3fe4b it does run, but not at f9e209f. 
Program received signal SIGSEGV: Segmentation fault - invalid memory reference.

dj1an, 2018-10-15:
got the Same Error with alpine 3.6

`Program received signal SIGSEGV: Segmentation fault - invalid memory reference.
Backtrace for this error:
`

ZL3IO, 2018-10-15:
Hi Pavel
Any chance to create a complete image that can be copied to a SD card. We don't care what Linux system as long as we can get a working image. Thank you in advance, Holger

pavel-demin, 2018-10-15:
Thanks for testing the new version on Red Pitaya.

While looking for how to solve this problem, I've found that I've broken the snr calculations with one of the previous commits. Fixing the snr calculations also fixed the segmentation fault problem.

dj1an, 2018-10-15:
Thanks for fixing the code Pavel, the new version seems to work.

pavel-demin, 2018-10-16:
> Any chance to create a complete image that can be copied to a SD card.

The latest image from one of my previous comments is still OK. Here is a link to this image:
https://www.dropbox.com/sh/5fy49wae6xwxa8a/AABFgDgTsFkDqAZnRc3g8iAFa/sdr/red-pitaya-alpine-3.6-armhf-20180403-ft8.zip?dl=1

[71 more replies]

closed 2019-03-24 by pavel-demin