pavel-demin/red-pitaya-notes #691: Making SDR Receiver compatible with HPSDR start at boot

bjornekelund, 2018-03-28:
### Description of the setup:

 * Device: Red Pitaya 
 * SD card image: 0.98-615
 * Application: SDR Receiver compatible with HPSDR - March 2018 version
 * Other relevant information:

### Description of the problem:
I am trying to make the receiver start at boot time. I tried following the instructions on 
pavel-demin.github.io, putting a copy of start.sh in the root ("/") directory but it seems to crash directly after start or not start at all. 

I also tried adding the following lines to rc.local: (from N6TV's guide on Hamsci.org)

cat /opt/redpitaya/www/apps/sdr_receiver_hpsdr/sdr_receiver_hpsdr.bit >/dev/xdevcfg
. /opt/redpitaya/www/apps/sdr_receiver_hpsdr/start.sh

With the same result. The strange thing is that if I start the receiver from the RP web interface and then stop it by leaving the web page, the start.sh script works until the next reboot.

My conclusion is that invoking the web interface seem to do some kind of initialization that is required for operation. Any advice on how to add this to my boot-up procedure would be welcome.

Thanks

### Steps to reproduce the problem:

 1. Reboot
 2. /opt/redpitaya/www/apps/sdr_receiver_hpsdr/start.sh

pavel-demin, 2018-03-28:
> SD card image: 0.98-615
>  I tried following the instructions on pavel-demin.github.io

The SD card image 0.98-615 is incompatible with my instructions.

My instructions work and were checked many times by many users. However, they should be followed exactly step by step without mixing them with other instructions.

When working with 0.98-615, the two commands from the description of the problem should work. It's exactly how the web interface starts the application. The first command configures the FPGA, the second command starts the server.

> Reboot
> /opt/redpitaya/www/apps/sdr_receiver_hpsdr/start.sh

In case of the SD card image 0.98-615, if you run only the second command (`start.sh`) without configuring the FPGA, then it's normal that it crashes.

The correct sequence is
```
reboot
cat /opt/redpitaya/www/apps/sdr_receiver_hpsdr/sdr_receiver_hpsdr.bit >/dev/xdevcfg
/opt/redpitaya/www/apps/sdr_receiver_hpsdr/start.sh
```

bjornekelund, 2018-03-28:
Dear Pavel,

Thank you for rapid and clear response. 

The solution turned out to be very embarrassing. 

It is spelled "chmod +x rc.local".

It now works perfectly.

pavel-demin, 2018-03-28:
I'm glad that you've found a solution. Closing this issue.

closed 2018-03-28 by pavel-demin