pavel-demin/red-pitaya-notes #695: qspectrumanalyzer & soapy_power not working with the SDR transceiver - Ubuntu 16.04

satfan52, 2018-04-14:
Hello,

I am trying to use the red-pitaya as a spectrum analyser using qspectrumanalyzer & soapy_power as a client to the SDR transceiver but it gives erroneous results in my Ubuntu 16.04 environment. The applications launches ok and the sweep starts fine but the spectrum displayed is very "creative" (to say the least, see picture enclosed).

the parameters I use are: driver=redpitaya, addr=192.168.0.102, port=1001

The problem could be with the SDR transceiver,  qspectrumanalyzer & soapy_power or in my own environment. However since my rtl-sdr dongle works "normally" with qspectrumanalyzer & soapy_power I suspect that the problem is not in my environment. 

If the problem is not (or not only) with SDR transceiver, my next step will be to report the problem to the developper of the qspectrumanalyzer & soapy_power applications

Regards
Peter

PS: No network throughput issues, I use a dedicated router to handle the traffic between the PC and the RP.

[image]
[image]
[image]

pavel-demin, 2018-04-14:
Hello Peter,

Thank you for the screenshots. I'd suggest to set 'Crop [%]' to at least 20.

Best regards,

Pavel

pavel-demin, 2018-04-14:
I've just tried to run qspectrumanalyzer and I've obtained some reasonably looking spectra with the following settings:
 - sample rate: 1.250 MHz
 - start frequency: 1.000 MHz
 - stop frequency: 50.000 MHz
 - bin size: 1.000 kHz
 - interval: 10 s
 - crop: 20 %

satfan52, 2018-04-14:
Dear Pavel,

Thanks for your reply. I understand your point about cropping (20%) and setting a larger interval (10s). However it does not solve the issue. I have tried the very same settings that you did (see pictures) and in my ubuntu 16.04 environment, the spectra produced displays additional non-existing spurs  at roughly 29 Mhz and 42 Mhz. Those spurs are not minor ones, they are of the same power than the existing real ignals at 28Mhz at 41 Mhz!. It just impossible to use soapy_power and the SDR transceiver to make some reasonably accurate spectra measurements. 

Please have at the picture below that compare the spectra obtained using osmocom-fft and soapy_power...
 
[image]
[image]
[image]
   
Best Regards
Peter

pavel-demin, 2018-04-14:
The problem of multiple peaks can be solved by adding `--tune-delay 0.1` to the 'Additional parameters' field in the 'Settings' window.

pavel-demin, 2018-04-14:
Here is a screenshot showing my settings:
[image]

satfan52, 2018-04-14:
Yes !!! 
This solves the issue.
I would have never thought about changing this parameter!!!
Thanks :-)

closed 2018-04-14 by satfan52