pavel-demin/red-pitaya-notes #699: SDR tranciever 10m extension

fransnissen, 2018-04-30:
### Description of the setup:

 * Device: Hamlab (redpitaya sdr)
 * SD card image:
 * Application: SDR Tranciever
 * Other relevant information:

### Description of the problem:
I'm working HF with my hamlab (redpitaya sdr) . the nexst project i'm working on is to use the hamlab sdr with a 2m transverter. the transverter is working from 26- 30 Mhz (10 and 11m). actualy the TX on the redpitaya software is only 28-30Mhz. 
so my question, is it possible to extend the TX capability, specialy for the 10m band.

regard,

Frans Nissen

### Steps to reproduce the problem:

 1.
 2.
 3.

pavel-demin, 2018-04-30:
> so my question, is it possible to extend the TX capability, specialy for the 10m band.

I think that it should be possible. PowerSDR mRX PS has various transverter related settings in the XVTRs menu.

dish-eme-sdr, 2018-05-01:
Bands and band limits are a function of the software used with RedPitaya.

In my case, I am using PowerSDR mRX PS with RedPitaya.

In the first setup screen (General), enter Extended in the (Region box) and the RedPitaya will transmit anywhere it is capable of receiving.

In another part of the program, it is possible to enter the offset of the transverter.  I have been able to program mine so that 26 MHz reads 144 Mhz on the display and 30 Mhz reads 148 MHz so no mental arithmetic is needed to figure out what frequency the system is tuned to.

I do the same using 432, 902, 1296, 2304, and 10 GHz transverters.

fransnissen, 2018-05-03:
Thank for this info, exact what i need.

closed 2018-05-06 by pavel-demin