pavel-demin/red-pitaya-notes #708: 2e TX output on redpitaya

fransnissen, 2018-05-22:
### Description of the setup:

 * Device: Hamlab / Stemlab sdr
 * SD card image: 
 * Application: Hpsdr / hamlabsdr
 * Other relevant information:

### Description of the problem:
I'm using the redpitaya for HF, and built myself a preRX filter bank for every hamband and switche it by i2c. 
My nexst project is to use the Hamlab / redpitaya for my 2m/70cm to 28mhz transverter. actualy using the 1 rx and tx output is a bit complicated, i have to bypass the TX LPF and TX BPF. 
it would be easy when i could use for RX and specialy TX the 2e output on the Redpitaya.

I know it is for now not possible to use the second TX output. my question is would in in the near future possible in the software? it would help me a lot.

Kind regard,
Frans Nissen

### Steps to reproduce the problem:

 1.
 2.
 3.

pavel-demin, 2018-05-22:
Hello Frans,

Currently, OUT2 outputs the envelope signal that can be used with the envelope elimination and restoration (EER) power amplifiers.

I can try to add some logic that would allow to select what signal goes to what output.

Best regards,

Pavel

fransnissen, 2018-05-22:
That would help me a second TX output on these pedpitayas sdr's

regard

Frans

dish-eme-sdr, 2018-05-22:
Maybe a bit off topic, but you mentioned something about the function of OUT2 that I did not know:

=
"Currently, OUT2 outputs the envelope signal that can be used with the envelope elimination and restoration (EER) power amplifiers."
=

Someplace along the way I completely missed that OUT2 outputs the envelope signal when in EER mode.  Now that I know this I may give an EER amp a try!

Currently I am running a conventional linear amplifier chain and using IN2 (RX2) for the adaptive predistortion feedback signal (PURESIGNAL in PowerSDR OpenHPSDR MRX) and it works GREAT thanks to your clever coding on the RedPitaya.

I am curious when in EER mode and using OUT2 for the Envelope signal, can I still use RX2 for the PURESIGNAL input port or is that disabled in EER mode??? 

And:

Is there a way to add a software controlled step attenuator to the Rx input / feedback path similar to what Anan does?  I have seen mention of some kind of add on board that can be used with RedPitaya but have not been able to nail the source and info down.

Feel free to move or delete this post if needed as I realize it's not 100% on topic but is somewhat related as it does relate to the function of OUT2 and the software controlled attenuator is something that most certainly could come in handy to set system gain when connecting outboard transverters to the RedPitaya.  As it is now I have to use external pads to set the gain.

ka6s, 2018-05-22:
Look up Charlie 25;

pavel-demin, 2018-05-22:
I've added some logic that allows to select what signal goes to what output with [this commit](https://github.com/pavel-demin/red-pitaya-notes/commit/77ca831432fd5f5d0f601fd4052f17e60bd49d7c).

Now, the sdr-transceiver-hpsdr program expects five command line arguments:
 - the first three arguments are for the inputs/receivers;
 - the last two arguments are for the outputs (OUT1 and OUT2), where 1 corresponds to the TX signal and 2 corresponds to the envelope signal.

To send the TX signal to OUT2, [the last line is start.sh](https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/sdr_transceiver_hpsdr/app/start.sh#L9) should be edited and the last argument should be set to 1:
```
sdr-transceiver-hpsdr 1 2 2 1 1 &
```

pavel-demin, 2018-05-22:
> I am curious when in EER mode and using OUT2 for the Envelope signal, can I still use RX2 for the PURESIGNAL input port or is that disabled in EER mode???

The envelope signal is independent of the PureSignal algorithm. Both can be enabled simultaneously. However, it's not impossible that there could be some side effects due to crosstalk that could degrade the performance of the PureSignal algorithm. It's to be tested.

> Is there a way to add a software controlled step attenuator to the Rx input / feedback path similar to what Anan does?

If you have a controllable step attenuator, then it's most probably possible to interface it with the Red Pitaya board and with the SDR application.

Please open a new issue if you want to continue the discussion about the step attenuator control.

dish-eme-sdr, 2018-05-22:
"sdr-transceiver-hpsdr 1 2 2 1 1 &"

THANK YOU !!!

satfan52, 2018-05-23:
Hello, I confirm both the port selections would be very useful, it would  be great if this settings could be controlled not only at the start of the program but also when the transceiver is running for instance via specific band codes using Hermes bit 1 to 4 or alternatively through some other mechanism that Could be controlled via a customised button in QUISK

Regarding the control of a digital attenuators, I currently use Hermes bit 5, 6, 7 with a customised button in QUISK for that purpose but that only provides 8 attenuation steps, actually 6 only considering that the activation of the LNA and the external narrow bandwidth Filter must also be controlled via those 3 Hermes bits. Is there a way to control the digital attenuators via the SPI interface so I can exploit all 64 attenuation steps of my 7 usd PE4302 digital attenuators (6 bits ) and if yes how to do this practically ? I would like to use a customised button in Quisk. 

Regards 
Peter

pavel-demin, 2018-05-23:
The logic that allows to select what signal goes to what output is now in [the latest release](https://github.com/pavel-demin/red-pitaya-notes/releases/tag/20180523).

@dish-eme-sdr & @satfan52 - Please [open a new issue](https://github.com/pavel-demin/red-pitaya-notes/issues/new) if you want to continue the discussion about the attenuators. I don't think that GitHub provides a way to move comments from one issue to another. So, please copy them yourself.

Closing this issue.

closed 2018-05-23 by pavel-demin