pavel-demin/red-pitaya-notes #712: How to digitally control external devices via the GUI through the HPSDR compatible SDR transceiver  HPSDR ?

satfan52, 2018-05-23:
Hello Again,

1) Regarding the selection of the Output Port 

a) Thanks for giving the possibility to control the output port at the time the compatible HPSDR transceiver is started, that is quite nice. I would welcome the exact syntax of the command to use for that purpose. Also for the reception ports by the way.

b) For optimal transverter support,  it should also be possible to select "output port 2" instead of "output 1" depending on the "band selected" by the user via the GUI or alternatively  by pressing a customized "Switch" button in the GUI (QUISK in my case).  I also would like to avoid for obvious reasons to  have to start and stop the application every time the desired output port must be changed.  

2) Regarding the control of Digital attenuators
 
 I currently use Hermes bit 5, 6, 7 with a customised button in QUISK button for that purpose but that only provides 8 attenuation steps, actually 6 only considering that the activation of the LNA and of  the external narrow bandwidth bandpass filter must also be controlled via those 3 Hermes bits. 

My attenuator is a PE4302  that is controlable over a dynamic range of -31.5 dB using 64 steps of 0.5 dB (so 6 bits) and I would like to control it from the GUI too using a custom QUISK button just like for the Output port as explained in 1b above. Of course the desired attenuation setting (ie, the 6 control bits) would have to be communicated to the PE4302 attenuator via the SPI serial interface.  I would like to be able to set the  6bits attenuation level from a customised button in QUISK or whatever already existing user controlable settings in PowerSDR.

Maybe it is already possible to do that with the current HPSDR transceiver code but I do not know what are the parameters to use in QUISK  and how to set-up the SPI serial communication with the PE4302 so that that the 6 control bits are transfered appropriatelly.

Please note that the issue might be more complex that it seem at first, because of course, different attenuators can be potentially used on INPUT port 1 and INPUT Port 2 and because there are even situation where it is desirable to control an external attenuator or a Variable Gain Amplifier in the OUTPUT path on output port 1 (or 2 ;-)  ) 

3) Regarding other type of operational information which is pertinent to transfer from to the red-pitaya and the GUI to an external digital device in a serial way via the SPI interface.

At first I am thinking of the "band" selected on receiver 2 and 3 as well as the frequency of reception on receiver 1,2,3 and the frequency of transmission on output port 1 (or 2 ;) )   Why ? These info are useful for building a transverter (to set the frequency of a digital LO and to select a given type of front end filter) . It is also usefull to select different bandpass filters,  antennas or transverters  to be used on input port 1, input port 2, output port 1 and output port 2.  For instance it is nice to TX/RX on 40m on using an antenna connected to input port 1 and output port 1 and to have the possibility to TX/RX on 2m using a transverter connected to input port 2 and output port 2.  

For LEO satellite operation I would like to be able to set a button with the type of satellite cross band operation required, implying that for a specific type of cross band satellite operation I must receive on input port X and transmit on ouput port Y and for another type of satellite cross band of operation it might be another combination. The type of cross band operation is typically a GUI info to be set by the user via a custom button in QUISK or some already existing user controlable settings in PowerSDR.   

Here the central question  is how to get such basic operational info from  redpitaya and the GUI and then to have it transfered serially via SPI to to the device of interest to control or more realistically to a micro-controler like an Arduino that will then take catre of switching the right devices based on the info received from the redpitaya and the GUI.

My apologies for these rather basic questions but I miss that part of the puzzle.

Thanks
 Regards
 Peter

pavel-demin, 2018-05-23:
> 1.a. For optimal transverter support, it should also be possible to select "output port 2" instead of "output 1" depending on the "band selected" by the user via the GUI or alternatively by pressing a customized "Switch" button in the GUI (QUISK in my case).

I don't think that this functionality is supported by the HPSDR protocol.

> Regarding the control of Digital attenuators.
> My attenuator is a PE4302 that is controlable over a dynamic range of -31.5 dB using 64 steps of 0.5 dB (so 6 bits).

Here is a link to a post by Peter DC2PD showing a couple of pictures of his attenuator based on PE43711 and PCA9555:
http://forum.cq-nrw.de/viewtopic.php?t=141#p1504

PCA9555 can be easily connected to the Red Pitaya board via its I2C interface. [sdr-transceiver-hpsdr.c](https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/sdr_transceiver_hpsdr/server/sdr-transceiver-hpsdr.c) contains several functions communicating with the PCA9555 chips.

> Here the central question is how to get such basic operational info from redpitaya and the GUI and then to have it transfered serially via SPI to the target device to control (or more realistically if there are several devices to control, to a micro-controler like an Arduino that will then take care of switching the right devices based on the info received from the redpitaya and the GUI).

I'm afraid that I can't answer this question.

pavel-demin, 2018-05-24:
> I would welcome the exact syntax of the command to use for that purpose. Also for the reception ports by the way.

I've just added a description of the new command line arguments to my notes about the SDR transceiver compatible with HPSDR:

http://pavel-demin.github.io/red-pitaya-notes/sdr-transceiver-hpsdr/#configuring-inputs-and-outputs

ka6s, 2018-05-30:
> In the example you give, your are mapping the output signal to the two
> OUTPUT ports at the same time, this not a typo, right ?
>
> Right.
>
> a) Why is it easier to use a PCA9555 compared to using an Arduino to
> communicate serially with the redpiraya for instance ?
>
> In my opinion, PCA9555 is easier to use compared to Arduino because it
> doesn't require programming on the PCA9555 side.
>
> b) When I look at the content of the existing sdr-transceiver-hpsdr.c file
> I think (?) I see that some input/output pins of the PCA9555 are set based
> on variables or functtions (?) such as rx_freq1, rx_freq2, tx_freq, etc..,
> Right ? My understanding is that these variables (or are they functions ?)
> are actually the receiving frequency of the 3 different receivers and the
> tx frequency. Right (?) or Wrong ? I am striving to understand the nature
> of those functions
>
> I've added a table showing the signals sent to the PCA9555 chip at address
> 3 (0x23):
>
> http://pavel-demin.github.io/red-pitaya-notes/sdr-transceiver-hpsdr/#i2c-connections
>
> Most of the signals sent to the PCA9555 chips at addresses 0, 1 and 3 are
> [... 21 lines cut ...]
> directly via the GPIO pins. This way, you don't need to modify Arduino.
>
> For example, here is a link to a video showing Quisk running on ASUS
> Tinker Board <https://www.amazon.de/dp/B01N35PQ9U>:
> https://www.youtube.com/watch?v=U6u-rPjQ3bs

Just to add to Pavel's suggestions. See GitHub.com/goorx/pihpsdr which is
repository containing a complete gui for hpsdr with buttons and knobs ala
PowerSDR. Several folks have replaced the raspberry pi with the
tinkerboard..it is a very successful combo.

satfan52, 2018-05-31:
Dear Pavel (and Ka6s),

Thank you for this very comprehensive overview, things are MUCH clearer for me now.

Consolidating the info about all possible (existing) options is actually very useful for me and I hope it will be for others too.  I still have some questions of course, but I will probably be able to answer them myself as I will testing some of the solutions which are referred here.  

Thank you !
Regards

pavel-demin, 2018-06-03:
> On the other hand I have no idea if programming on the ASUSTINKER board (for instance) is easier or more complex than programming on the Arduino.

I'd say that programming a small single board computer (Tinker Board, Raspberry Pi, etc.) running Linux doesn't differ much from programming a desktop or a laptop running Linux. Basically, it's just programming a computer running Linux. It's possible to connect to it via SSH or directly plug a screen, a keyboard and a mouse. Then it's possible to do all the code development directly on the computer. In case of Quisk,, the GPIO pins can be controlled directly from the Python code via the [GPIO library](https://github.com/TinkerBoard/gpio_lib_python).

satfan52, 2018-06-04:
Ok got the point. This GPIO  Python library  could be used from within the QUISK hardware description file to activate specific GPIO pins following actions defined for customised buttons of the QUISK UI interface . That seems indeed like a level of complexity that I could manage. Not sure though why it makes more sense to use the AsusTînker board vs  RawsberyPI as this library  also runs on the RawsberyPI.

satfan52, 2018-06-06:
Dear Take,

Thank you for this info. I own the very same model of attenuator than the one displayed on your pictures, so quiet usefull!

Regards
Peter
ON7YI

JS1MAV, 2018-06-06:
Peter,

I am not using QUISK anymore and I do not know S-ATT of HPSDR will be able to apply for QUISK. My motivation to add S-ATT was to enable auto calibration function of HPSDR pure-signal and it works.

My next desire is to control output level by another PE4302 but SPI on E2 is currently limited to control one chip. This seems to be a common issue of RedPitaya and I hope somebody will provide the solution.  I personally prefer SPI interface than IC2.

Regards,

take

de JA5AEA

closed 2018-06-08 by satfan52