pavel-demin/red-pitaya-notes #718: Reg Development machine

falcon98, 2018-06-29:
Dear Pavel,

I recently completed some projects from Anton Potonik's blog on Red Pitaya and have developed some basic understanding. For Anton's project my dev machine had Ubuntu with Vivado and from this machine I used to transfer and execute bit files on my RP through SSH. Also later I am able to execute projects from Windows directly.
I just wanted to know if I can continue with the same development machine or do I need to install Debian version as mentioned by you in notes. 
Can I generate project from your tcl files and transfer bit files to RP with my present Ubuntu/Windows machine and execute them or it is really important to set up debian based system?

Thanks for your support.

Regards,

pavel-demin, 2018-06-29:
Ubuntu should be perfectly fine. I have a machine running Ubuntu 18.04 and I can build all my Red Pitaya projects without any problem.

The Debian based system is not a strict requirement. My instructions are just an example of how a minimal development machine could be configured.

The [Pre-requirements section](http://pavel-demin.github.io/red-pitaya-notes/led-blinker/#pre-requirements)  in my LED blinker notes contains a list of packages that are required for some of my scripts. These packages can be installed on both Debian and Ubuntu.

falcon98, 2018-06-29:
Hi Pavel,
Thanks a ton for your quick reply , I will try the led blinker project in Ubuntu and also update here for future reference.

Thanks

falcon98, 2018-06-30:
Hi Pavel,

As expected, the led blinker worked fine on Ubuntu system.
Now I will try working on ADC_test code and execute server program on my RP board and client program on Ubuntu host system.
Thanks

closed 2018-06-30 by falcon98

BeauWebber, 2018-06-30:
Hi All,
Is this fast ADC or the slow ADC on E2 ?
I am very interested in advice re getting the slow DAC to work with Pavel’s .bit code.
Cheers,
                Beau Webber

-- Kentforlife.net - the email service for alumni of the University of Kent

pavel-demin, 2018-06-30:
Hi Beau,

I think that outputting constant levels to the PWM pins can be done with an additional cfg register and a sigma-delta modulator. The code for a sigma-delta modulator can be found at [this link](http://www.fpga4fun.com/PWM_DAC_2.html). However, I'm afraid that it'll bring a lot of noise.

If you're interested, I can add this functionality to the NMR configuration. 

Cheers,

Pavel

BeauWebber, 2018-06-30:
Hi Pavel,
Yes if you could add that and I will try and see how it works.
I have a huge amount of filtering on that pin, as I only need constant levels.
I use it to just set the variable gain on the NMR receiver.
Cheers,
                Beau
PS I am in holiday in France at the moment, for another week, so no hurry.