pavel-demin/red-pitaya-notes #721: Red Pitaya with GnuRadio

w2naf, 2018-08-19:
### Description of the setup:

 * Device: Red Pitaya 125-14
 * SD card image: red-pitaya-alpine-3.6-armhf-20180523
 * Application: sdr-receiver-hpsdr
 * Other relevant information:

### Description of the problem:

Hi, Pavel.

My goal is to set up a Red Pitaya to work with GnuRadio and MIT's DigitalRF as a multi-band HF receiver that will save time-stamped IQ files for later analysis. However, when I try running, red-pitaya-notes/projects/sdr_transceiver/gnuradio/trx_am.grc, I get "socket.error: [Errno 111] Connection refused". I've verified that the IP address is correctly set to the Red Pitaya an the port is set to 1001. I've tested the receiver with SkimmerServer on a Windows computer, and it works fine. Do you have any suggestions?

Also, do you think it is possible to run GnuRadio and DigitalRF all on the Red Pitaya without a remote computer?

Thanks for your help,
Nathaniel W2NAF

w2naf, 2018-08-19:
Hi again,

I see some work regarding running everything on the red pitaya is referenced in #686 . Do you know what the status is on this?

Thanks again,
Nathaniel

w2naf, 2018-08-20:
Hi yet again, Pavel,
I made some progress... I figured out that the GnuRadio script works with the sdr_transceiver, not sdr-receiver-hpsdr. Now that I have that worked out, do you have any comments/suggestions on using the multiband receiver in the manner that I described above?

Thanks!

~Nathaniel W2NAF

pavel-demin, 2018-08-20:
Hi Nathaniel,

If you need more than two receivers available in the sdr_transceiver configuration, then you can try to use 
sdr_receiver_hpsdr with the [hpsdr_hermes GNU Radio blocks](https://github.com/Tom-McDermott/gr-hpsdr).

> My goal is to set up a Red Pitaya to work with GnuRadio and MIT's DigitalRF as a multi-band HF receiver that will save time-stamped IQ files for later analysis.
> Also, do you think it is possible to run GnuRadio and DigitalRF all on the Red Pitaya without a remote computer?

I'd say that you'll still need a remote computer to store the output HDF5 files. Moreover, I think that a more efficient solution would be to write a small C program based on [sdr-transceiver.c](https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/sdr_transceiver/server/sdr-transceiver.c) or [sdr-receiver-hpsdr.c](https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/sdr_receiver_hpsdr/server/sdr-receiver-hpsdr.c) that would receive the I/Q samples from the FPGA and would write them to the HDF5 files using the [Digital RF C library](https://github.com/MITHaystack/digital_rf/tree/master/c).

Could you please provide more details about your storage requirements? How many bands are you planning to record? At what sample rate?

Best regards,

Pavel

w2naf, 2018-08-20:
Hi Pavel,
Thanks for your help. I'll take a look at the hpsdr receiver option. I agree that bypassing GNU Radio entirely would probably be more efficient and more desirable. I only have minimal experience working with the C code, so I will probably need some help doing that.

I would like to record as many HF slices as possible, probably 6 to 8 192-kHz slices. If I just have 6 bands, I would do the ham bands at 1.8, 3.5, 7, 14, 21, and 28 MHz. I will likely have between 512 GB to 2 TB available, depending on the hardware I choose, and I would like the recorded to be able to run unattended for up to 2 weeks.

Thank you,
Nathaniel

w2naf, 2018-08-21:
Hi Pavel,

I just made plans to visit the DigitalRF authors at MIT Haystack in a couple of weeks to work on this with them in person. I'll let you know how it goes. Thanks!

~Nathaniel

jacintomfr, 2018-08-24:
Hi Pavel, 

I'm using Red Pitaya with gnuradio and it works very well, any chance to have control for pre-amplifier + attenuator on GPIO pins (like ATT and PRE on red pitaya source as 0/1 select) ? for me is enough and I already have a good grc (modified from the W7FU original grc file) see screenshot, I really don't want to change to hpsdr grc block because I'm happy what I have now, BPF is not even needed as I have a very RF silent zone, just a FM notch filter and LPF is automatic. Thanks.

Jacinto

[image]

pavel-demin, 2018-08-24:
Hi Jacinto,

Your GRC setup looks great!

I'd prefer not to modify the FPGA configuration of the sdr_transceiver application.

I'd propose to create a separate GRC block that would control a [MCP23017](https://www.microchip.com/wwwproducts/en/MCP23017) chip connected to the I2C pins of the [extension connector E2](http://redpitaya.readthedocs.io/en/latest/developerGuide/125-14/extent.html#extension-connector-e2).

Another possibility is to use a [FT232H](http://www.ftdichip.com/Products/ICs/FT232H.htm) chip connected via USB to your computer. Here is an example of a [readily available module](https://learn.adafruit.com/adafruit-ft232h-breakout). Looking at the [GPIO example](https://learn.adafruit.com/adafruit-ft232h-breakout/gpio), I'd say that it should be very easy to interface with GRC.

What do you think?

Best regards,

Pavel

jacintomfr, 2018-08-24:
Hi Pavel, 

Until two months ago I didn't know anything of gnuradio, I only modify the grc file to support red pitaya and this was after try many times, but creating a grc block I really don't know where to start ... and python is difficult for me. 
Anyway, the way it is is great and I love it! thank you for supporting gnuradio on red pitaya ... for the moment att and pre will be activated by switch's, but I will study the two possibilities you described, maybe there is already grc module (to install easy on gnuradio) for FT232H, will search internet.

Jacinto

jacintomfr, 2018-08-24:
Hi Pavel, 

I just remember that I have a module with one PCA9555 + 2 ULN2803A ready to work, tested with red pitaya and hpsdr protocol, for me and you (as you already worked with this I2C chip), do you think is more easy to create a grc block then the FT232H USB board ?

Jacinto

pavel-demin, 2018-08-24:
> I just remember that I have a module with one PCA9555 + 2 ULN2803A ready to work, tested with red pitaya and hpsdr protocol, for me and you (as you already worked with this I2C chip), do you think is more easy to create a grc block then the FT232H USB board?

Yes, it would be easier to create a grc block for PCA9555.

jacintomfr, 2018-08-24:
That's good news, 

so, when you have time, I will appreciate a little help how to create the grc block, please ...

Jacinto

chrisdebian, 2025-06-22:
Does this issue still need to be open?

Chris

closed 2025-06-22 by pavel-demin