pavel-demin/red-pitaya-notes #722: Running on Windows 10

dnhkng, 2018-08-20:
I found a way to run this on a Windows 10 development machine.

You have to first build the cores, and then the project. Then you can open the project and generate the bitstream as usual. 

**a) Open Vivado, and in the TCL console, paste this code:**

_cd c:/#your-path-to-the-code#/red-pitaya-notes-master

set argv [list axi_axis_reader_v1_0 xc7z010clg400-1]
source scripts/core.tcl
set argv [list axi_axis_writer_v1_0 xc7z010clg400-1]
source scripts/core.tcl
set argv [list axi_bram_reader_v1_0 xc7z010clg400-1]
source scripts/core.tcl
set argv [list axi_bram_writer_v1_0 xc7z010clg400-1]
source scripts/core.tcl
set argv [list axi_cfg_register_v1_0 xc7z010clg400-1]
source scripts/core.tcl
set argv [list axi_sts_register_v1_0 xc7z010clg400-1]
source scripts/core.tcl
set argv [list axis_accumulator_v1_0 xc7z010clg400-1]
source scripts/core.tcl
set argv [list axis_adder_v1_0 xc7z010clg400-1]
source scripts/core.tcl
set argv [list axis_alex_v1_0 xc7z010clg400-1]
source scripts/core.tcl
set argv [list axis_averager_v1_0 xc7z010clg400-1]
source scripts/core.tcl
set argv [list axis_bram_reader_v1_0 xc7z010clg400-1]
source scripts/core.tcl
set argv [list axis_bram_writer_v1_0 xc7z010clg400-1]
source scripts/core.tcl
set argv [list axis_constant_v1_0 xc7z010clg400-1]
source scripts/core.tcl
set argv [list axis_counter_v1_0 xc7z010clg400-1]
source scripts/core.tcl
set argv [list axis_decimator_v1_0 xc7z010clg400-1]
source scripts/core.tcl
set argv [list axis_fifo_v1_0 xc7z010clg400-1]
source scripts/core.tcl
set argv [list axis_gate_controller_v1_0 xc7z010clg400-1]
source scripts/core.tcl
set argv [list axis_gpio_reader_v1_0 xc7z010clg400-1]
source scripts/core.tcl
set argv [list axis_histogram_v1_0 xc7z010clg400-1]
source scripts/core.tcl
set argv [list axis_i2s_v1_0 xc7z010clg400-1]
source scripts/core.tcl
set argv [list axis_iir_filter_v1_0 xc7z010clg400-1]
source scripts/core.tcl
set argv [list axis_interpolator_v1_0 xc7z010clg400-1]
source scripts/core.tcl
set argv [list axis_keyer_v1_0 xc7z010clg400-1]
source scripts/core.tcl
set argv [list axis_lfsr_v1_0 xc7z010clg400-1]
source scripts/core.tcl
set argv [list axis_negator_v1_0 xc7z010clg400-1]
source scripts/core.tcl
set argv [list axis_oscilloscope_v1_0 xc7z010clg400-1]
source scripts/core.tcl
set argv [list axis_packetizer_v1_0 xc7z010clg400-1]
source scripts/core.tcl
set argv [list axis_phase_generator_v1_0 xc7z010clg400-1]
source scripts/core.tcl
set argv [list axis_pps_counter_v1_0 xc7z010clg400-1]
source scripts/core.tcl
set argv [list axis_pulse_generator_v1_0 xc7z010clg400-1]
source scripts/core.tcl
set argv [list axis_pulse_height_analyzer_v1_0 xc7z010clg400-1]
source scripts/core.tcl
set argv [list axis_ram_writer_v1_0 xc7z010clg400-1]
source scripts/core.tcl
set argv [list axis_red_pitaya_adc_v2_0 xc7z010clg400-1]
source scripts/core.tcl
set argv [list axis_red_pitaya_dac_v1_0 xc7z010clg400-1]
source scripts/core.tcl
set argv [list axis_stepper_v1_0 xc7z010clg400-1]
source scripts/core.tcl
set argv [list axis_tagger_v1_0 xc7z010clg400-1]
source scripts/core.tcl
set argv [list axis_timer_v1_0 xc7z010clg400-1]
source scripts/core.tcl
set argv [list axis_trigger_v1_0 xc7z010clg400-1]
source scripts/core.tcl
set argv [list axis_validator_v1_0 xc7z010clg400-1]
source scripts/core.tcl
set argv [list axis_variable_v1_0 xc7z010clg400-1]
source scripts/core.tcl
set argv [list axis_variant_v1_0 xc7z010clg400-1]
source scripts/core.tcl
set argv [list axis_zeroer_v1_0 xc7z010clg400-1]
source scripts/core.tcl
set argv [list dna_reader_v1_0 xc7z010clg400-1]
source scripts/core.tcl
set argv [list gpio_debouncer_v1_0 xc7z010clg400-1]
source scripts/core.tcl
set argv [list port_selector_v1_0 xc7z010clg400-1]
source scripts/core.tcl
set argv [list port_slicer_v1_0 xc7z010clg400-1]
source scripts/core.tcl
set argv [list pulse_generator_v1_0 xc7z010clg400-1]
source scripts/core.tcl
set argv [list shift_register_v1_0 xc7z010clg400-1]
source scripts/core.tcl_

This will generate the cores in the tmp directory. It takes a while, grab a coffee.

**b) Edit the /scripts/project.tcl file:**
find:
"set_property IP_REPO_PATHS tmp/cores [current_project]"
and add directly below it:
"update_ip_catalog"

**c) Build the project**
set argv [list led_blinker xc7z010clg400-1]
source scripts/project.tcl

That should save creating a virtual machine.

BeauWebber, 2018-08-20:
Thanks David, that looks good, I will give it a try sometime. Cheers, Beau

-- Kentforlife.net - the email service for alumni of the University of Kent

pavel-demin, 2018-08-21:
Hi David,

Thanks for sharing this idea!

I've just added a script that builds all the cores and added `update_ip_catalog` to [project.tcl](https://github.com/pavel-demin/red-pitaya-notes/blob/master/scripts/project.tcl) with [this commit](https://github.com/pavel-demin/red-pitaya-notes/commit/d0a5764b6ce600a6c95162314f52021bdba6c4ff).

Here is an updated version of your instructions:

**a) Open Vivado, and in the TCL console, paste this code:**
```
cd c:/#your-path-to-the-code#/red-pitaya-notes-master
source helpers/build-cores.tcl
```

This will generate the cores in the tmp directory. It takes a while, grab a coffee.

**b) Build the project**
```
set argv [list led_blinker xc7z010clg400-1]
source scripts/project.tcl
```

Best regards,

Pavel

closed 2019-05-14 by pavel-demin