pavel-demin/red-pitaya-notes #723: Other SBC than Red Pitaya possible ?

hanslammerts, 2018-08-20:
Hi Pavel,
This is not a issue, but more a question.
I don't own a RB, but I'm still following this project with great interest. I'd like to have something similar but then for a small X86 based, 64 bit computer running Linux, and connected to (in my case) a SDRPlay receiver, or even an RTL-SDR dongle.
Just a very short question:

Do you think this can be done, or would one really need the special features of a Red Pitaya ?
I was able to compile ft8d on Debian 9 on this very x86_64 machine, but I'm unsure how to proceed.

Thanks,
Hans PA3CRZ

pavel-demin, 2018-08-21:
Hi Hans,

I'm not sure if I understand what you're planning to achieve. Are you planning to use a SDR receiver together with a FT8 decoder? If so, then why not use the complete WSJT-X software? Looks like it runs even on Raspberry Pi3:
https://www.rtl-sdr.com/using-a-raspberry-pi-3-and-rtl-sdr-as-a-40m-ft8-jt65-jt9-monitor/

Best regards,

Pavel

hanslammerts, 2018-08-21:
Hi Pavel,

I'm sorry. I wasn't clear enough.
What I'm trying to do is to have a small computer (x86, Debian), to which a SDRplay receiver is connected, to run some kind of FT8 decoder, but without the graphical interface. This may seem silly, but all I want to do is to receive signals continuously, maybe even change the receiving frequency of the SDRplay remotely, decode FT8, and send the decoded messages over UDP (or TCP) to my network. That way I think I could be able to run i.e. GridTracker on my wifi laptop whereas receiving and decoding is being done elsewhere.

Maybe I'm completely wrong, but I imagined your software, in a modified form, could be what I'm looking for.
Since the FT8 frequencies are fixed, and I only want to receive (for now) I do not need the entire graphical package that WSJT-X is.
Then again, I could be barking up the completely wrong tree. Hope you can enlighten me a bit.

Thanks,
Hans

pavel-demin, 2018-08-22:
Thanks for the clarification.

I think that it should be possible to record the I/Q samples with [the I/Q recorder for SDRplay](https://github.com/SDRplay/examples) and when to decoded them with the FT8 decoder. Maybe you'll also need [the CSDR tools](https://github.com/simonyiszk/csdr) for the sample rate conversion.

closed 2018-10-05 by pavel-demin