pavel-demin/red-pitaya-notes #724: VNA Cnnect

Rolf-L, 2018-08-21:
### Description of the setup:

 * Device:
 * SD card image:
 * Application:
 * Other relevant information:

### Description of the problem:

### Steps to reproduce the problem:

 1.
 2.
 3.

Rolf-L, 2018-08-21:
How can I prevent the signal DIO0 from being set to "1" when I connect the VNA program? This generates the PTT signal in my TRX and turns on the connected PA. Only the bias current flows, but it is about 4A and heats up the PA after a short time. Can I prevent this by a further command line in the SD Card image file latest version and, if so, what are the commands?

Rolf-L, 2018-08-21:
sorry, I dont want to close it

pavel-demin, 2018-08-21:
The easiest solution is still to edit [line 566 in vna.py](https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/vna/client/vna.py#L566) and to replace `self.set_gpio(1)` with `self.set_gpio(0)`.

In the latest GUI version, there is no place left for additional control elements. So, I don't know how to add a GUI element for controlling the GPIO pins.

pavel-demin, 2018-08-22:
> If I understand that correctly, I'll have to do the correction in the source code and recompile,

The recompilation isn't required. One of the ideas behind the VNA client is that it should be very easy to modify. The code is in [vna.py](https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/vna/client/vna.py) and the GUI is in [vna.ui](https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/vna/client/vna.ui). Both files can be edited.

So, the following steps should be enough;
- open [vna.py](https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/vna/client/vna.py#L566) in a good text editor like for example [Notepad++](https://notepad-plus-plus.org/download/);
- go to the line 566;
- replace `1` with `0`;
- save the file.

pavel-demin, 2018-08-22:
> I can only find vna.py on GitHub, but not on the SDCard in the vna folder.

The `vna.py` file is on your computer in the folder from where your start the the `vna.exe` program.

Rolf-L, 2018-08-22:
Hi Pavel,
If you know how to do it, it's really easy :-)
Thanks again for your quick help.

best regards and have a nice day
Rolf

chrisdebian, 2025-06-22:
@Rolf-L Hi, Rolf; is this issue still valid, does it need to stay open?

Chris