pavel-demin/red-pitaya-notes #726: mcpha installation problem

jankon0, 2018-08-22:
Dear Pavel,

I have trouble getting the mcpha application to run on my Redpitaya. I had a original Redpitaya image on the SD card and copied your zip-file to the FAT32 partition. The mcpha control program however did not connect to the Redpitaya (error cannot connect), and also the webinterface did not show your programs. Is it possible to use the zipfile to run your applications on the original Redpitaya Linux? If yes, which files do I need to copy, all of them or just the /apps folders?

I also tried installing the image with Alpine linux you provide, but there my problem is that the size of the FAT32 partition is only 11 MB, which is not enough to copy the applications from your zip-file onto it. Somehow resizing the partition doesn´t work:

root@red-pitaya:~# resize2fs /dev/mmcblk0p1
resize2fs 1.42.12 (29-Aug-2014)
resize2fs: Bad magic number in super-block while trying to open /dev/mmcblk0p1
Couldn't find valid filesystem superblock.

I do not understand how the small partition size is supposed to contain all the apps you wrote, or are they stored on the Extv4 partition?

Thanks a lot for your help!

Best regards,

Janko

Rolf-L, 2018-08-22:
Hi Janko,
I had the same problem with Alpine version.
You have to format the SDCard to use the full size or take a new formated SDCard.
I hope this would help :-)

best regards
Rolf

pavel-demin, 2018-08-22:
> I had a original Redpitaya image on the SD card and copied your zip-file to the FAT32 partition.

If you have a original Red Pitaya image on the SD card, then it's possible to install and run the  MCPHA application via the web interface without manually unzipping anything. Just run the Application marketplace application, then select and install the MCPHA application. After that the MCPHA application should appear on the main page of the Red Pitaya web interface.

pavel-demin, 2018-08-22:
> I do not understand how the small partition size is supposed to contain all the apps you wrote, or are they stored on the Extv4 partition?

Normally, the Alpine Linux image should be copied to an empty FAT formatted SD card. I think that any freshly unpacked SD card should work out of the box.

jankon0, 2018-08-22:
Thanks for your quick reply! I managed to install the app via the market place and it is working fine now.

closed 2018-08-22 by jankon0