pavel-demin/red-pitaya-notes #732: About development environment requests！

cleverlau, 2018-09-14:
Thanks for taking the time to open an issue! 
Dear parvel,
            I recently came into contact with redpitaya, and there are still many doubts here. If I want to use redpiataya to develop my own things, which development environment do use? Which environment are you using?
Thank you very much for your attention.

pavel-demin, 2018-09-14:
> Which environment are you using?

I'm using Xilinx Vivado Design Suite 2018.1 and I'm going to switch to 2018.2.

More details about my environment can be found at the following links:

http://pavel-demin.github.io/red-pitaya-notes/development-machine/
http://pavel-demin.github.io/red-pitaya-notes/led-blinker/
https://www.dropbox.com/sh/5fy49wae6xwxa8a/AACl--BhQvcNgjeQLRaiX9dha/ClubVivado2016_Pavel_Demin.pdf?dl=1

cleverlau, 2018-09-15:
First of all, thank you very much for your answer.
 I am a Chinese student. I also use the Xilinx vivado design kit 2018.2. I don't know if it is the reason for our network here. You can't open the connection. But I know what development environment to use, thank you again.

http://pavel-demin.github.io/red-pitaya-notes/development-machine/
http://pavel-demin.github.io/red-pitaya-notes/led-blinker/
https://www.dropbox.com/sh/5fy49wae6xwxa8a/AACl--BhQvcNgjeQLRaiX9dha/ClubVivado2016_Pavel_Demin.pdf?dl=1

cleverlau, 2018-09-15:
I refreshed the web link again, this time opened the connection you gave, thank you very much for your help.

http://pavel-demin.github.io/red-pitaya-notes/development-machine/
http://pavel-demin.github.io/red-pitaya-notes/led-blinker/
https://www.dropbox.com/sh/5fy49wae6xwxa8a/AACl--BhQvcNgjeQLRaiX9dha/ClubVivado2016_Pavel_Demin.pdf?dl=1

closed 2018-10-05 by pavel-demin

cleverlau, 2018-10-10:
Hello, Pavel, bother. 
When I run your routine “dac_test”, I always get the following error. After looking for a long time, I still can't find the answer. I have to ask for help.
best regards！

Closed #732.