pavel-demin/red-pitaya-notes #736: USB Audio with SDR Transceiver compatible with HPSDR

JS1MAV, 2018-10-03:
### Description of the setup:

 * Device: STEMlab 125-14
 * SD card image: Alpine SD Card Image
 * Application: SDR transceiver compatible with HPSDR
 * Other relevant information:

### Description of the problem:
Hi Pavel,

I have enjoyed WM8731 audio codec but I am attempted to use my external USB codec (E-MU 0204)  followed the description of "Audio from the HPSDR #269" in 2016. Now in 2018 , Alpine SD card image provides both alsa-utils and Linux kernel alsa sound set-up by default and I confirmed aplay, arecord, alsamixer e.t.c. commands usable and I am happy to hear much better sound quality than WM8731. 

Then, I added the last line of start.sh with "| aplay -fS16_BE -c2 -r48000 " and ran the program and expected the receiver sound from external USB codec but no sound come out.

I with to hear from you what I am missing to be able to use external USB audio.

Regards,

take

de JA5AEA
### Steps to reproduce the problem:

 1.
 2.
 3.

pavel-demin, 2018-10-04:
Hi Take,

The USB sound wasn't working well. The problems were discussed in https://github.com/pavel-demin/red-pitaya-notes/issues/305.

So, it was removed with the commit https://github.com/pavel-demin/red-pitaya-notes/commit/f6461973afd5ee82a7a57f934dfd343e68428477.

As far as I know, the most stable solution is to use a sound card attached to a computer. With a ASIO driver, this combination has the shortest delays.

Best regards,

Pavel

JS1MAV, 2018-10-06:
Hi Pavel and others,

Does anybody know "Audio Injector Ultra 2 sound card"?

It is designed for Rasberry Pi and provides 192kHz input/output and digital output + 2 mics and amazing three layer daughter boards configuration!

Used GPIO are

The shared pins are : 
5v power : 2, 4 
I2C (shared) : 3, 5 
GND : 6, 9, 14, 17, 20, 25, 30, 34, 39 
3.3v power : 1, 17

The reserved pins are : 
I2S : 12, 35, 38, 40 
Reset : 29

Price is around $25.

So far, this card appeals me as the replacement of my VM8731 proto card.

I am expecting to have any advice and comments positive and negative.

Regards,

take 

de JA5AEA

ja5aea, 2019-02-13:
Hi Pavel,

Sorry, It is been a long time after I talked about ""Audio Injector Ultra 2 sound card". At that time, it is still under kick starter project status. Then, I found the company offers VM8731 based sound cards with separate line-in and line-out connectors. Thus, I ordered one called "Zero Soundcard" and connected to my RedPitaya.

Headphone and microphone connection works.  RCA line-out also works. RCA line-in can be activated if RedPitaya .c file is appropriately modified to activate "Line In" setting of OpenHPSDR Transmit tag. Sound quality is enough for communication purpose. So, this is promising for me as the solution of TAPR hardware compatible architecture. However, I can hear frequent (less than one second period and typical synch noise nature) pop-up noise. I tried to resolve the problem with several configuration changes of Ether configuration with my Redpitaya and OpenHPSDR buffer sizes. Some improvement were observed but I can not totally remove it.

Now, I wish to ask you and readers whether somebody notice these pop-up noises with VM8731 board and know solutions to fix.

Any comments?

Regards,

take

de JA5AEA

P.S: I broke my WM8731 audio-codec obtained. Thus I can not see the difference between two at this moment but I remember pop-up noises were there but sound quality itself did not attract me to analyze these noises.

pavel-demin, 2019-02-18:
I have a couple of questions about your current setup.

Do you know what is the frequency of the oscillator installed on you new WM8731 board? If it's not 12.288 MHz, some parameters of the WM8731 should be changed.

Do you have the same pop-up noises when the audio is played through the computer sound card?

ja5aea, 2019-02-19:
Hi Pavel,

In fact, I have separate chats with Manfred and he raised the same question.  Separately he educated me about recent OpenHPSDR optional adoption of "PRO Latency" and "Resamper" and in fact they gave me quite low latency and comfortable audio quality at VAC connection to my PC sound card. 

Unfortunately, the crystal is only part mounted on the bottom side of PCB, therefore, it is hard to read 
[image]
frequency number after assembled. So today, I searched Audio Injector's forum but I could not find it. Thus finally, I decided to de-assembled it and found the number. It says "12.000". 

Do you have any specific parameter changes for 12.000MHz crystal?

I am waiting your response soon.

Regards,

take

de JA5AEA

pavel-demin, 2019-02-19:
I've just checked the WM8731 documentation. Looks like the only parameter to change is the USB mode bit in the Sampling Control register.

It can be done by adding 
```
i2c_write_addr_data8(i2c_fd, 0x10, 0x01);
```
just after the [line 484 in sdr-transceiver-hpsdr.c](https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/sdr_transceiver_hpsdr/server/sdr-transceiver-hpsdr.c#L484).

ja5aea, 2019-02-20:
Hi Pavel,

Quick report.

Much improved. I am also referring WM8731 manual and confirmed your instruction puts ADC and DAC  to 48kHz sampling mode with 12.000MHz MCLK FREQUENCY in Table 21 page 44.

I will precisely check the quality of both receiver and transmitter and report you later.

I noticed WM8731 is actually working with Master Mode i.e. WM8731 CODEC is providing clocks (BCLK, ADCLRC, DACLRC) to RedPitaya ARM at line 484 (set format register). Would you elaborate on the purpose of this master mode setting?

Regards,

take

de JA5AEA

pavel-demin, 2019-02-20:
Hi Take,

Thanks for confirming that setting USB mode bit works with the 12 MHz oscillator.

I've chosen the master mode because I don't know how to generate the 3.072 MHz clock using the resources available on the Red Pitaya board.

Best regards,

Pavel

pavel-demin, 2019-02-20:
I've just checked the slave mode description in the WM8731 documentation and found one more reason why I've chosen the  master mode. The slave mode requires an additional connection for MCLK or CLKOUT  that isn't always available on the boards with a crystal oscillator. For example, the MCLK and CLKOUT pins aren't connected to the external connector on the MIKROE-506 codec board.

Rolf-L, 2019-02-20:
Hi Take,
I'm using the WM8731 Codec board from MikroElektronika.
This Codec board is a part of my SDR TRX using the Red Pitaya Board.
For more see: 
http://dj7th.de (english side of Projekte)
https://www.mikroe.com/audio-codec-proto-board.
vy 73 de Rolf -DJ7TH-

ja5aea, 2019-02-21:
Hi Rolf,

Your assembly is beautiful. 

Pavel,

Thank you for describing the constrains why Audio codec have to run on master mode.

However, I can imagine RedPitaya ARM processor would be busy enough to handle over-run and under-run of three independent clocked blocks connected through UDP packets  i.e. 12.288MHz audio A/D and D/A clock, 125MHz RP A/D and D/A clock and OpenHPSDR on PC RTC clock.  We can not expect these three clocks are always synchronized within the tolerance not to produce the pop-up noise. That means  it depends on each station's environments.  I know I was stupid to have a terrible and night mare experiment to run the audio codec with 12.000MHz to demonstrate the issue.

I am comparing audio quality of VAC connection and Audio Injector Zero audio codec. When I listen on the local AM station, I notice occasional pop-up noise with audio codec. However, SSB monitoring on HF, there is no difference between VAC and audio codec. I wish to hear readers comments and experience who own WM8731 audio codecs.

Regards,

take

de JA5AEA

closed 2019-05-14 by pavel-demin